visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA #384: Ambient Temperature operates in Celsius, even when the unit is set to Fahrenheit

rosetown-kelly, 2022-10-04:
If I set the unit to F, the unit changes in the web interface, mqtt and on the hot tub display. 

However, the unit for the ambient temperature still seems to operate in C. This seems to result in inaccurate virtual temperature readings, as well as heating time estimates.

rosetown-kelly, 2022-10-05:
Thank you for that. However, when setting the temperature in the web UI, the scale for

rosetown-kelly, 2022-10-05:
[image]

Thank you for that tip on the different MQTT settings - I will definitely keep that in mind when setting up MQTT.

However, when setting the temperature on the web UI, the scale for ambient temperature remains on a Celsius scale, even when the spa is set to Fahrenheit. Pictured above, I have the spa set to F, but the highest ambient temperature I can set is 50. The current ambient temperature is 20C/68F. 

That is what lead me to believe the spa was still using C values for the ambient temperature, however maybe it's just a HTML display issue and if I set it via MQTT it would function properly.

visualapproach, 2022-10-05:
Ah I see! Well, there is two solutions: you can move closer to the pole, or I can fix the html.

Thank you for clarifying!

visualapproach, 2022-10-08:
@rosetown-kelly Hi again. I was checking the ambient slider and it does change unit already. However it didn't update the sliders properly. It is now fixed. Easiest way to update is to klick on the hamburger menu in top right corner and go to file upload. Choose the new websockets.js file. Done. (First ofc download that file from github)

closed 2023-03-13 by visualapproach