visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA #391: 4 wire virtual temp rising when pump off and "God mode" off

cobaltfish, 2022-10-19:
On Dev at commit 8d1ed9ace0ba9760135e84a0cf193d41eac9352e

I think this has introduced a new bug when not in "God" mode.  I've noticed with a set point of greater than ambient temp, and God mode off, the virtual temp keeps rising.  The timer for heating also keeps ticking over.
[image]
I rolled back to commit _ebf4638 4w: VT filter sensor readings_ and could see that the heating timer is no longer climbing when pump is stopped, and control not taken
I'm guessing its either something about the new hysteresis functionality, or the refactoring of the CIO::updateStates() for pump state / heat red state, but haven't got any further than diffing the code between the commits and looking at what has changed.

I will try and find time to cherry pick code between the 2 commit ids and see if I can give a better pointer to cause

visualapproach, 2022-10-19:
Thanks, I'll look at it and you are welcome to dig into it and see if you can spot the bug.

cobaltfish, 2022-10-19:
I think I'm getting there.
After swapping builds repeatedly, I noted differences in the "STATES" packet
in the old build, when in "normal" mode, and TGT < TMP, RED = 0, when TGT > TMP, **RED = 0**
in the new build  if TGT < TMP, RED = 0, and when TGT > TMP, **RED = 1**
I restored the "old" hysteresis and antifreeze code in updatePayload(), and commented out the new in loop(), and it seems to be working correctly again

visualapproach, 2022-10-19:
Looking at it now, and I've found one bug so far. Namely it tried to regulate temperature when not in "god" mode:
line 234 changed to:
`if(_cio.states[HEATSTATE] == 1) // in "normal" mode heatstate will be 2 (unknowable)`
Looking for more bugs...

visualapproach, 2022-10-19:
@cobaltfish I *think* that bug was doing what you described. Same error found and squashed in the antifreeze/overtemp functions. Can you try the latest dev plz?

cobaltfish, 2022-10-20:
Just deployed and testing now.  
Target temp at 39, actual at 30 and the heating time wasn't ticking up.  

I think I've found a new bug :-(
When controlling through the web UI, I can't turn off heating if in red mode!
If I do the following:
Current temp: 32
Target Temp: 37
Take Control -> On
Pump -> On
Heat -> On
then it starts heating, in Red mode

If I then click on Heat again to turn it off, the toggle moves, but then returns to on.
I also tried using the command queue method, of scheduling heat =0 with no repeat and target time of now, and no change
If I turn Pump off while it is still heating, the pump stops, but the heat mode stays red!
[image]

However, if I drop the temp target to less than current temp, say 30C, with heat mode in green, I can stop the heating as expected

I'm afraid I can't test the anti-freeze as the air and water temps are too high.  But I had looked at the anti-freeze section and wondered if it would need something to do with god mode.  
I guess the question is, do you want it to take control if the temp drops below 10, or do you only want to protect if its in control?

visualapproach, 2022-10-20:
Doh , I'll check it out tonight. Guess you saw my comments in the antifreeze code. Yes it can be handled differently depending on opinion. I'd like it to start heating and preserve other settings like target temp without heating to target until told so, but it's kind of not worth the effort. 
Thanks for your patience

cobaltfish, 2022-10-20:
Yep, looks good to me.  I'll let it do its scheduled run overnight and confirm in the morning :-)

As ever, appreciate your speedy replies

cobaltfish, 2022-10-21:
All good now, it ran for 4 hours, and both heat and pump stopped on schedule, and disabling control / god mode and the heat/heat counter behaving as expected:
[image]
(Ignore anything before midnight, that was just me manually testing)

closed 2022-10-21 by cobaltfish

visualapproach, 2022-10-21:
Awesome! Big thanks for helping! Have a nice weekend