visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA #394: Home Assistant - Ambient Temperature not controllable

RetroVape, 2022-10-24:
Layzspa was autodetected by the broker integration, but the ambient temperature is not controllable like the other controllable entities.

visualapproach, 2022-10-25:
4 or 6 wire? The ambient was supposed to be handled by a home automation system rather than manual, so it was intentionally omitted. I guess it may be convenient to arrange a selector helper or something.
Please show what you publish and the responses plz.

RetroVape, 2022-10-25:
Thank you for sharing your automation, exactly what I need. I am still new to HA. I have setup an automation with switches, and thought entities needed to be changeable in the GUI for HA automation to understand how to control the ambient. I didn’t think I could just publish an MQTT topic if the temperature outside changes.

I must have had a typo in MQTT test publish within the broker configuration. It looks to be working now after trying today. Ugh haha.

visualapproach, 2022-10-25:
Yeah it's a bit of a threshold to get the head around the yaml but google is our friend. Good to hear it worked out

closed 2022-10-26 by RetroVape