visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA #396: Virtual Temp in Fahrenheit is wrong

RetroVape, 2022-10-26:
Looks like the virtual temp for Fahrenheit is being converted incorrectly?

[image]
[image]

visualapproach, 2022-10-26:
Does it show the wrong value when flipping back to F again? My device does not do that when I test. What FW version do you have?

visualapproach, 2022-10-26:
You can try uploading `WebSocket.js` file from the data/ folder. Just download it from the dev branch and upload it via the web GUI file uploader menu.

RetroVape, 2022-10-26:
Same result after uploading the latest WebSocket.js

bottom of the web page says firmware version 10-8-22.

visualapproach, 2022-10-26:
Can you expand the STATES message and post here?
[image]

RetroVape, 2022-10-26:
[image]

visualapproach, 2022-10-27:
I wonder if it is related to a unit-press on the real display maybe. Don't know but I'll look at it in a few days when I get the time.

RetroVape, 2022-10-27:
Going off you last comment, I did change that from the control panel when I first setup. 

Switching back and forth in app and control panel didn’t do anything. However, I put it into C and then restarted the esp, and that did it. I am now reading correct.

visualapproach, 2022-10-27:
Okay that's good. Hard to know what went wrong. I set mine to F and restarted and it still works.

RetroVape, 2022-10-27:
haha whiplash, just checked it, back to 217.

RetroVape, 2022-10-27:
DO we need to run it with the pump on but heater off for a while to get a base line?

RetroVape, 2022-10-27:
Wondering if it jumps up like that when it goes through a heating cycle, I will turn off the heater to let it cool and watch it after I turn it back on.

RetroVape, 2022-10-27:
Well that was fast, no, it just jumps back up after a while.

visualapproach, 2022-10-27:
Can you upload the latest dev version so we have same base plz

visualapproach, 2022-10-27:
> DO we need to run it with the pump on but heater off for a while to get a base line?

To calibrate rate of cooling, yes, and you need to let it cool by at least 2 degrees, and not at ambient. But it shouldn't cause this error.

RetroVape, 2022-10-27:
Ok I will do the dev branch

RetroVape, 2022-10-27:
I installed the dev branch, it was fine until I reconnected the mqtt connection!

RetroVape, 2022-10-27:
this is the only automation I have, otherwise it’s just vanilla mosquito broker.

alias: "Update Layzspa ambient temperature "
description: ""
trigger:
  - platform: state
    entity_id: sensor.openweathermap_temperature
condition: []
action:
  - service: mqtt.publish
    data:
      topic: layzspa/command
      payload_template: >-
        {"CMD":14,"VALUE":{{ states("sensor.openweathermap_temperature")|round
        }},"XTIME":0,"INTERVAL":0}
mode: single

RetroVape, 2022-10-27:
Installed the dev branch, seems to still be doing it.

RetroVape, 2022-10-27:
Just does it after a little time. After 5 minutes it switched without changing the ambient.

I didn’t even realize the significance of the number, but it is doing the Celsius to Fahrenheit conversion again…

visualapproach, 2022-10-28:
Can you post your `settings.txt`? (menu/directory - click on settings.txt)

visualapproach, 2022-10-28:
Never mind, I managed to reproduce the fault. Only thing left is to find out wth is wrong... Wait a moment please

visualapproach, 2022-10-28:
Fix uploaded to dev.

RetroVape, 2022-10-28:
Dude thank you! I really appreciate it, this is a great project!

I will be able to test it tomorrow.

RetroVape, 2022-11-02:
I am back up and running and looks to be working. 

Took me awhile, I think I killed my esp in the process. Every time I would reset the esp it would get stuck with an exception I think on boot, at least that is what the serial monitor said. On whim I uploaded to a new esp and everything seems to work.

I really appreciate the quick fix!

RetroVape, 2022-11-02:
No on to automating the pump while not heating. That is what I plan to do with this feature, was that the intent?

visualapproach, 2022-11-02:
I also got boot resets after an OTA upload. When using USB it worked with the same exact code. I think VSC compiles wrongly sometimes (there is a thread here about that somewhere) or maybe there is a bug in the ESP or something.
The intention was to get an idea of the real temperature when pump is off, and to get finer resolution if possible. I was thinking you could get a more accurate temp keeping also, instead of +/- one degree, it can potentially be +/- 0.1 or so.

closed 2022-11-02 by RetroVape