visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA #403: Lay-Z-Spa restarts when web interface left open

rosetown-kelly, 2022-11-06:
Whenever I leave the web admin interface open on my device, the ESP8266 restarts frequently. When restarting, I will hear the chime sequence and the control panel will display show the IP address, before resuming normal operation. 

This only happens when the web interface is open, and even happens when the tab is in the background. However, as long as I don’t leave the interface open, everything works perfectly. I can administer everything perfectly over MQTT for extended periods without issue. 

The restarts don’t happen at a consistent frequency, but generally will happen within 2 minutes of the interface being open.

visualapproach, 2022-11-06:
FW_version?
What's the RSSI?
What's the reboot reason?

rosetown-kelly, 2022-11-06:
Firmware version: 2022-07-21
RSSI hovers between -45 and -50
Last boot: 2022-11-05, 8:41:41 p.m. Hardware Watchdog

visualapproach, 2022-11-06:
Thanks, I think you should update to latest version. 3 months ago (sept) changes were made to avoid wdt resets. I'd go for the dev branch which is latest code.

rosetown-kelly, 2022-11-06:
Thanks. I have been trying to update firmware for a bit now. I have a custom pinout however, and the new firmware seems to not like when I set a custom pinout:

I have: 

 bwc.begin(D3,D2,D1,D5,D6,D7,D8); //no params = default pins

I get the error:
main.cpp
too many arguments in function call C/C++(140) Line 22, Col 13

And if I try to upload: 

src\main.cpp:22:33: error: no matching function for call to 'BWC::begin(const uint8_t&, const uint8_t&, const uint8_t&, const uint8_t&, const uint8_t&, const uint8_t&, const uint8_t&)'

visualapproach, 2022-11-06:
Okay, calling back.begin() with arguments is deprecated and I forgot to remove that remark in main.cpp. Instead edit lib/bwc/bwc_6w_type1.h and change the pin out for your PCB there. I will update the remark

rosetown-kelly, 2022-11-06:
Perfect. Now that I have that figured out, I am running the latest firmware. Will update if the issue with Hardware Watchdog resets persists. 

Thanks again for all you do!

closed 2023-03-13 by visualapproach