visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA #415: target temperature at "238" then "20" after ESP restarts

torei, 2022-12-13:
steps to reproduce:
- in "SPA Config" disable the restore of states on startup
- back "Home" and disable all buttons but "Unit" (set this to celsius)
- set target temperature to 38°
- if you want to validate that the state restore is disabled, enable the pump (this should be disabled after startup)
- "ESP Restart"

result:
"Unit" is restored. that is fine - at least in my opinion.
somehow it seems that the startup sets the temperature to +200 = 238 in my case.
after a while it goes down to the minimum value, 20.
issue does not occur on "restore states on startup" enabled.

expected result:
the target temperatur should be restored, like it is usual behaviour in the original firmware (i think)

before restart:
[image]

after startup:
[image]

visualapproach, 2022-12-17:
@torei I'm looking in to it. I don't get the "238" but it drops to 20 which is not expected/wanted.

visualapproach, 2022-12-17:
update: I know where the 238 comes from. At startup the pump unit is always F and the ESP is converting 0 F to -18 C. -18 is the same as 238 depending on the variable type is signed or not. Working on...

visualapproach, 2022-12-17:
This issue, and also the reboot issue should be fixed now. Hope it's working for you too.

torei, 2022-12-17:
cool, thanks.
will try tomorrow

visualapproach, 2022-12-17:
I found another strange issue I need to solve. It sets target = actual temp and the physical buttons don't respond for a long time... I will inform when fixed

visualapproach, 2022-12-17:
Fixed! The pump started in power-off mode sometimes.

torei, 2022-12-19:
it seems like it is writing the cmdq.txt in every loop.
i guess that somehow the `_saveCmdqNeeded` is not properly set to false when it comes to `_saveCommandQueue()`

visualapproach, 2022-12-19:
Okay, I'll check

torei, 2022-12-19:
i think its set by `BWC::_handleCommandQ(void)`

visualapproach, 2022-12-19:
How do you know that it is saving cmdq.txt? The code looks fine with setting the save flag to false, and it's only called when adding or executing a command. Did you download the code today? I made some changes yesterday after our conversation.

visualapproach, 2022-12-19:
If you're running it on the bench it will add Gettarget command every 5 seconds or so until it gets a reading. Which won't happen.

torei, 2022-12-19:
oh okay, i will try to test it on the pump then.
no esp reset on hammering commands works :]

visualapproach, 2022-12-19:
Great 👍🏼

torei, 2022-12-20:
how do you debug with a pump attached?
is it possible OTA?

visualapproach, 2022-12-20:
Depends. You can send info over websockets. Or use a wireless serial but then you'd need to setup that library. Or you can connect a usb but keeping the 5V from CIO and USB separate.

torei, 2022-12-20:
okay, i cannot test that `_saveCommandQueue()f` for now.
but restart of ESP works fine, looks good :]
thanks for the fix

btw, "Ambient Temperature" and "Brightness" are not restored yet but i bet you have that on your screen.

closed 2022-12-20 by torei

visualapproach, 2022-12-20:
> btw, "Ambient Temperature" and "Brightness" are not restored yet but i bet you have that on your screen.

done. Thanks for your help!