visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA #422: brightness and ambient temp always reset to default value

AttiMM, 2022-12-27:
Hi @visualapproach, marry Xmas and happy new year! :-)
I'd like to let you know that some controls (in web GUI, Home page) always reset to their default value: brightness: 7, ambient temp: 20.
Why and how can I resolve?
Thx,
Attila

visualapproach, 2022-12-27:
Thanks and happy holidays! Those issues were addressed 7 days ago in the development branch. Check your firmware and report if it doesn't work.

AttiMM, 2022-12-28:
Dear @visualapproach 
I'm sorry for my many questions, but I'm stuck again...

1. I could find and download the new release (v3.5.1). I compared the new files with the previous ones in v3.5.0 and I can see there are files modified under \lib\BWC and \src - which means to me I should upload the new firmware via VSC Upload first, and then I can upload all the modified and new files under \data - am I right?

2. After I uncompressed WiFi-remote-for-Bestway-Lay-Z-SPA-3.5.1.zip and opened \Code\6-wire-version\platformio.ini in VSC, I  modified \lib\BWC\model.h according to my tub, and also the last lines of platform.ini for the OTA upload:
```
; upload_port = COM4
 upload_protocol = espota
 upload_port = 192.168.0.16
 upload_flags = 
    --auth=esp8266
```
3. using the right arrow in left bottom, I try to Upload the firmware OTA according to the given info above but it's failed mainly because of "unrecognized command-line option '--auth=esp8266' "
[image]

What have I done wrong?

visualapproach, 2022-12-28:
1. Correct 
2. Good
3. No idea. Try "clean all" under platformio menu and recompile

visualapproach, 2022-12-28:
Did you comment out the 'esptool' line also?

AttiMM, 2022-12-28:
Yes, of course. Clean all is done, but Upload failed again :(

visualapproach, 2022-12-29:
It should look like this (IP instead of layzspa.local is ok). Try restarting your computer, it looks weird.
```
; Uncomment the lines below by removing semicolons and edit IP for OTA upload.
; You have to upload via USB cable the first time. (upload_protocol = esptool)
; Make sure to use a data-USB cable. There is power only cables that wont work.
; upload_protocol = esptool
; edit com port if not found automatically (if you have CH340 UART in some versions of platformio)
; upload_port = COM4

upload_protocol = espota
upload_port = layzspa.local
upload_flags =
  --auth=esp8266
```

AttiMM, 2022-12-29:
I'm sorry, restarting computer didn't solve this problem :(

visualapproach, 2022-12-30:
I test compiled the development version as is. It worked fine. So you must have a broken setup or maybe you edited something unintentionally. If you try to compile it (not upload it) directly as is after download and it does not work, you have something weird in your VCS installation.

closed 2022-12-31 by visualapproach

AttiMM, 2022-12-31:
Please try it again after you set up for OTA uploading. If I set up for USB uploading (commented out the lines the problematic lines) I can build (and upload via USB) it without any problem.