visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA #429: Ready in: Never

AttiMM, 2023-01-11:
Hi @visualapproach ,
It seems to me that Ready in (hours) is not working. 
I started to heating up from 20°C to 39°C. After a lot of hours the current temperature was 35°C but "Ready in:" showed "Never". And so it continued until it reached the target temp.
I always set up Ambient Temp according to the real temp.
Firmware: 2022-12-20 Model: MALDIVES2021 RSSI: -70

See some screenshots.
[image]
[image]
[image]

visualapproach, 2023-01-11:
It's possible that it's because it's not calibrated yet. To calibrate: let the pump on while cooling a couple of degrees. Keep heater off and don't change anything.

AttiMM, 2023-01-11:
Hm... I didn't know about the calibration method - I'll try it!

visualapproach, 2023-01-11:
Also FYI see my discussion post on wip v4

closed 2023-03-13 by visualapproach