visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA #436: Bootloop with latest

eried, 2023-01-27:
Hi, I tried to install latest release and my esp stopped responding :O so I had to disassemble the pump. I want to say that:

MQTT: CONNECTED
Firmware: 2022-12-20
Model: MIAMI2021

This version has been running flawlessly now (I had a very old one before) but I really like the new UI and the new selectors for temperature. Any idea why it could have entered in a non responsive OTA loop?

BTW, I still think the brightness should go up only 1 level when the UI pressed :P but that could be an option, I find it much more pleasant, specially if is in 0 brightness all the time as I set mine.

Skeletorjus, 2023-01-27:
I have experienced bootloops after OTA. Have you tried clean/rebuild and flashing via cable?

visualapproach, 2023-01-27:
What @Skeletorjus said and kudos to @torei for the selectors. Roger on the brightness. Will try to remember in v4. Wip

eried, 2023-01-27:
ok thanks, via cable strangely some versions where rebooting when the webgui was being opened. Since it was wet and cold I just tried to find a version that worked ok (2022-12-20; which is working perfect, no reboots, no weird data points on temperature now). I was wondering if it was something expected via OTA

visualapproach, 2023-01-27:
I think the OTA sometimes get corrupted. If it's the platformio, the esp or some interrupt disturbing the process I don't know. It only happens rarely.

closed 2023-02-16 by eried