visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA #437: Boots, plays chime, displays version and IP, then.... nothing.

thedavestewart, 2023-02-04:
So the headline says it all I think. It boots. Plays the chime. Displays the startup info on the screen as expected. Then I can connect to wifi, and get into the control panel, and nothing responds. 

I'm using:
Pump Model S100105 (It's the Saluspa "Laguna" from Costco)
PCB v2
TXS0108E LLC
Wires are done up 1:1 (picture attached)
Programmed using the MIAMI2021 Definitions and PCB_V2

I've tried double checking and tracing all my solder joints appear to be good. I've tried flashing and re-flashing the 8266 several times... the boot up chime and display leads me to believe that I'm at least on the right track, but no joy so far. 

Other interesting things: In the web UI, if I tap any of the buttons (heater, pump, bubbles, etc.) it very quickly flips on and off, but no action on the pump side. 

Anyone who's encountered anything similar or can point me in a direction would be greatly appreciated! 

[image]

visualapproach, 2023-02-05:
Suggestions:
Try red LLC and rewire and use PCB_V2B. Probably work better.
Or try without LLC. You can solder to the empty holes by the ESP. Use V2B here as well.

thedavestewart, 2023-02-05:
I'll try the V2B rewire without the LLC first - I've got Red LLC's on order, but they're gonna take a while to get here. Thanks!! I'll post results here.

thedavestewart, 2023-02-05:
Happy to report: 
Wired per V2B spec. Re-flashed using V2B in model.h 
KEPT the blue LLC in place. 
Powers on and functions like a charm!!! 

So maybe (as I was reading in other threads) the red LLC isn't necessarily the fix? 

Thank you again for pointing me in the right direction. 

(Also, if you or anyone reads this, and there's something dangerous/improper/harmful with leaving the blue LLC in place, let me know and I wont install it permanently yet)

rgregg, 2023-02-06:
I'm just in the process of building the same setup for the same pump, so I'm going to give this a try as well. I'm having difficulty parsing the difference between V2 and V2B configuration on the wiring. Is the delta just cross the D3 and D5 pinout wires between the two plugs?

thedavestewart, 2023-02-06:
@rgregg If you were wiring it straight through before (like in my picture above) where the top plug is male, bottom is female, and Wires are 1-6. Your color scheme might vary, but going by my photo, 
1 (yellow)
2 (white)
3 (blue)
4 (red)
5 (green)
6 (black). 

Abbreviations are as follows: 
Male Plug = MP
Female Plug = FP

So putting it all together:
MP1 = The First Wire (Yellow) on the Male Plug. Etc. 

Top Solder Pad (H1A)- Male Plug (except for 1 wire)
Solder in this order (based on Pad Numbers):
5V - MP1 (My Color: Yellow)
G  - MP2 (My Color: White)
D1 - MP3 (My Color: Blue)
D2 - MP4 (My Color: Red)
D3 -FP5 (My Color: Green)
NC - MP6 (My Color: Black)

Bottom Solder Pad (H2A) - Female Plug (except for 1 wire)
Solder in this order (based on Pad Numbers):
5V - FP1 (My Color: Yellow)
G  - FP2 (My Color: White)
D4 - FP4 (My Color: Red)
D5 - MP5 (My Color: Green)
D6 -FP3 (My Color: Blue)
D7 - FP6 (My Color: Black)

Sorry for the overly pedantic writeup... I'm visual and I have an easier time with colors. :P

rgregg, 2023-02-06:
That was super helpful, thanks @thedavestewart

rgregg, 2023-02-10:
Wanted to leave my confirmation here as well - I wired per the V2B instructions, used the blue LLC, no resisters/capacitors, and everything is working amazingly well. So great to be able to automate / remote control the hot tub now!

visualapproach, 2023-02-10:
> Sorry for the overly pedantic writeup... I'm visual and I have an easier time with colors. :P

@thedavestewart Be proud instead! I wish everybody would be that precise when reporting here. Even myself ;-)

rgregg, 2023-02-10:
@visualapproach is there a source doc for the PDF installation guide? I'd like to help by updating it to have the latest instructions but can't edit the PDF directly.

closed 2023-02-18 by thedavestewart