visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA #460: error when compiling v3.5.1 in Arduino: cannot pass objects of non-trivially-copyable type

Pilot001, 2023-03-04:
### Discussed in https://github.com/visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA/discussions/459

<div type='discussions-op-text'>

<sup>Originally posted by **Pilot001** March  4, 2023</sup>
During compilation of v3.5.1 in Arduino I get the follow error message:
```
bestway.ino: In function 'void handleESPInfo()':
bestway:2559:31: error: cannot pass objects of non-trivially-copyable type 'class String' through '...'
     ESP.getMaxFreeBlockSize() );
                               ^
```
Any ideas how to fix that?
Thank you in advance!
And BTW: Thanks for this great project!</div>

visualapproach, 2023-03-04:
You are welcome. No need to double post please. Closing issue

closed 2023-03-04 by visualapproach

Pilot001, 2023-03-04:
I'm sorry, this was a suggestion of github, to create an issue out of the discussion.

visualapproach, 2023-03-04:
Aha, no worries mate