visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA #464: Bestway 2021 Square Model Installation Instructions

eemitev, 2023-03-11:
I am noticing that most of the instructions and diagrams are targetting the "Eggshaped" pump. Are there any pictures or references on how to open the "2021 Square Model"?

Thanks!

visualapproach, 2023-03-11:
There is a thread somewhere about that. Look in issues, both open and closed. And maybe in discussions.

eemitev, 2023-03-11:
I went digging through old issues. I did not find pictures of the disassembly process but I found some text instructions.

It appears that the pump should be open by spliting in half. The tilting control panel with the buttons and display is *not* the right spot to start taking apart. Control panel should be left alone.

https://github.com/visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA/issues/54#issuecomment-1084504124

closed 2023-03-11 by eemitev