visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA #472: [FEATURE RQ] BestWay Pool heater

GSzabados, 2023-03-17:
**Is your feature request related to a problem? Please describe.**
The other day I have seen this product in a shop and thought that setup looks quite similar to what any of the pumps have.
https://www.bestwaystore.co.uk/pool-heater.html

**Describe the solution you'd like**
I think that there might be people who would like to start and stop that heater remotely according their PV output.

**Describe alternatives you've considered**
As the device has an RCD a smart plug cannot start it.

**Additional context**
I don't have this product, neither a pool. But thought it would be cool to have support for it.

Jay-Ti, 2023-03-17:
I highly doubt the insides are similar to the blowup hottubs heaters. I only got the cartridge sized heater so i cant really tell. I can only tell getting rid of the rcd and connecting it to an rcd protected wall plug isnt hard.

GSzabados, 2023-03-17:
> I can only tell getting rid of the rcd and connecting it to an rcd protected wall plug isnt hard.

I am even not sure how a smart plug would handle it. I've read after my suggestion, that it is total 2.8kW and you need a dedicated socket/breaker for it.

Jay-Ti, 2023-03-17:
Youre from the uk your lines also run on 230v 16A so 3600W,no?
As long as you dont have any other big draw on the line you should be fine with existing wirering building code also wants RCDs for your homes.

closed 2023-12-01 by github-actions[bot]