visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA #481: display not working or showing dim temp but no buttons work

starsx2016, 2023-03-26:
**Hardware (please complete the following information):**
 - Pump model 54123 Miami eggshape pre2021
 -  6 wires
 - PCB type v2
 - Level shifter type red one

**Software (please complete the following information):**
 - Version latest 4.1.1

hi,
since a update to newest version of firmware , display don't work anymore or show very dim the temp and no buttons work but web interface work perfectly. when I plug the cio directly to dsp without the esp it work like it should.

I don't know if it's a power/current issue (may be not enough current to make the display work properly)
do you thing a cap between vcc and gnd could help?

thanks

Matt

visualapproach, 2023-03-26:
I am working on general stability improvements. Let's see if your problems remain after next update. Not sure when I will upload it. In a couple of days maybe

visualapproach, 2023-03-29:
Uploaded new code (still named 4.1.1 but with some fixes)

starsx2016, 2023-03-29:
hi!
still no luck after updating to your bug fix version
[image]
[image]
[image]

visualapproach, 2023-03-29:
Why does it say PRE2021 when you have a 54123?

starsx2016, 2023-03-29:
it’s 54124 miami (2019-2020) 6-wires

starsx2016, 2023-03-29:
[image]

starsx2016, 2023-03-29:
i will give it a try

starsx2016, 2023-04-01:
I try again the version I had last year and still the same issues, so I check if my pin def. match yours but I notice that my pinout goes like this:

       cio.                             DSP
yours | mine.          YOURS | MINE
5v.     |   5v                5V.     |.    5V
gnd   |   GND.          GND.   |.   GND
D1     |  D7.                D6.     |.   D5
D2    |  D2.                D4.     |.   D4
D5    |  D1.                D3.     |.    D3
NC   |  NC.                D7.     |.    D6

so I can control everything from the web without problem in this config pinout, but display no, but still weird as last year it was working fine on 5.2 with the same pinout.....
where in the code I can change pin assignation?

thanks

Matt

visualapproach, 2023-04-02:
You set the pinout on the hardware config page, not in the code.

starsx2016, 2023-04-05:
yeah, I did that but still no luck with 4.1.1 . I made some change to my custom pub (thicker track and shorter) so I'll see when I'll get them in a week...

thanks for now!

starsx2016, 2023-04-23:
Hi!
problems fixed! with my new pub with thicker tracks and a new llc (red one), everything work fine on latest version on my old pump and on my new square pump (S100104)2021 Tahiti .

thanks for all the good works you done with this!

Matt

closed 2023-11-23 by visualapproach