visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA #487: [BUG] Calibration not working on v4 with 4 wire spa

cobaltfish, 2023-04-06:
Having got v4 running on my spa, I've noticed a couple of things
First, that leaving the spa running on pump only, no heat, it doesn't calibrate.  In settings.json VTCAL remains false too.
I also noted if settings.json has it set to true the settings page doesn't show that it is true

I've done some debugging and made a fork which I think fixes the problems.

I first add saveDebugInfo to dump strings from the `_updateVirtualTempFix_ontempchange()` method in lib/BWC_unified/bwc.cpp so I could see where it was getting to, what it thought it was doing and why in all the various returns.

I saw quite a few times that even though the heater hadn't run for quite some time it was exiting at `if(_heatred_change_timestamp_ms < _temp_change_timestamp_ms) return;`
my debug text showed that `heated too recently _heatred_change_timestamp_ms: 800169  _temp_change_timestamp_ms: 2571028`
I dug through the previous versions, and noted that v4 looks to use a timestamp, rather than a relative time since change. And that v3 would loop through an array of state_age and set any that have changed to 0, rather than setting it to a point in time
I'm not sure what the best fix is, but when I switched the logic, it then didn't return when it had been running on pump/filter only for 5+ mins
I did similar to `if(cio->cio_states.heatred || cio->cio_states.bubbles || (_bubbles_change_timestamp_ms < _temp_change_timestamp_ms)) return;` as this also started exiting before calculations could be made

In order to resolve the settings page not showing the status of calibrated, I ensured in the json object returned for settings
I added into `BWC::getJSONSettings()`
`doc[F("VTCAL")] = _vt_calibrated;`

It took me a while to prove that it had worked (too much insulation, it was taking 3+ hours at times to get the 1 degree change) but I have put it into a branch / fork are here:
https://github.com/cobaltfish/WiFi-remote-for-Bestway-Lay-Z-SPA/commits/Fix/VirtualTempCalculations

I think this should work for 6 wire too

visualapproach, 2023-04-06:
Cool, I'll check it out. Thank you

visualapproach, 2023-04-09:
I was unpatient and edited the code and added your name to the commit. Good job spotting the error.

closed 2023-04-09 by visualapproach