visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA #488: PlatformIO Compile Error

dietmar-1, 2023-04-07:
I tried to compile latest version 4.1.1, and platformio terminated compilaton with an error concerning umm_malloc - Memory Manager:

```
In file included from src\main.cpp:1:0:
src\main.h:30:40: fatal error: umm_malloc/umm_heap_select.h: No such file or directory
 #include <umm_malloc/umm_heap_select.h>
```
 
to solve the problem for me, i change the [env:] section in platformio.ini to

```
[env:nodemcuv2]
platform = espressif8266@^3

```
I hope this works...

visualapproach, 2023-04-08:
Maybe you had an earlier version of the platform cached, in which case a "clean all" would help

goofy79, 2023-04-23:
I have the same problem, with this 

[env:nodemcuv2]
platform = espressif8266@^3

visualapproach, 2023-04-23:
You probably have an old installation. Clean all and try platform = espressif8266@^4

visualapproach, 2023-04-23:
Click PIO icon, Quick access/Platforms, updates. Update platform ESP8266

goofy79, 2023-04-23:
This work -> Clean all and try platform = espressif8266@^4
Update of the ESP8266 Platform won't work

closed 2023-11-23 by visualapproach