visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA #493: [OTHER]

paulandrewcrouch, 2023-04-14:
- Pump model, for example S100202 (Square)
 - 6 wires to pump, 4 to display (in a 6 plug)
 - PCB v2
 - Red LLC
 - Software (from GitHub yesterday)
 
 Thought all was going well, Node MCU flashed, integrated into Home Assistant with MQTT, on the Wi-Fi fine, website loads ok.  Soldered the cables onto the board and thought amazing, just need to plug in.  
 
Sadly, display on the pump all lights/buttons light up and 888 on the display.   Can access the web page but the buttons do nothing. 

Any advice appreciated.

visualapproach, 2023-04-14:
Have you configured the hardware in the web ui?

paulandrewcrouch, 2023-04-14:
> Have you configured the hardware in the web ui?

To the best I can….  because I don’t know which one to select I’ve tried the different 6 wire pump models and different 4 wire displays.   Have I missed something obvious…?   Thanks for the quick response.

Jay-Ti, 2023-04-14:
You have a pump and most likely a Display belonging to said pump, right? Then There should also be a Model number somewhere on the pump, if not, send a picture, and maybe we can take some educated guesses.

paulandrewcrouch, 2023-04-14:
> You have a pump and most likely a Display belonging to said pump, right? Then There should also be a Model number somewhere on the pump, if not, send a picture, and maybe we can take some educated guesses.

[image]

paulandrewcrouch, 2023-04-14:
[image]

paulandrewcrouch, 2023-04-14:
[image]
[image]

visualapproach, 2023-04-15:
> > Have you configured the hardware in the web ui?
> 
> To the best I can…. because I don’t know which one to select I’ve tried the different 6 wire pump models and different 4 wire displays. Have I missed something obvious…? Thanks for the quick response.

Aha, I'm pretty sure you have a 4 wire on both ends. The spa manufacturer seems to throw in whatever they have on hand sometimes. In this case a 6-wire cable. So you should set both the pump and display to a 4-wire model. Both should be the same.  Don't know which, but there is only 5, of which only 54138 and 54173 has hydrojets.

paulandrewcrouch, 2023-04-15:
I was full of hope, but I'm definitely all out of ideas now, tried every option (together), lots of permutations if I try different pump from board (which I have tried a few).   Even took it apart today to see if I could see model numbers but I don't know where to look.   I must be so close, yet so far away!!

Thanks for you help though, much appreciated.   Am determined to sort this.

visualapproach, 2023-04-15:
Usually the number is on a big sticker on the outside of the pump.

paulandrewcrouch, 2023-04-15:
[image]

paulandrewcrouch, 2023-04-15:
[image]
[image]
[image]

visualapproach, 2023-04-15:
Okay so you have a S100202. It's not listed so you may be the first to try it. I'll see if we can figure out what goes on. Call you back with advice.

paulandrewcrouch, 2023-04-15:
> Okay so you have a S100202. It's not listed so you may be the first to try it. I'll see if we can figure out what goes on. Call you back with advice.

That really kind.  I wondered if I should swap some wires (like in the instructions) where the red is swapped over?

visualapproach, 2023-04-17:
Hi again @paulandrewcrouch . From your last message I understand that you haven't checked the pinout. With that I mean checked which wire from the pump is TX and which is RX. Correct?
Since there are only two combinations you can of course just try to flip them. But if there is more issues than just this, it would be beneficial to actually know that the wiring is correct.

How to find CIO TX (transmit line):
Turn pump OFF.
Disconnect both the ESP and display from the pump.
Turn pump ON
Use a multimeter to check voltage between GND and the data pins on the cable coming from the pump. The one with a reading more than 0V is CIO TX. Do not short the pins!

The CIO TX wire should now be connected according to your PCB setting. If using PCB V2 as in the picture, it should be connected to D2. The other wire from CIO (RX) goes to D1.
DSP RX would be the same wire as CIO TX before we added the ESP module. So it goes to D5.
[image]

paulandrewcrouch, 2023-04-17:
Ok, so tried a few different combinations today too (just checked and all works fine when plugged back in normally so nothing blown).   Across the four active wires, I'm getting about 4.78V from all three (the 5V feed and the other two wires, presumably Tx and Rx).   Same every time, all lights come on and stay on.

visualapproach, 2023-04-17:
"Four active lines" is ambiguous. Did you measure whilst connected to the esp or as I suggested?

visualapproach, 2023-04-17:
Okay, I guess there is pull-ups on the lines then. Very well, update via web ui to the beta version and go to page layzspa.local/debug-on/
You need to enter the address manually with the trailing '/'
Let it run for a few minutes and go to layzspa.local/dir/
Click on log.txt and post the text here
You may want to go to .../debug-off/ to stop recording

When 'take control' is not active the esp copies the messages between the display/cio. So it should work as long as the protocol is right. Serial 9600 is assumed in your case.

paulandrewcrouch, 2023-04-17:
1681759272:0,0,0,0,0,0,0,0,0,0,0	,0,0,0,0,0,0,0
1681759465:0,0,0,0,0,0,0,0,0,0,0	,0,0,0,0,0,0,0

paulandrewcrouch, 2023-04-17:
I’ve resoldered onto another board and also put in spade connectors so I can easily turn them.   Yes tried changing the pins over, to no avail.  

Am a bit stumped.  Will try doing it  in the software rather than the wires to see if it makes any difference.

could it be the LLC?

wrong PCB?

paulandrewcrouch, 2023-04-18:
Tried the software change - no difference.   I feel I've done something wrong, but don't know what...   Connecting the PCB, Node MCU and LLU seem to not allow an acknowledgement signal, yet log.txt didn't put up any activity.

T1gg1, 2023-04-18:
Same Issue here
Wemos D1 mini with blue LLC. Changed the LLC, same issue. 
For one minute the esp shows the correct temperature. I clicked some switche but they switched back. After I changed the hardware config, the bubbles startet for about 3 seconds. But I could reproduce it :D

My log is the following:

1681846621:0,0,0,0,0,0,0,0,0,0,0	,F8,DA,0,0,0,0,0
1681846621:0,0,0,0,0,0,0,0,0,0,0	,E3,BA,0,0,0,0,0
1681846621:0,0,0,0,0,0,0,0,0,0,0	,71,FF,0,0,0,0,0
1681846621:0,0,0,0,0,0,0,0,0,0,0	,6A,B5,0,0,0,0,0
1681846621:0,0,0,0,0,0,0,0,0,0,0	,97,6A,0,0,0,0,0
1681846622:0,0,0,0,0,0,0,0,0,0,0	,FF,FF,0,0,0,0,0
1681846622:0,0,0,0,0,0,0,0,0,0,0	,2A,7C,0,0,0,0,0
1681846622:0,0,0,0,0,0,0,0,0,0,0	,89,27,0,0,0,0,0
1681846624:0,0,0,0,0,0,0,0,0,0,0	,7B,F3,0,0,0,0,0
1681846624:0,0,0,0,0,0,0,0,0,0,0	,DD,4E,0,0,0,0,0
1681846624:0,0,0,0,0,0,0,0,0,0,0	,8F,FE,0,0,0,0,0
1681846624:0,0,0,0,0,0,0,0,0,0,0	,D8,61,0,0,0,0,0
1681846624:0,0,0,0,0,0,0,0,0,0,0	,3B,3B,0,0,0,0,0
1681846624:0,0,0,0,0,0,0,0,0,0,0	,EA,FF,0,0,0,0,0
1681846624:0,0,0,0,0,0,0,0,0,0,0	,6D,54,0,0,0,0,0
1681846625:0,0,0,0,0,0,0,0,0,0,0	,29,B6,0,0,0,0,0
1681846625:0,0,0,0,0,0,0,0,0,0,0	,4A,71,0,0,0,0,0
1681846625:0,0,0,0,0,0,0,0,0,0,0	,F1,23,0,0,0,0,0
1681846625:0,0,0,0,0,0,0,0,0,0,0	,BE,7B,0,0,0,0,0
1681846625:0,0,0,0,0,0,0,0,0,0,0	,71,40,0,0,0,0,0
1681846625:0,0,0,0,0,0,0,0,0,0,0	,5F,82,0,0,0,0,0
1681846625:0,0,0,0,0,0,0,0,0,0,0	,D6,73,0,0,0,0,0
1681846626:0,0,0,0,0,0,0,0,0,0,0	,FF,FF,0,0,0,0,0
1681846626:0,0,0,0,0,0,0,0,0,0,0	,68,BC,0,0,0,0,0
1681846626:0,0,0,0,0,0,0,0,0,0,0	,45,D6,0,0,0,0,0
1681846626:0,0,0,0,0,0,0,0,0,0,0	,BB,6B,0,0,0,0,0
1681846626:0,0,0,0,0,0,0,0,0,0,0	,EF,CE,0,0,0,0,0
1681846626:0,0,0,0,0,0,0,0,0,0,0	,85,9,0,0,0,0,0
1681846627:0,0,0,0,0,0,0,0,0,0,0	,FA,2D,0,0,0,0,0
1681846627:0,0,0,0,0,0,0,0,0,0,0	,FF,FF,0,0,0,0,0
1681846627:0,0,0,0,0,0,0,0,0,0,0	,1B,38,0,0,0,0,0
1681846627:0,0,0,0,0,0,0,0,0,0,0	,E1,5B,0,0,0,0,0
1681846627:0,0,0,0,0,0,0,0,0,0,0	,BE,78,0,0,0,0,0
1681846627:0,0,0,0,0,0,0,0,0,0,0	,F9,FF,0,0,0,0,0
1681846628:0,0,0,0,0,0,0,0,0,0,0	,D6,58,0,0,0,0,0
1681846628:0,0,0,0,0,0,0,0,0,0,0	,FC,D3,0,0,0,0,0
1681846628:0,0,0,0,0,0,0,0,0,0,0	,F2,27,0,0,0,0,0
1681846628:0,0,0,0,0,0,0,0,0,0,0	,C,9D,0,0,0,0,0
1681846628:0,0,0,0,0,0,0,0,0,0,0	,82,78,0,0,0,0,0
1681846628:0,0,0,0,0,0,0,0,0,0,0	,96,FC,0,0,0,0,0
1681846629:0,0,0,0,0,0,0,0,0,0,0	,5B,DE,0,0,0,0,0
1681846629:0,0,0,0,0,0,0,0,0,0,0	,AA,F2,0,0,0,0,0
1681846629:0,0,0,0,0,0,0,0,0,0,0	,99,1F,0,0,0,0,0
1681846629:0,0,0,0,0,0,0,0,0,0,0	,8C,75,0,0,0,0,0
1681846629:0,0,0,0,0,0,0,0,0,0,0	,A3,AF,0,0,0,0,0
1681846630:0,0,0,0,0,0,0,0,0,0,0	,24,1E,0,0,0,0,0
1681846630:0,0,0,0,0,0,0,0,0,0,0	,8E,63,0,0,0,0,0
1681846630:0,0,0,0,0,0,0,0,0,0,0	,19,F9,0,0,0,0,0
1681846630:0,0,0,0,0,0,0,0,0,0,0	,C6,E0,0,0,0,0,0
1681846630:0,0,0,0,0,0,0,0,0,0,0	,73,5B,0,0,0,0,0
1681846631:0,0,0,0,0,0,0,0,0,0,0	,77,21,0,0,0,0,0
1681846631:0,0,0,0,0,0,0,0,0,0,0	,D4,17,0,0,0,0,0
1681846631:0,0,0,0,0,0,0,0,0,0,0	,87,19,0,0,0,0,0
1681846631:0,0,0,0,0,0,0,0,0,0,0	,B4,17,0,0,0,0,0
1681846631:0,0,0,0,0,0,0,0,0,0,0	,FF,FF,0,0,0,0,0
1681846632:0,0,0,0,0,0,0,0,0,0,0	,89,54,0,0,0,0,0
1681846632:0,0,0,0,0,0,0,0,0,0,0	,3D,DC,0,0,0,0,0
1681846632:0,0,0,0,0,0,0,0,0,0,0	,E8,E5,0,0,0,0,0
1681846632:0,0,0,0,0,0,0,0,0,0,0	,8C,AA,0,0,0,0,0
1681846632:0,0,0,0,0,0,0,0,0,0,0	,D4,67,0,0,0,0,0
1681846634:0,0,0,0,0,0,0,0,0,0,0	,B7,22,0,0,0,0,0
1681846634:0,0,0,0,0,0,0,0,0,0,0	,3F,45,0,0,0,0,0
1681846634:0,0,0,0,0,0,0,0,0,0,0	,87,3D,0,0,0,0,0
1681846635:0,0,0,0,0,0,0,0,0,0,0	,CE,FF,0,0,0,0,0
[... 246 lines cut ...]
1681846696:0,0,0,0,0,0,0,0,0,0,0	,87,B6,0,0,0,0,0
1681846698:0,0,0,0,0,0,0,0,0,0,0	,4F,76,0,0,0,0,0
1681846698:0,0,0,0,0,0,0,0,0,0,0	,33,A4,0,0,0,0,0
1681846698:0,0,0,0,0,0,0,0,0,0,0	,7D,5F,0,0,0,0,0
1681846699:0,0,0,0,0,0,0,0,0,0,0	,88,CB,0,0,0,0,0
1681846699:0,0,0,0,0,0,0,0,0,0,0	,FD,FF,0,0,0,0,0
1681846699:0,0,0,0,0,0,0,0,0,0,0	,3F,7B,0,0,0,0,0
1681846699:0,0,0,0,0,0,0,0,0,0,0	,B8,C2,0,0,0,0,0
1681846699:0,0,0,0,0,0,0,0,0,0,0	,F7,FF,0,0,0,0,0
1681846700:0,0,0,0,0,0,0,0,0,0,0	,FB,FF,0,0,0,0,0
1681846700:0,0,0,0,0,0,0,0,0,0,0	,3C,9C,0,0,0,0,0
1681846700:0,0,0,0,0,0,0,0,0,0,0	,DB,35,0,0,0,0,0
1681846700:0,0,0,0,0,0,0,0,0,0,0	,D,B,0,0,0,0,0
1681846701:0,0,0,0,0,0,0,0,0,0,0	,E4,F8,0,0,0,0,0
1681846701:0,0,0,0,0,0,0,0,0,0,0	,8C,D4,0,0,0,0,0
1681846701:0,0,0,0,0,0,0,0,0,0,0	,7B,F,0,0,0,0,0
1681846702:0,0,0,0,0,0,0,0,0,0,0	,9A,5,0,0,0,0,0
1681846702:0,0,0,0,0,0,0,0,0,0,0	,6,1B,0,0,0,0,0
1681846702:0,0,0,0,0,0,0,0,0,0,0	,71,7E,0,0,0,0,0
1681846702:0,0,0,0,0,0,0,0,0,0,0	,4B,E6,0,0,0,0,0
1681846703:0,0,0,0,0,0,0,0,0,0,0	,F3,FF,0,0,0,0,0
1681846703:0,0,0,0,0,0,0,0,0,0,0	,A9,61,0,0,0,0,0
1681846703:0,0,0,0,0,0,0,0,0,0,0	,E9,83,0,0,0,0,0
1681846703:0,0,0,0,0,0,0,0,0,0,0	,97,F4,0,0,0,0,0
1681846703:0,0,0,0,0,0,0,0,0,0,0	,84,DE,0,0,0,0,0

T1gg1, 2023-04-18:
After I switched the pins:

1681847014:0,0,0,0,0,0,0,0,0,0,0	,73,5E,0,0,0,0,0
1681847014:0,0,0,0,0,0,0,0,0,0,0	,89,FF,0,0,0,0,0
1681847014:0,0,0,0,0,0,0,0,0,0,0	,DB,9,0,0,0,0,0
1681847016:0,0,0,0,0,0,0,0,0,0,0	,AF,11,0,0,0,0,0
1681847016:0,0,0,0,0,0,0,0,0,0,0	,73,9B,0,0,0,0,0
1681847016:0,0,0,0,0,0,0,0,0,0,0	,D3,14,0,0,0,0,0
1681847016:0,0,0,0,0,0,0,0,0,0,0	,24,2A,0,0,0,0,0
1681847016:0,0,0,0,0,0,0,0,0,0,0	,14,F,0,0,0,0,0
1681847016:0,0,0,0,0,0,0,0,0,0,0	,7F,FA,0,0,0,0,0
1681847017:0,0,0,0,0,0,0,0,0,0,0	,A0,96,0,0,0,0,0
1681847017:0,0,0,0,0,0,0,0,0,0,0	,6F,44,0,0,0,0,0
1681847017:0,0,0,0,0,0,0,0,0,0,0	,FE,D3,0,0,0,0,0
1681847017:0,0,0,0,0,0,0,0,0,0,0	,4D,B4,0,0,0,0,0
1681847017:0,0,0,0,0,0,0,0,0,0,0	,54,EA,0,0,0,0,0
1681847017:0,0,0,0,0,0,0,0,0,0,0	,DF,FA,0,0,0,0,0
1681847017:0,0,0,0,0,0,0,0,0,0,0	,6A,74,0,0,0,0,0
1681847018:0,0,0,0,0,0,0,0,0,0,0	,18,D3,0,0,0,0,0
1681847018:0,0,0,0,0,0,0,0,0,0,0	,92,EF,0,0,0,0,0
1681847018:0,0,0,0,0,0,0,0,0,0,0	,38,11,0,0,0,0,0
1681847018:0,0,0,0,0,0,0,0,0,0,0	,3F,7A,0,0,0,0,0
1681847018:0,0,0,0,0,0,0,0,0,0,0	,23,9,0,0,0,0,0
1681847018:0,0,0,0,0,0,0,0,0,0,0	,D9,70,0,0,0,0,0
1681847018:0,0,0,0,0,0,0,0,0,0,0	,BE,F0,0,0,0,0,0
1681847019:0,0,0,0,0,0,0,0,0,0,0	,D6,89,0,0,0,0,0
1681847019:0,0,0,0,0,0,0,0,0,0,0	,1E,E9,0,0,0,0,0
1681847019:0,0,0,0,0,0,0,0,0,0,0	,C8,3A,0,0,0,0,0
1681847019:0,0,0,0,0,0,0,0,0,0,0	,BE,D1,0,0,0,0,0
1681847019:0,0,0,0,0,0,0,0,0,0,0	,F,BD,0,0,0,0,0
1681847019:0,0,0,0,0,0,0,0,0,0,0	,9F,23,0,0,0,0,0
1681847020:0,0,0,0,0,0,0,0,0,0,0	,EE,FC,0,0,0,0,0
1681847020:0,0,0,0,0,0,0,0,0,0,0	,FF,FF,0,0,0,0,0
1681847020:0,0,0,0,0,0,0,0,0,0,0	,5C,F6,0,0,0,0,0
1681847020:0,0,0,0,0,0,0,0,0,0,0	,B4,99,0,0,0,0,0
1681847020:0,0,0,0,0,0,0,0,0,0,0	,CE,5F,0,0,0,0,0
1681847020:0,0,0,0,0,0,0,0,0,0,0	,3E,1C,0,0,0,0,0
1681847021:0,0,0,0,0,0,0,0,0,0,0	,77,3B,0,0,0,0,0
1681847021:0,0,0,0,0,0,0,0,0,0,0	,FC,9E,0,0,0,0,0
1681847021:0,0,0,0,0,0,0,0,0,0,0	,AF,7E,0,0,0,0,0
1681847021:0,0,0,0,0,0,0,0,0,0,0	,BE,19,0,0,0,0,0
1681847021:0,0,0,0,0,0,0,0,0,0,0	,3C,F6,0,0,0,0,0
1681847022:0,0,0,0,0,0,0,0,0,0,0	,3F,F2,0,0,0,0,0
1681847022:0,0,0,0,0,0,0,0,0,0,0	,FD,FF,0,0,0,0,0
1681847022:0,0,0,0,0,0,0,0,0,0,0	,8,81,0,0,0,0,0
1681847022:0,0,0,0,0,0,0,0,0,0,0	,B7,33,0,0,0,0,0
1681847022:0,0,0,0,0,0,0,0,0,0,0	,DF,AF,0,0,0,0,0
1681847023:0,0,0,0,0,0,0,0,0,0,0	,AA,DF,0,0,0,0,0
1681847023:0,0,0,0,0,0,0,0,0,0,0	,CF,8F,0,0,0,0,0
1681847023:0,0,0,0,0,0,0,0,0,0,0	,F,90,0,0,0,0,0
1681847023:0,0,0,0,0,0,0,0,0,0,0	,22,BC,0,0,0,0,0
1681847023:0,0,0,0,0,0,0,0,0,0,0	,62,BD,0,0,0,0,0
1681847024:0,0,0,0,0,0,0,0,0,0,0	,FC,46,0,0,0,0,0
1681847024:0,0,0,0,0,0,0,0,0,0,0	,A9,DC,0,0,0,0,0
1681847024:0,0,0,0,0,0,0,0,0,0,0	,4E,FA,0,0,0,0,0
1681847025:0,0,0,0,0,0,0,0,0,0,0	,D7,98,0,0,0,0,0

visualapproach, 2023-04-18:
@T1gg1 you didn't state what pump model you have.

T1gg1, 2023-04-18:
Same as paulandrewcrouch

T1gg1, 2023-04-18:
Logfile with that config:  4 wire, 54123

1681847983:0,0,0,0,0,0,0,0,0,0,0	,0,0,0,0,0,0,0
1681848356:0,0,0,0,0,0,0,0,0,0,0	,0,0,0,0,0,0,0
1681848403:FE,FE,FF,FF,FC,F8,1E,0,0,0,0	,0,0,0,0,0,0,0
1681848405:FE,FE,FE,FE,FF,F9,FC,0,0,0,0	,0,0,0,0,0,0,0
1681848447:FF,FF,FE,FE,FE,F9,FC,0,0,0,0	,0,0,0,0,0,0,0
1681848618:FC,FE,FE,FE,FF,F9,FC,0,0,0,0	,0,0,0,0,0,0,0
1681848620:FE,FE,FE,FE,FE,F8,8E,0,0,0,0	,0,0,0,0,0,0,0

visualapproach, 2023-04-18:
Ok can you log when set to a 4-wire in config. And use the latest beta version uploaded an hour ago

visualapproach, 2023-04-18:
Weird data from cio and nothing from dsp...
Does anyone of you guys happen to have a logic analyzer?

T1gg1, 2023-04-18:
4 wire, 54154

1681849551:FE,FE,FF,FC,FF,F8,FC,0,0,0,0	,0,0,0,0,0,0,0,0,0,0,0
1681849554:FE,FF,FF,FE,FE,FA,FE,0,0,0,0	,0,0,0,0,0,0,0,0,0,0,0

T1gg1, 2023-04-18:
@visualapproach yes I have one

visualapproach, 2023-04-18:
Cool, would be helpful to see logs on the two data wires without the esp present.

T1gg1, 2023-04-19:
[digital.csv](https://github.com/visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA/files/11274999/digital.csv)
I did it threeway so I connected channel 0/1 to display and pump. It started, I turned it on, turned the bubbles on, turned bubbles off and turned the complete pump off. I hope that help.
Should I connected it otherway? 
Channel0 = fourth contact
Channel1 = third contact

T1gg1, 2023-04-19:
I don´t know if that help and how but I hope so :D

visualapproach, 2023-04-19:
Can you post a screenshot of the waveform graph where the timing is visible? I looked at the data in excel but don't have a good program to view it in present form and I couldn't be bothered to make one. It looks like a higher frequency than the usual 9600 baud. More like 57600 from a guesstimate

T1gg1, 2023-04-19:
[image]
[image]
[image]
Like this?
With async serial 9600 analyzer it shows the third picture
I tried it with 57600 and then it shows the foruth picture
[image]

visualapproach, 2023-04-19:
nice, what program are you using and what's the filetype of the samples?

visualapproach, 2023-04-19:
okay, I'll look at it when I have time. If you post the file :-)

T1gg1, 2023-04-20:
[Session 0.zip](https://github.com/visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA/files/11280549/Session.0.zip)
I put the plug in, activate it, reduce the temperature, start the bubbles, end the bubbles and deactivate it and pull the plug

visualapproach, 2023-04-20:
Thx, it seems like another protocol than the rest of the models. I'm interested in cracking it, but its not obvious to me how to go about it yet. Are you able to see the chip names and we may be lucky to find a datasheet.

paulandrewcrouch, 2023-04-20:
It’s so strange that a production line pump would have such variations.   I did wonder if it was simply baud rate.  

Also so annoying following your tremendous work we seem to have a model that for some reason wants to play differently. 

Thanks for all your support so far.

T1gg1, 2023-04-21:
[image]
[image]
[image]
@visualapproach for the display it is a esp wrom 02??? the rest is completely sealed. I can not see the backside.
Thank you for your help

@paulandrewcrouch as usual. If you find a way to not use their cloud and app, they will find a way that you have to.

paulandrewcrouch, 2023-04-21:
I did wonder what bypassing that would do.   Later models appear to still work though, so we just have an odd variant?

wyliecoyoteuk, 2023-04-21:
I have an earlier pump with an ESP wifi module, #54184
Currently using V4, PCB v2 and pre2021.
Remote control works, but none of the buttons or display work.

visualapproach, 2023-04-21:
I didn't realize we were talking about wi-fi pumps. Can you make a simple drawing of how it's connected? Will probably not make any difference, but I'm just curious.

paulandrewcrouch, 2023-04-21:
4 wires from CIO to DSP, then 4 from DSP to wireless chip.   Perhaps I could just tasmotize the chip (with a backup) that's already there?

paulandrewcrouch, 2023-04-21:
Sorry 6 from DSP to chip.

visualapproach, 2023-04-21:
So CIO -- 4wires -- DSP -- 6wires -- ESP32?
Is it possible to tap off the signals between ESP32 and DSP?

visualapproach, 2023-04-22:
You may have an option here:
https://community.home-assistant.io/t/lay-z-spa-hot-tub-wi-fi-pump-automation/229006/46

paulandrewcrouch, 2023-04-22:
Yep done that successfully, it just still uses the cloud and intercepts APIs, what I loved about yours was the local nature of it and some of the extra controls.   I'm tempted to try bypassing the internal WiFi and see if it acts as a normal pump.   I assume the delay is waiting for the ESP32 to boot up.

I want the thing off the cloud.   Will check the wires later today.   I'm genuinely really grateful for you taking all this time out to reply.   Very kind of you.

T1gg1, 2023-04-22:
@visualapproach you see it in my picture. On my pump from DSP to ESP are 4 wires.
Can´t access the pins because of the sealing

wyliecoyoteuk, 2023-04-22:
> I didn't realize we were talking about wi-fi pumps. Can you make a simple drawing of how it's connected? Will probably not make any difference, but I'm just curious.
I thought that might be the reason for the difference in protocols.

paulandrewcrouch, 2023-04-22:
My apologies - 4 wires all the way through, sealed, but would pull out I’m sure. 
[image]
[image]
[image]

paulandrewcrouch, 2023-04-23:
I think it's fair to deduce from the letters that White=Rx, Red=Tx, Yellow=GND, Black=5V.   I presume flashing with Tasmota 32 would be dangerous rather than smart?   I could use Tasmotizer to back up.

I don't believe I've got the tools to tap off the signals, I would be willing to get the plug out of the gel though to access the 4 pins.

T1gg1, 2023-04-23:
@visualapproach are you sure it is so complicated? I saw guys in threads with the same pump and your code works. Why @paulandrewcrouch and I have a pump with problems? Maybe we did something wrong? Or the producer cracked the firmware?😅

visualapproach, 2023-04-23:
@paulandrewcrouch I guess Tasmota wouldn't help if we don't know how the ESP32 is supposed to communicate with the display. If we did it could be doable. Or with ESPHOME or similar. Or a port of this project preferably ;-). Now I don't think you should risk bricking your pump to get information that I'm not sure we can use.
@T1gg1 No, I'm not sure it's complicated but the file you provided (thx) is discouraging. The timings are all over the place. Pulses as short as tens of nanoseconds and up to milliseconds. It doesn't look like a plain old serial signal. And when comparing line 0 with line 1 in your file, the signals starts at almost exactly the same time which makes me think they are both generated from the same device (instead of one beeing transmitted from cio and the other from dsp). I don't know, maybe a measurement problem, or noise, but it's confusing. Maybe sampling the signals from cio only, without the dsp/ESP plugged in will tell us something.
If you're talking about @wyliecoyoteuk I got the impression he has an older version of the pump so it could be different. Or are there other reports that you refer to?
@wyliecoyoteuk It's weird that it works from web ui but not from display. If you're on the beta version you can start logging by going to layzspa.local/debug-on/ and stopping it after a few minutes at layzspa.local/debug-off/
You now have a log.txt file you can examine or post here. Press some buttons on the display while logging.

T1gg1, 2023-04-23:
@visualapproach I can´t find the thread. No it wasn´t @wyliecoyoteuk, it was in a german forum.
I can cut the DSP and analyze it again. Then I will upload the new file

paulandrewcrouch, 2023-04-27:
> @paulandrewcrouch I guess Tasmota wouldn't help if we don't know how the ESP32 is supposed to communicate with the display. If we did it could be doable. Or with ESPHOME or similar. Or a port of [...]

The ESP32 is in the DSP.   Do you think it could simply be bypassed, I presume if unplugged then it's like a normal pump?   Maybe I just stick with the API integration - which does work fine, I just wanted to take it local.

wyliecoyoteuk, 2023-04-27:
The ESP in my pump is on a separate board. disconnecting it just stopped the pump working

paulandrewcrouch, 2023-04-27:
> The ESP in my pump is on a separate board. disconnecting it just stopped the pump working

Ahhh, won't try that then.   Thanks.   I think it's time to put the pump back together and use the chip that's already there.   The API integration with Home Assistant is perfectly fine - I just wanted it 100% local.

visualapproach, 2023-04-27:
I totally understand that but there's not much I can do with available data at the moment.
Maybe if you get the display driver chip model but it may be erased or impossible to read through the gel.

T1gg1, 2023-04-28:
[Session Without Disp.zip](https://github.com/visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA/files/11354790/Session.Without.Disp.zip)
This is the session without display plugged in

[10 more replies]

closed 2023-12-01 by github-actions[bot]