visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA #494: [BUG] 4 Wire NO54144 heating at half power

cobaltfish, 2023-04-20:
When using the spa over the weekend, I noticed it seeming to struggle to maintain temp, even though it wasn't cold outside.

Looking at energy consumption via smart meter / plug socket, I noted that it was consuming about 1kw, rather than about 2kw.
I then ran the heaters directly rather than remotely, and saw that it was using 1.6-1.8kw

I noticed the "Set full power" command in the schedule command queue, but doing that had made no difference to power consumption.

I tried changing the HEATBITMASK1 and HEATBITMASK2 in Code/lib/cio/CIO_54144.h so it didn't have the soft start, and once deployed, saw the increased power consumption:
```
-        const uint8_t HEATBITMASK1 =   B00110000;  //48  heater stage 1 = 50%
-        const uint8_t HEATBITMASK2 =   B01000000;  //64  heater stage 2 = 100%
+        const uint8_t HEATBITMASK1 =   B01110000;  //48  heater stage 1 = 50%
+        const uint8_t HEATBITMASK2 =   B01110000;  //64  heater stage 2 = 100% 
```
(I am reset back to source on version 2023-04-18-2103 via firmware uploader at present, and code updated to commit 8b33bc0)

At first I thought it might be a race condition somewhere, as I sometimes find I need to press "Take Control" a few times for it to apply.   But I think that was one of the things you just fixed in 8b33bc0 / fix 4W bug in beta

I've had a further look through the code, and I think the case / action for the command SETFULLPOWER is missing within bwc.cpp.

I see it in the Commands array of enums.h, but nowhere else.   

In v3 in Code/4-wire-version/lib/BWC4W/BWC_8266_4w.cpp there is a case SETFULLPOWER within BWC::_handleCommandQ(void):
```
case SETFULLPOWER:
          if(_commandQ[0][1]){
            if(ALLOWEDSTATES[_currentStateIndex][HEATTOGGLE] == 2) {
              _cio.heatbitmask = HEATBITMASK1 | HEATBITMASK2;
            }
          }
          else
            _cio.heatbitmask = HEATBITMASK1;
          break;
```

Should there be something like this for the v4 branch?  

**Hardware:**
 - Pump model 54144
 - 4 wire
 - PCB V1
 - Level shifter TXS0108E

**Software:**
 - Firmware: 2023-04-18-2103
 - Branch: development_v4

visualapproach, 2023-04-20:
Thanks for your report, I'll look into it

visualapproach, 2023-04-23:
fixed in beta version. I had a local stack variable instead of a class member, meaning the timer for heater2 delay would never trigger since it counted zeroes all the time. Doh!
The command "set full power" is just hanging there not doing anything. I may make it possible to choose full/half power from mqtt if I get in the mood.

cobaltfish, 2023-04-23:
Just deployed it, and ran a 5 min test of the pump / heater.  Looks good to me, was seeing it max out at about 1.8Kw rather than the 1.0Kw previously.  
Sorry if I sent you looking in the wrong direction for the full power stuff.  I think I had even looked at the _HEATER2_DELAY_MS but clearly didn't follow it through the code to see what that did end to end.  I'll confess it now, I'm more of a sys admin 
 than programmer / developer, so its just that I know enough to be dangerous!

closed 2023-04-23 by cobaltfish

visualapproach, 2023-04-23:
I'm impressed that you followed the code to find the bitmasks. The non blocking code is hard to follow even for me, and I wrote it Lol. Good to hear it works now, have a great day.