visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA #498: Calibrated not checked but virtual temp and time to ready working

Mr-Groch, 2023-04-28:
Hi, I've upgraded from old 3.x version (from 2nd half of 2022) to newest last week. I'v noticed that in web ui "Calibrated" checkbox appeard, that can't be checked manually and it is always unchecked. My virtual temp and time to ready is working quite well. I'm also using HA automation to update ambient temp value. Pump is on 12h daily, in past 5 days I haven't been using lay-z-spa (heater, air bubbles), and Calibration still unchecked...

visualapproach, 2023-04-28:
Hi. I'll look for potential bugs. Here's some info about the calibration:
Checkmark is designed to show if the automatic calibration is done. You may only uncheck it, to do a new calibration.
Water pump needs to run for 5 minutes or more before calibration starts.
During calibration you may not have/switch on heat, bubbles or jets.
Measured temperature must drop at least one degree.
Water temp must not be same or close to ambient temperature.
Default value before calibration should be in the right ballpark, but intention is to make it as precise as possible with the calibration. Nice to hear you confirming that.

If all of the above is true for you, please tell me.

Mr-Groch, 2023-04-28:
>Water pump needs to run for 5 minutes or more before calibration starts.

It is working 12h daily (Automation in HA is switching it ON at 8AM and OFF at 8PM) for about 2 weeks now

>During calibration you may not have/switch on heat, bubbles or jets.

I hasn't switched any of it since about 5 days

>Measured temperature must drop at least one degree.
>Water temp must not be same or close to ambient temperature.

This is happening almost every day from about 5PM to 8PM, this is example where temperature dropped 2 degrees when pump was on, and ambient temp was more than 5 degrees lower:

[image]

visualapproach, 2023-04-28:
Just to be picky: was the heater OFF all the time during temp decrease, or did some automation or maybe a glitch flick it? (Check history for the climate control by pressing the three dots menu)

visualapproach, 2023-04-28:
I was more interesed in the logbook below the graphs. You don't have to post, just check if heater was turned on during above period.

Mr-Groch, 2023-04-28:
And longer period, with all my recorder hostory (7 days)

[image]

visualapproach, 2023-04-28:
Ah, great, seems like it should work. I'm looking into it

visualapproach, 2023-04-28:
There was a recent bugfix regarding this from @cobaltfish. Please update to v4.1.2

Mr-Groch, 2023-04-28:
Yes, that was the case. I've udated to newest release and "calibrated" checkbox became checked an hour later. Thanks!

closed 2023-04-29 by visualapproach