visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA #499: Control after update does not work

DandeMC, 2023-04-29:
Hello,

I updated my PCB from a version from 2022 to the current one. Before that I ran Erase on PlatformIO to avoid errors. My pump is version 54129 and has 6 cables. The PDF describes the 54129 with 4 cables. Unfortunately I don't know which PCB version I have either. I think the V1. I'm trying to attach a photo.
My problem is that I can see changes in the web interface that I make on the pump (C to F) but can't control anything from the web interface. Currently I have set "6 wire, pre 2021" as hardware.
Is it possible that I have to delete the ESP in another way because there is still data from the previous version? If yes how?

[image]

visualapproach, 2023-04-29:
That should be same pinout as PCB_V1. What remaining data are you talking about? Did you build and upload the filesystem? There is no need to erase anything on the ESP. It will be overwritten with a new upload.

DandeMC, 2023-04-29:
Yes I transferred both. Do you know why I can't control it on the web interface? And why is my pump model in the documentary with 4 wires and my is with 6?

visualapproach, 2023-04-29:
What remaining data are you talking about?
> Do you know why I can't control it on the web interface?

No, but it will be less difficult to find out if you answer the question.
> And why is my pump model in the documentary with 4 wires and my is with 6?

You have to ask the manufacturer why they do such strange things as labelling two different pumps with the same model number.

So did you set the hardware page to PRE2021, PRE2021 and PCB_V1. Saved and restarted?

DandeMC, 2023-04-29:
It saved the wifi settings. After the erase, it was gone.

Yes I had set it to pre2021 and PCB v1. It was the only way to see the right infos in the web interface. But I can't control.

visualapproach, 2023-04-29:
It saves the wifi between updates in a special memory, which is convenient and normal.
Can you control from display panel?

DandeMC, 2023-04-29:
Aah okay. Nice to know ☺️

Yes, on control panel I can control it and it updates on the web interface the display and shows the correct data. Only on web interface I can't control it.

visualapproach, 2023-04-29:
go to http://layzspa.local/debug-on/ (you can click the link) then press LOCK on display. Then click here http://layzspa.local/debug-off/
Go to http://layzspa.local/dir/ and copy the log.txt file here

DandeMC, 2023-04-29:
Its a little bit short. Is this right? 

[log.txt](https://github.com/visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA/files/11359990/log.txt)

visualapproach, 2023-04-29:
Yes it's good. You have PRE2021. Can you show the main web page also?

DandeMC, 2023-04-30:
Now it gets very weird. I have always tested it with Edge on Windows 11 and had the problems there. I was just on my Macbook with Edge and had no problems. Everything works there. Then I cleared the cache on the PC and opened a private window in Edge. Same mistake. All addons turned off. Same mistake. It also works on my iPhone with Safari. Do you have any idea why that is?

[image]

visualapproach, 2023-04-30:
It explains why I only ever use edge to download chrome

DandeMC, 2023-04-30:
Okay? And whats the problem? Can we solve it?

wyliecoyoteuk, 2023-04-30:
Use Chrome, Chromium, Opera or Firefox.
Edge is based on the same Chromium code as Chrome.
I wouldn't use Edge or Bing because of the privacy issues.
I use Firefox and Duckduckgo.

DandeMC, 2023-05-01:
Don't we want to find a solution for this?

wyliecoyoteuk, 2023-05-01:
The problem is almost certainly with the Edge browser on Windows, not the web page.
There may be a setting in Windows or Edge that is stopping the controls working, it could be a permissions issue, or windows security settings blocking the execution of code.
Download Firefox or Chrome browser and try that, if it still fails, then it is a Windows setting.
As you say it works on your Mac with Edge and on your iPhone with Safari.
Are you using Windows 11 by any chance?  
I don't use Windows at home but I support customers who do, and I know that a lot of stuff is currently broken in 11.

wyliecoyoteuk, 2023-05-01:
I have just tested it with Edge on a Windows 11 VM, and it works perfectly.
So it is probably a setting in your Edge or Windows installation.

closed 2023-11-23 by visualapproach