visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA #501: [BUG] Command queue

goofy79, 2023-04-30:
I have Daily Commander to Switch On and off the Pump for „Automatic cleaning“ but id i Heat the Pool, this take for example 1 Day.
If the command for Switch on the Pump and Switch Off in this Heating Time, the heater will stop heating. 

I habe the fw Version [2023-04-24-1854]

visualapproach, 2023-04-30:
Not exactly sure what you mean but I think what you want to do is
Turn off pump (will also turn off heat)
Turn on heat (will also turn on pump. If you just turned on pump, the heat will be off)

goofy79, 2023-04-30:
ok i have daily commands which turn the pump on at 8am and off at 10am, then on again at 12pm and off at 2pm and then finally on at 9pm and on at 11pm clock off
If I want to heat up my pool for tomorrow, I turn on the heater until it has the desired temperature.
But that doesn't work in this case, because at 10:00 a.m. at the latest, the pump switches off and the heating is also switched off.
So I can no longer heat my pool, at least not properly as long as I have the commands active.

Turning the heating on at 10:00 a.m. doesn't make sense either, since I don't want to heat the pool every day.

Possibly it would help to query via software, if it is a command that has the pump off and the heating is on at the same time, then don't do anything.

visualapproach, 2023-05-01:
Okay I understand what you mean now. It's admittedly a limitation. I As I see it, the best option would be to have an "enable command queue" switch. When user manually change the states the ESP could ask if command queue should be turned off. Doing it by default would not be wanted by all users all the time. On the other hand, how would we confirm it if the user just presses a display button. I guess it could be done on the 6 wire systems with some text+input, but not on the 4 wire.
Another option would be to use "set ready" function, which could inhibit the command queue until after the "ready time".

visualapproach, 2023-05-02:
I looked into this and it's a bit involved. It's on my todo list for now

visualapproach, 2023-08-06:
Update: We can now edit single commands. Make the settings as usual and press "SET" on a specific command to replace it with the new data. Deleting and more features are on the horizon :-)
You are welcome to try beta version.

closed 2023-11-23 by visualapproach