visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA #504: [BUG] Loading Web Interface causes device to drop pings and never loads

jondunning, 2023-05-03:
**Describe the bug**
Loading Web Interface causes device to drop pings and never loads. 

**To Reproduce**
Steps to reproduce the behavior:
Open Chrome and browse to IP of the device

**Expected behavior**
Web should load and pings not expected to drop

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Hardware (please complete the following information):**
 - Helsinki Spa
 - 6 wires
 - PCB v1

**Software (please complete the following information):**
Firmware: 2023-04-24-1854

**Additional context**
MQTT seems to still operate during this issue

jondunning, 2023-05-03:
Screen recording below;

https://user-images.githubusercontent.com/6370173/236046811-69ffb9cf-4ef5-4a54-9d48-2b19e7ef0b08.mov

jondunning, 2023-05-03:
Unifi Screenshot of connection between ESP and AP

[image]

visualapproach, 2023-05-03:
Hi! I don't know why you have that problem. I thought maybe the ESP's own AP is on but then MQTT wouldn't work I guess. Maybe something with the flash file system? Does the .../dir/ page work? Or .../info/
Or maybe something is blocking in your router? Guest network perhaps? Have you tried with any other applications on the ESP?

jondunning, 2023-05-04:
Both directories load fine, results below;

**/dir/**

[bootlog.txt](http://192.168.200.175/bootlog.txt) Size: 1057 Bytes [remove](http://192.168.200.175/remove/?FileToRemove=bootlog.txt)
[chkupdatefw.html.gz](http://192.168.200.175/chkupdatefw.html.gz) Size: 1452 Bytes [remove](http://192.168.200.175/remove/?FileToRemove=chkupdatefw.html.gz)
[cmdq.json](http://192.168.200.175/cmdq.json) Size: 9 Bytes [remove](http://192.168.200.175/remove/?FileToRemove=cmdq.json)
[config.html.gz](http://192.168.200.175/config.html.gz) Size: 3055 Bytes [remove](http://192.168.200.175/remove/?FileToRemove=config.html.gz)
[debug.html.gz](http://192.168.200.175/debug.html.gz) Size: 759 Bytes [remove](http://192.168.200.175/remove/?FileToRemove=debug.html.gz)
[favicon.ico](http://192.168.200.175/favicon.ico) Size: 9662 Bytes [remove](http://192.168.200.175/remove/?FileToRemove=favicon.ico)
[favicon.png](http://192.168.200.175/favicon.png) Size: 1817 Bytes [remove](http://192.168.200.175/remove/?FileToRemove=favicon.png)
[function.js.gz](http://192.168.200.175/function.js.gz) Size: 491 Bytes [remove](http://192.168.200.175/remove/?FileToRemove=function.js.gz)
[furelise.tune](http://192.168.200.175/furelise.tune) Size: 524 Bytes [remove](http://192.168.200.175/remove/?FileToRemove=furelise.tune)
[hwcfg.json](http://192.168.200.175/hwcfg.json) Size: 69 Bytes [remove](http://192.168.200.175/remove/?FileToRemove=hwcfg.json)
[hwconfig.html.gz](http://192.168.200.175/hwconfig.html.gz) Size: 2479 Bytes [remove](http://192.168.200.175/remove/?FileToRemove=hwconfig.html.gz)
[hwtest1.jpg](http://192.168.200.175/hwtest1.jpg) Size: 10533 Bytes [remove](http://192.168.200.175/remove/?FileToRemove=hwtest1.jpg)
[hwtest2.jpg](http://192.168.200.175/hwtest2.jpg) Size: 9432 Bytes [remove](http://192.168.200.175/remove/?FileToRemove=hwtest2.jpg)
[hwtestinfo.html.gz](http://192.168.200.175/hwtestinfo.html.gz) Size: 879 Bytes [remove](http://192.168.200.175/remove/?FileToRemove=hwtestinfo.html.gz)
[index.html.gz](http://192.168.200.175/index.html.gz) Size: 2711 Bytes [remove](http://192.168.200.175/remove/?FileToRemove=index.html.gz)
[index.js.gz](http://192.168.200.175/index.js.gz) Size: 3497 Bytes [remove](http://192.168.200.175/remove/?FileToRemove=index.js.gz)
[main.css.gz](http://192.168.200.175/main.css.gz) Size: 1852 Bytes [remove](http://192.168.200.175/remove/?FileToRemove=main.css.gz)
[manifest.json](http://192.168.200.175/manifest.json) Size: 149 Bytes [remove](http://192.168.200.175/remove/?FileToRemove=manifest.json)
[menu.png](http://192.168.200.175/menu.png) Size: 223 Bytes [remove](http://192.168.200.175/remove/?FileToRemove=menu.png)
[mqtt.html.gz](http://192.168.200.175/mqtt.html.gz) Size: 1445 Bytes [remove](http://192.168.200.175/remove/?FileToRemove=mqtt.html.gz)
[mqtt.json](http://192.168.200.175/mqtt.json) Size: 259 Bytes [remove](http://192.168.200.175/remove/?FileToRemove=mqtt.json)
[remove.html.gz](http://192.168.200.175/remove.html.gz) Size: 788 Bytes [remove](http://192.168.200.175/remove/?FileToRemove=remove.html.gz)
[settings.json](http://192.168.200.175/settings.json) Size: 295 Bytes [remove](http://192.168.200.175/remove/?FileToRemove=settings.json)
[states.txt](http://192.168.200.175/states.txt) Size: 34 Bytes [remove](http://192.168.200.175/remove/?FileToRemove=states.txt)
[success.html.gz](http://192.168.200.175/success.html.gz) Size: 671 Bytes [remove](http://192.168.200.175/remove/?FileToRemove=success.html.gz)
[upload.html.gz](http://192.168.200.175/upload.html.gz) Size: 862 Bytes [remove](http://192.168.200.175/remove/?FileToRemove=upload.html.gz)
[visualapproach.png](http://192.168.200.175/visualapproach.png) Size: 8177 Bytes [remove](http://192.168.200.175/remove/?FileToRemove=visualapproach.png)
[webconfig.html.gz](http://192.168.200.175/webconfig.html.gz) Size: 1537 Bytes [remove](http://192.168.200.175/remove/?FileToRemove=webconfig.html.gz)
[wifi.html.gz](http://192.168.200.175/wifi.html.gz) Size: 2073 Bytes [remove](http://192.168.200.175/remove/?FileToRemove=wifi.html.gz)
[wifi.json](http://192.168.200.175/wifi.json) Size: 260 Bytes [remove](http://192.168.200.175/remove/?FileToRemove=wifi.json)

**/info/**

Stack size:          268 
Free Heap:           20488 
Min  Heap:           17168 
Core version:        3.1.2 
CPU fq:              160 MHz
Cycle count:         410041422 
Free cont stack:     1520 
Sketch size:         677312 
Free sketch space:   2465792 
Max free block size: 16208

visualapproach, 2023-05-04:
Great, so it all seems normal and the ESP server can respond. Only serving the html files not working. Did it work in earlier versions or is this your first install?
If you open the inspect/console in the browser, what is the output when trying to load the page?

jondunning, 2023-05-04:
I've re-flashed the file system ota and it's now loading fine, very strange as it was working then suddenly stopped without any changes.

dasfliege, 2023-05-07:
I'm having the same issue again as well. Running latest dev version.
Did a clean flash several times now, re-uploaded filesystem, etc.... First the UI was working until i rebootet the ESP. Now i don't get the UI to open at all. Favico is loading but the rest of the UI stays blank. /info/ and /dir/ are working fine though:

[image]

[image]

visualapproach, 2023-05-07:
Do you have multiple pages open in different browsers? Or mqtt updates super frequently?

dasfliege, 2023-05-07:
No. Also tried different browsers. MQTT is disabled.

visualapproach, 2023-05-07:
Okay, my guess is web server is getting OOM (out of memory). If you want you can uncomment all the -Ddebug lines in platformio.ini and build+upload, and check the serial monitor for OOM message. You need to have it hooked up with USB to a computer. It's weird though that only some of you have this problem. Maybe connected to a certain router or something. Does it make a difference if you're on dhcp or static IP? Just fumbling in the dark here.

dasfliege, 2023-05-08:
This drives me crazy. The behavior is completely random.
- uncommented all ddebug line, rebuilt+uploaded -> UI was loading perfectly
- Changed hardware to 54173 and PCB to v2, restarted ESP -> not working anymore. barely able to ping the device and UI not loading.
- Set a static IP in config.h, rebuild+upload ->  not working anymore. barely able to ping the device and UI not loading.

I don't see any OOM errors in the serial console, nor anything else then the startup sequence:
[image]

visualapproach, 2023-05-08:
Followed your points and still works perfectly here. 
Btw - You can/should set the IP static in the web ui-network config.
I added some output to the code. Will upload new beta version in a moment. It will give this output:
```Start
.
192.168.4.64
End of setup()
File size: 2711
HTTP > file sent: /index.html.gz (2711 bytes)
File size: 1852
HTTP > file sent: /main.css.gz (1852 bytes)
File size: 491
HTTP > file sent: /function.js.gz (491 bytes)
File size: 3497
HTTP > file sent: /index.js.gz (3497 bytes)
File size: 223
HTTP > file sent: /menu.png (223 bytes)
File size: 8177
HTTP > file sent: /visualapproach.png (8177 bytes)```

dasfliege, 2023-05-08:
Did another clean flash with the latest beta. Ping is alright, but UI is not loading. Everytime is retry to access the UI, ping is breaking up for a few seconds.
This is all i get in the console:
[image]

jondunning, 2023-05-08:
@dasfliege Are you having the same issue when running without the hot tub connected?

visualapproach, 2023-05-08:
> Did another clean flash with the latest beta. Ping is alright, but UI is not loading. Everytime is retry to access the UI, ping is breaking up for a few seconds.
> 
> This is all i get in the console:
> 
> [image]
> 
> 

All file sizes are wrong, I wonder if there is something fishy with your file system. If you can make it work long enough to make an "update" to beta from web ui and then check output again, maybe it will be right

dasfliege, 2023-05-08:
> @dasfliege Are you having the same issue when running without the hot tub connected? i.e Just the esp connected via usb?

Yes, i had it on my bench all the time

dasfliege, 2023-05-08:
> All file sizes are wrong, I wonder if there is something fishy with your file system. If you can make it work long enough to make an "update" to beta from web ui and then check output again, maybe it will be right

I'll give it a try. But atm i don't get it to work at all...
Tried erasing flash, re-flash with or without pre-set wifi settings, etc etc... 

I gonna try it on another computer as well

dasfliege, 2023-05-08:
How do you guys exactly perform a fresh flash?
- Do you perform an "erase flash" prior to each flash operation?
- Do you physically unplug / re-plug the ESP after flashing and before uploading the filesystem?

visualapproach, 2023-05-09:
- No
- No
[image]
1 build filesystem image
2 upload filesystem image
ESP connected with USB for upload

jondunning, 2023-05-09:
I get mine working I just flashed the Filesystem OTA. It's not been connected to the Hot Tub since though (waiting on replacement pump) and also not connected to the same AP because I have it connected via USB in my office.

I'll go and place it outdoors now and see if it's an AP thing. Shooting in the dark but worth testing. I'll report back shortly.

jondunning, 2023-05-09:
OK interesting....It's started misbehaving when connected to my AP outdoors. I have many other esp devices (running tasmota) connected to this AP that are fine. Garden AP details below;

[image]

[image]

[image]

dasfliege, 2023-05-09:
What's pretty confusing though, is that everything was working perfect at the exact same location with an older firmware. I just bricked it by flashing a firmware that fixes my MQTT problem (hydrojets not visible in HA). So i highly doubt that the RF environement has anything to do with it.

visualapproach, 2023-05-09:
Older firmware is still available. Wondering about your file sizes. Can you post screenshot of layzspa.local/dir/

dasfliege, 2023-05-09:
[image]

I don't remember what firmware it was before the MQTT fix. Do you know it?

jondunning, 2023-05-09:
I put my WiFi back to to transmit power to auto and it still works but I notice a channel switch from 6 to 1. This does feel like a WiFi related issue. I’ll run more tests tomorrow

olskar, 2023-05-13:
> Changed transmit TX from Auto to High and the esp now seems to be responding.
> 
> Is it possibly related to this - [esp8266/Arduino#6471](https://github.com/esp8266/Arduino/issues/6471)

I have the same issue. How do I do this?

olskar, 2023-05-13:
I have the same problem, it seems very random. What I THINK is some kind of pattern in it for me is that the issue does not present itself when running the esp with usb, only when connected to the pump.

visualapproach, 2023-05-13:
This is my /dir/
Check the file sizes. If not the same it's probably where you want to check first. At least @dasfliege.
Note that save files will be different, but the html should match.
[image]

dasfliege, 2023-05-14:
Except of the image files, all others are smaller. How can it be, that it still works sometimes if all files seem to be not complete?

dasfliege, 2023-05-14:
Just did an update to Beta 1551 through the UI. And now most of the filesizes matches.
I'll check how it runs now and give feedback.

Interesting though, Current firmware now show 1137. Maybe you forgot to update the fw version in the code?

[image]

visualapproach, 2023-05-14:
That's a good question! I'd try to do updates to beta from web ui to take out some unknowns like your computer.

visualapproach, 2023-05-14:
Hm, have to check but you're probably right

dasfliege, 2023-05-14:
Seems to run perfectly fine until now.
One thing: I do see all informations in MQTT, but can't control the pump. I first have to activate "Take Control" via the UI and then it also works in MQTT. Is this by design? Though in an earlier release i was able to control the pump via MQTT without having to go to the WebUI first...

visualapproach, 2023-05-14:
I'm on it 😸

visualapproach, 2023-05-14:
There you are. Beta version.

jondunning, 2023-05-15:
> I have the same problem, it seems very random. What I THINK is some kind of pattern in it for me is that the issue does not present itself when running the esp with usb, only when connected to the pump.

Just to confirm, you are connected to a Unifi Access Point? If so, can you share the model of the hardware (Just looking for a pattern).

Also, when connected to USB is your ESP connecting to the same AP as when it's connected to the Hot Tub. Mine wasn't, hence the question.

dasfliege, 2023-05-16:
> There you are. Beta version.

Did the update, but getting E13 now on every boot :-( Had to disconnect the ESP for now, as i installed it in the pump and the pool is heating up now. But i have another ESP here for testing.

visualapproach, 2023-05-16:
*sigh* that's a bummer. Let me know how your other ESP works.

closed 2023-12-01 by github-actions[bot]