visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA #507: Hardware test fail

xDCriss, 2023-05-08:
Hardware:
 - Pump model NO54123 Hawaii Hydrojet
 - 4 wires
 - PCB type V1
 - Level shifter type -TXS0108E

Software:
 - Version 4.1.2

Hello everyone, this is the first such project, I went through a few problems (mainly with programming because it's black magic for me) and I stopped at the following moment. The device is programmed when connected to a PC via USB, everything works well, remembers the setting, works smoothly. After connecting to the pump, no reaction and error E13 (I have already determined that it is a connection error with the display). Today I did a Hardware test and the following message popped up: "CIO to DSP pin 3 fail! CIO to DSP pin 4 fail! CIO to DSP pin 5 fail! DSP to CIO pin 3 fail! DSP to CIO pin 4 fail! DSP to CIO pin 5 fail!". 
I suspect that for more advanced people it will be a trifle, but at the moment I totally don't know what to do.
[image]

[image]

visualapproach, 2023-05-09:
Did you connect the connectors? If so it means no connection between esp and end of cable. Measure with a DMM between solder points and wires etc

visualapproach, 2023-05-09:
A broken LLC will cause the errors  not the other way around, but that's probably what you meant. If that's the case you can solder directly to esp and not use the LLC

visualapproach, 2023-05-16:
@xDCriss there was a bug in hardware test for 4 wire pumps. It should be fixed in new beta version (2023-05-16). Sorry for the inconvenience

xDCriss, 2023-05-16:
Just checked beta.

In my case, the changes are not great.

On the first attempt, I connected the connectors together and did a hardware test. I got this feedback "CIO to DSP pin 3 0% badCIO to DSP pin 4 fail!CIO to DSP pin 5 fail!DSP to CIO pin 3 fail!DSP to CIO pin 4 fail!DSP to CIO pin 5 fail!"
On the second attempt, I completely desoldered the connectors and used a jumper between pin 2-6, and 3-7, I did a test and it showed me this "CIO to DSP pin 3 success!CIO to DSP pin 4 fail!CIO to DSP pin 5 fail!DSP to CIO pin 3 success!DSP to CIO pin 4 fail!DSP to CIO pin 5 fail!"

Any ideas?

xDCriss, 2023-05-17:
I will try tomorrow solder directly to esp, maybe that will change something.

visualapproach, 2023-05-17:
Yes. If you want you could move just wire 4 to another pair of pins and change the PIN numbers on the HW config page. Either the chosen esp pins are toast, or the LLC on that channel, or bad solder joints. In either case it may work on other pins.
Tip: use flux when soldering!

xDCriss, 2023-05-18:
There is little progress on my case, I changed pins D3 to D2, D2 to D1, D6 to D5, D7 to D4 and now I have online control of my hot tub but when I connect the display still error E13

xDCriss, 2023-05-18:
This is quite a strange situation because once I saw that some communication with the display because a second before error E13 showed the right temperature

visualapproach, 2023-05-18:
Why did you move all wires? Wire 3 was good according to the hardware test so you should only move the other wires from the bad pins. You probably have the bad connection going to the display now.
Use jumper wires to connect 2 pairs of pins, like you did earlier. Configure them in hw config and save+restart. Note that CIO TX should be paired with DSP RX. Make the hardware test. You need success on pins 3 and 4. Pin 5 is not relevant in your case.
If not successful, change pins in hw config page and move the jumpers accordingly until your get success. Then connect the dsp+cio according to the pinout you got in hw config.

closed 2023-11-23 by visualapproach