visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA #516: Issue updating from 2023-05-10-1137

thedavestewart, 2023-05-19:
So it's been working like a charm for ages now, and I've been applying updates as they come out (because I love being a guinea pig apparently). 

Currently on: 2023-05-10-1137

Web UI Shows: 
Current: 2023-05-10-1137
Master: 2023-05-13-1551
Beta: 2023-05-16-2110

If I click the "MASTER" update as usual, it does what it's more or less always done - pauses for a while. Looks like the site connection errors out. Page cycles back to the home page after some time. But the version won't update. Keeps going in a loop. Even tried moving the access point right next to the darn thing just to make sure it wasn't a connectivity issue. Also tried rebooting the ESP, both via webUI and hard power off at the pump. 

Other than that, it's been just fine, and STILL is perfectly functional, just can't seem to update.

visualapproach, 2023-05-19:
It was just a wrong date in the firmware. It's corrected now, thx

thedavestewart, 2023-05-19:
Thanks! 2023-05-19-1021 fixed it. Closing. 

(Awesome work as always!!)