visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA #520: Update Firmware

Dizzy1985, 2023-05-26:
Wie kommt man von
Firmware-Version: 29.06.2022 auf die aktuelle Version V4 1.3
Geht das irgendwie over the Air oder muss den
 Esp per Hardware Verbindung Flashen?

visualapproach, 2023-05-26:
You can update over the air by editing platformio.ini. See the manual. After that, the best way to update the filesystem is by clicking 'check hardware update' page. Also next version you want to update to, will be a click away

deltamimi, 2023-06-04:
Hallo Zusammen,

leider bin ich im Umgang mit PlatformIO nicht so gut und habe probleme die neue Firmware aufzuspielen.
mir ist nicht ganz klar ob ich das Modul nochmals mit dem PC per USB verbinden muss oder es auch per Wlan geht. Bei meinem Versuch habe ich eine Fehlermeldung bekommen. Error 1
Die Aktuelle FW ist die 2022-07-21 Modul: PRE2021

Es wäre super wenn wir einer den Vorgang etwas genauer erklären könnte da mir die Erklärung im Handbuch nicht klar ist.

liebe grüße

closed 2023-08-06 by visualapproach