visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA #521: PCB_V2 not working with square S100101

HercegAndras, 2023-05-27:
Hi! 
Wanted to thank you for this project, looks very promising!

However it looks not working for me: after connecting the PCB, the square display remained in a kind of turned off state, and the web GUI also shows that the pump and other metrics are missing, switch buttons are not working.

Spec:
-pump: Bahamas (square controller, purchased in 2022, JUN, in EU. code: S100101)
-SW version: V4.1.3, just flashed the ESP y-day.
-PCB_V2, TXS0108E, 6 oin JST with resistors

What I did already:
-Flashed the ESP as it described in the docu via VSCode.
-Setup wifi on webGUI
-HW config: tried all of the 4 different 6 wire options.

The results in the first 3 optioms is the following:
WebUI available but not functional (see screenshot) and the pump controller interface looks like it is in a powered off state, no display/lights. If I press one of the buttons, it plays a short tone. (btw never heard it before)

Any idea how to get it work?
Many thanks!

[image]
[image]
[image]
[image]
[image]
[image]

visualapproach, 2023-05-27:
Hi @HercegAndras 
Pretty assembled board. The RC filters are supposed to be used with the wires in the B-ports. I don't know exactly how they affect  the signal when connected like this. There is several ways you can try, but I recommend that you first make the hardware test. If it passes you can try moving to the B-ports. Or removing the filters and use A-ports.
You could also connect the data wires (blue red green in picture) directly to the esp pins, and not use the LLC. You have free holes beside the nodemcu.
If you move the wires I suggest that you select PCB_v2B and use that pinout. Doesn't matter that you have earlier pcb if you follow the pinout on the hardware config page.

HercegAndras, 2023-05-27:
Many thanks for your quick reply and support ideas!
Actually my brother-in-law did the assembly and soldering, he has better equipment and day to day practice:)
Will do the HW test and move the wires to the B section, then post the updates here. Thanks!

HercegAndras, 2023-05-30:
Finally, it is working!
So the solution is: re-soldering the wires from A ports to B ports, AND using the PCB_V2B pinout!
MANY Thanks!

closed 2023-05-30 by HercegAndras

visualapproach, 2023-06-01:
Cool, thanks for sharing!