visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA #528: [OTHER]

Tops68, 2023-06-02:
**Hardware (please complete the following information):**
 - Pump model: 2021
 - 6 wires
 - PCB type: custom (direct wiring)
 - Level shifter type: /

**Software (please complete the following information):**
 - Version 2023-05-19-1021

**Your message**
Hi,
On reboot, I've found that the ESP sends brefly defaults values (not the latest) through MQTT.
This is very anoying because these wrong values make my domotic badly react.
Examples:
layzspa_message_FLT	0
layzspa_message_GRN	0
layzspa_message_RED	0
layzspa_message_AIR	0
layzspa_message_TMPC	-4
layzspa_message_TGTC	-18
layzspa_message_AMBC	20
layzspa_message_BRT	7
Could you improve this behavior?

visualapproach, 2023-06-03:
There is a 60 seconds delay before first mqtt message is sent after restart. This should be plenty of time for default values to be updated with real values. I fixed a bug that made that delay significantly shorter, but I didn't see your issue before the bug fix. You may try the latest beta to see if it makes any difference.

Tops68, 2023-06-03:
Don't you think this happens when ESP re-register with mqtt broker?
I'm going to upload the beta firmware for testing...

Tops68, 2023-06-03:
Issue solved with beta fw... and... great project, very usefull, thank you!

Tops68, 2023-06-16:
Hi,
Every one hour approximatively, I see in my logs that ESP sends data to MQTT even there is no changes.
Am I wright?

visualapproach, 2023-06-16:
Depends on how you configed 👍 
[image]

Tops68, 2023-06-16:
Strange... I've putted 600.

visualapproach, 2023-06-16:
So it sends every 10 minutes. Possibly HA side is filtering if same message is coming. You can use mqtt explorer (separate windows program) to investigate.

Tops68, 2023-06-16:
Ok, I'm going to investigate.
Conversely, does "0" mean spontaneous sending is disabled?

closed 2023-12-01 by github-actions[bot]