visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA #529: [BUG] SETFULLPOWER not working

Xamude, 2023-06-02:
**Describe the bug**
Heater always  heats at full Power (2000W) , seems SETFULLPOWER is not working to set it to 1000W. 

**To Reproduce**
Steps to reproduce the behavior:
Send over MQTT:
 {"CMD": 18,"VALUE": 0,"XTIME":0,"INTERVAL":0}
{"CMD": 3,"VALUE": 1,"XTIME":0,"INTERVAL":0}
Powermeter says ~ 2050 W Consumption 

no change when i first send CMD:3 and then CMD: 18, always ~2050

**Expected behavior**
Heat with 1000W 

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Hardware (please complete the following information):**
 - Pump model, for example 54123: 54173 Maldives (2018)
 - 4 or 6 wires : 4
 - PCB type: V1
 - Level shifter type: The Red one 

**Software (please complete the following information):**
 - Version: Firmware: 2023-05-08-1052

**Additional context**
Add any other context about the problem here.

visualapproach, 2023-06-02:
I am working on it. Thanks for a good report

visualapproach, 2023-06-03:
Please try latest beta version. I think it's fixed now.

Xamude, 2023-06-03:
Did a short test , Heater running on 1000W :) 
But now i get instant E 13 after ESP Boot. Did 5 resets but instand E 13 :(

visualapproach, 2023-06-03:
Ok, and it you downgrade to master it works without E13? (You can update to master from web ui)

Xamude, 2023-06-04:
No :(
{"CONTENT":"OTHER","MQTT":0,"HASJETS":true,"HASGOD":true,"MODEL":"NO54173","RSSI":-76,"IP":"192.168.170.43","SSID":"meins","FW":"2023-05-19-1021","loopfq":2}

loopfq: 2 ? with FW 2023-05-08-1052 it was around 500 

can i flash back to 2023-05-08-1052 do check ob this version run´s again ?

visualapproach, 2023-06-04:
Oh I thought you came from latest version. I think you only need to resave hardware config

Xamude, 2023-06-05:
> I think you only need to resave hardware config
Upgraded back to Beta and i think the resave did the Trick 
Heater running at 1000W now.
I´ll give feedback in a few day´s due the E13.

Thanks for the quick fix 👍

closed 2023-08-06 by visualapproach

Snotmann, 2024-08-02:
@Xamude how did you get the heater running at 1000 watt ? Did you feed the ESP with mqtt data only ?