visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA #533: [OTHER] Doesn't work: makes a chirp sound when 'on' button pressed, but nothing else.

normanstanleyspencer, 2023-06-04:
**Hardware (please complete the following information):**
 - Pump model: P05328
 - 4 or 6 wires: 6 wire
 - PCB type: d1_mini using this guide https://antanaitis.lt/2022/09/07/wi-fi-on-your-lay-z-spa-make-it-easy-to-remote-control/
 - Level shifter type, Not used (doesn't seem to need one according to the above guide)

**Software (please complete the following information):**
 - Version 4.1.3 (latest)

**Your message**
Hello.  Many thanks for the project.  I've completed a build using the above https://antanaitis.lt/2022/09/07/wi-fi-on-your-lay-z-spa-make-it-easy-to-remote-control/ guide.  All seems to work well - can communicate via WiFi and see the online control panel.

Problem is nothing seems to work on the physical controls when I press buttons on the unit.  When I try and switch the unit on, there is a chirp sound.  I can also press + and - and they do a whoosh-up and whoosh-down sound respectively.  Nothing is shown on the display.  I also can't control the unit via the online control panel.

Any ideas?  Many thanks.

visualapproach, 2023-06-04:
Hi there 
Never heard of that model before.
Did you set the hardware config in the web ui?

normanstanleyspencer, 2023-06-04:
Thanks - sorry, probably copied down the model number wrongly.
It's the Miami (recent purchase so the square pump).  Numbers on the unit are Model: S100102 and on the box: #60001.

Have set the hardware config in the web ui to:
CIO: 6 wire, 2021
DSP: 6 wire, 2021
PCB: Custom
Pinout CIO: 4, 3, 2
Pinout DSP: 7, 6, 5, 0

visualapproach, 2023-06-04:
Does the display show anything at startup? It sounds like the connection to dsp is fine since it makes sound when pressing a button. Probably the cio part that's struggling.
Do the hardware test (see my manual)
If ok go to [debug on](http://layzspa.local/debug-on/)
Run for a couple of minutes and press buttons on display. Then click here [debug off](http://layzspa.local/debug-off/)
Then post log.txt found here [dir](http://layzspa.local/dir/)

normanstanleyspencer, 2023-06-04:
Thanks for your help!

Supplied +5V and GND using a wall USB (I know it said not to, let me know if this messes things up and I'll find a pure 5V power source) - I assumed it shoudn't be a problem as only 5V and GND are connected, but I may be wrong.

Results here:
CIO to DSP pin 3 success!
CIO to DSP pin 4 success!
CIO to DSP pin 5 success!

DSP to CIO pin 3 success!
DSP to CIO pin 4 success!
DSP to CIO pin 5 success!

normanstanleyspencer, 2023-06-04:
CIO to DSP pin 3 success!
CIO to DSP pin 4 success!
CIO to DSP pin 5 success!

DSP to CIO pin 3 success!
DSP to CIO pin 4 success!
DSP to CIO pin 5 success!

Debug ON.  Log test log.txt below:

1685902138:C0,0,0,0,0,0,0,0,0,0,0	,0,0,0,0,0,0,0,0,0,0,0
1685902142:C0,0,0,0,0,0,0,0,0,0,0	,2,0,0,0,0,0,0,0,0,0,0
1685902143:C0,0,0,0,0,0,0,0,0,0,0	,1B,1B,0,0,0,0,0,0,0,0,0
1685902144:C0,0,0,0,0,0,0,0,0,0,0	,0,0,0,0,0,0,0,0,0,0,0
1685902159:C0,0,0,0,0,0,0,0,0,0,0	,2,0,0,0,0,0,0,0,0,0,0
1685902159:C0,0,0,0,0,0,0,0,0,0,0	,1B,1B,0,0,0,0,0,0,0,0,0
1685902160:C0,0,0,0,0,0,0,0,0,0,0	,0,0,0,0,0,0,0,0,0,0,0
1685902165:C0,0,0,0,0,0,0,0,0,0,0	,2,0,0,0,0,0,0,0,0,0,0
1685902166:C0,0,0,0,0,0,0,0,0,0,0	,1B,1B,0,0,0,0,0,0,0,0,0
1685902166:C0,0,0,0,0,0,0,0,0,0,0	,0,0,0,0,0,0,0,0,0,0,0
1685902181:C0,0,0,0,0,0,0,0,0,0,0	,2,0,0,0,0,0,0,0,0,0,0
1685902182:C0,0,0,0,0,0,0,0,0,0,0	,1B,1B,0,0,0,0,0,0,0,0,0
1685902182:C0,0,0,0,0,0,0,0,0,0,0	,0,0,0,0,0,0,0,0,0,0,0
1685902187:C0,0,0,0,0,0,0,0,0,0,0	,2,0,0,0,0,0,0,0,0,0,0
1685902188:C0,0,0,0,0,0,0,0,0,0,0	,1B,1B,0,0,0,0,0,0,0,0,0
1685902311:0,0,0,0,0,0,0,0,0,0,0	,1B,1B,0,0,0,0,0,0,0,0,0
1685902364:0,0,0,0,0,0,0,0,0,0,0	,0,0,0,0,0,0,0,0,0,0,0
1685902365:0,0,0,0,0,0,0,0,0,0,0	,1B,1B,0,0,0,0,0,0,0,0,0
1685902365:0,0,0,0,0,0,0,0,0,0,0	,1,0,0,0,0,0,0,0,0,0,0
1685902367:0,0,0,0,0,0,0,0,0,0,0	,1B,1B,0,0,0,0,0,0,0,0,0
1685902367:0,0,0,0,0,0,0,0,0,0,0	,3,0,0,0,0,0,0,0,0,0,0
1685902368:0,0,0,0,0,0,0,0,0,0,0	,1B,1B,0,0,0,0,0,0,0,0,0
1685902369:0,0,0,0,0,0,0,0,0,0,0	,10,12,0,0,0,0,0,0,0,0,0
1685902370:0,0,0,0,0,0,0,0,0,0,0	,1B,1B,0,0,0,0,0,0,0,0,0
1685902371:0,0,0,0,0,0,0,0,0,0,0	,12,12,0,0,0,0,0,0,0,0,0
1685902371:0,0,0,0,0,0,0,0,0,0,0	,1B,1B,0,0,0,0,0,0,0,0,0
1685902372:0,0,0,0,0,0,0,0,0,0,0	,8,9,0,0,0,0,0,0,0,0,0
1685902373:0,0,0,0,0,0,0,0,0,0,0	,1B,1B,0,0,0,0,0,0,0,0,0
1685902373:0,0,0,0,0,0,0,0,0,0,0	,11,12,0,0,0,0,0,0,0,0,0
1685902374:0,0,0,0,0,0,0,0,0,0,0	,1B,1B,0,0,0,0,0,0,0,0,0
1685902375:0,0,0,0,0,0,0,0,0,0,0	,13,12,0,0,0,0,0,0,0,0,0
1685902375:0,0,0,0,0,0,0,0,0,0,0	,1B,1B,0,0,0,0,0,0,0,0,0
1685902376:0,0,0,0,0,0,0,0,0,0,0	,2,0,0,0,0,0,0,0,0,0,0
1685902377:0,0,0,0,0,0,0,0,0,0,0	,1B,1B,0,0,0,0,0,0,0,0,0

visualapproach, 2023-06-04:
Does the display show anything at startup?

The comms to/from CIO is bad. Your solder joints are good according to the test. So what can be wrong? You're feeding 3.3V to a 5V system. It may or may not work. Try *short* cables to minimize voltagedrop and other interference.

normanstanleyspencer, 2023-06-09:
Having another go at this.
Have shortened the leads but still the same issue.
Display doesn't come on when turned on.

Excuse the soldering (have kept the original leads in hence what seems a mess) but here is the D1 mini: https://ibb.co/br7gr4y (it should handle 5V) and here is the pinout https://ibb.co/zrn6Mh1

Any help very much appreciated!

visualapproach, 2023-06-10:
If it's a Wi-Fi pump from factory chances are it won't work.
First off get yourself some solder flux. Proper solder joints is crucial. Remove everything and connect according to PCB V2B. Do not keep extra wires. They act as antennas and capacitors and will mess with the signals.

normanstanleyspencer, 2023-06-10:
Thanks.  Some progress.

ps it's not a WiFi pump

Tried with another D1 Mini board and using headers to ensure a good connection.

Same issue when using the pinouts I had previously (Pinout CIO: 4, 3, 2; Pinout DSP: 7, 6, 5, 0)

Changed the pinouts and connections to the V2B pinouts and now I have the LED display correctly (IP address + date) but still the same issue with the controls - I can't change anything, but now all buttons make a chip or other sound.

I am not using any extra capacitors or resistors.

Thanks again

visualapproach, 2023-06-10:
what if you log again?

normanstanleyspencer, 2023-06-11:
Thanks.  Results of tests:

CIO to DSP pin 3 success!
CIO to DSP pin 4 success!
CIO to DSP pin 5 success!

DSP to CIO pin 3 success!
DSP to CIO pin 4 success!
DSP to CIO pin 5 success!

1686491418:C0,1,FE,9F,FE,DB,FE,1,FF,F5,FF	,1B,1B,0,0,0,0,0,0,0,0,0
1686491482:C0,1,FE,9F,FE,DB,FE,1,FF,F5,FF	,2,0,0,0,0,0,0,0,0,0,0
1686491482:C0,1,FE,9F,FE,DB,FE,1,FF,F5,FF	,1B,1B,0,0,0,0,0,0,0,0,0
1686491483:C0,1,FE,9F,FE,DB,FE,1,FF,F5,FF	,13,12,0,0,0,0,0,0,0,0,0
1686491483:C0,1,FE,9F,FE,DB,FE,1,FF,F5,FF	,1B,1B,0,0,0,0,0,0,0,0,0
1686491484:C0,1,FE,9F,FE,DB,FE,1,FF,F5,FF	,11,12,0,0,0,0,0,0,0,0,0
1686491485:C0,1,FE,9F,FE,DB,FE,1,FF,F5,FF	,1B,1B,0,0,0,0,0,0,0,0,0
1686491485:C0,1,FE,9F,FE,DB,FE,1,FF,F5,FF	,12,12,0,0,0,0,0,0,0,0,0
1686491486:C0,1,FE,9F,FE,DB,FE,1,FF,F5,FF	,1B,1B,0,0,0,0,0,0,0,0,0
1686491486:C0,1,FE,9F,FE,DB,FE,1,FF,F5,FF	,10,12,0,0,0,0,0,0,0,0,0
1686491486:C0,1,FE,9F,FE,DB,FE,1,FF,F5,FF	,1B,1B,0,0,0,0,0,0,0,0,0
1686491487:C0,1,FE,9F,FE,DB,FE,1,FF,F5,FF	,3,0,0,0,0,0,0,0,0,0,0
1686491487:C0,1,FE,9F,FE,DB,FE,1,FF,F5,FF	,1B,1B,0,0,0,0,0,0,0,0,0
1686491488:C0,1,FE,9F,FE,DB,FE,1,FF,F5,FF	,1,0,0,0,0,0,0,0,0,0,0
1686491488:C0,1,FE,9F,FE,DB,FE,1,FF,F5,FF	,1B,1B,0,0,0,0,0,0,0,0,0
1686491489:C0,1,FE,9F,FE,DB,FE,1,FF,F5,FF	,0,0,0,0,0,0,0,0,0,0,0
1686491489:C0,1,FE,9F,FE,DB,FE,1,FF,F5,FF	,1B,1B,0,0,0,0,0,0,0,0,0
1686491489:C0,1,FE,9F,FE,DB,FE,1,FF,F5,FF	,8,9,0,0,0,0,0,0,0,0,0
1686491490:C0,1,FE,9F,FE,DB,FE,1,FF,F5,FF	,1B,1B,0,0,0,0,0,0,0,0,0
1686491494:C0,1,FE,9F,FE,DB,FE,1,FF,F5,FF	,11,12,0,0,0,0,0,0,0,0,0
1686491494:C0,1,FE,9F,FE,DB,FE,1,FF,F5,FF	,1B,1B,0,0,0,0,0,0,0,0,0
1686491496:C0,1,FE,9F,FE,DB,FE,1,FF,F5,FF	,10,12,0,0,0,0,0,0,0,0,0
1686491498:C0,1,FE,9F,FE,DB,FE,1,FF,F5,FF	,1B,1B,0,0,0,0,0,0,0,0,0
1686491499:C0,1,FE,9F,FE,DB,FE,1,FF,F5,FF	,11,12,0,0,0,0,0,0,0,0,0
1686491500:C0,1,FE,9F,FE,DB,FE,1,FF,F5,FF	,1B,1B,0,0,0,0,0,0,0,0,0
1686491500:C0,1,FE,9F,FE,DB,FE,1,FF,F5,FF	,11,12,0,0,0,0,0,0,0,0,0
1686491503:C0,1,FE,9F,FE,DB,FE,1,FF,F5,FF	,1B,1B,0,0,0,0,0,0,0,0,0

visualapproach, 2023-06-11:
Only thing I can think of is that the esp can't sink enough on its data pin. Try switching place between data D1 wire and audio D7 wire. Don't forget to update the hardware config page by switching those pin numbers.

normanstanleyspencer, 2023-06-11:
> <visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA/issues/533/1586183100@
> github.com>

Thanks - unfortunately no luck. I'll buy another cable and see if that
makes a difference (the wires were quite thin).

metaljay, 2023-06-12:
Just to add, I have a 2023 Paris with S100102 square pump. Also have a D1 mini and have the same strange sound error. I can get the board to connect and access the internet but no control over hot tub.
[image]
[image]

normanstanleyspencer, 2023-06-12:
> <visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA/issues/533/1587766419@

Thanks.  That sounds exactly my situation (albeit you have far nicer solder
joints)

davebuk, 2023-06-12:
Hi all.

I'm having similar issues.

Square pump model S200102. The pump showed no display and the D1 mini wasn't booting (no blue LED). With power still on I hit the manual reset push button on the D1 mini.

It now booted and I have full display and webUI control but the blue LED seems a lot dimmer than when I ran the D1 mini from my USB port.

Perhaps the 5VDC supply is not quite powerful enough to boot the pump system and D1 mini from a cold boot? I'll see if I can get some voltage measurements tomorrow.

metaljay, 2023-06-12:
> Hi all.
>
> I'm having similar issues.
>
> Square pump model S200102. The pump showed no display and the D1 mini wasn't booting (no blue LED). With power still on I hit the manual reset push button on the [...]

Can you post some photos of your wiring and soldering please

davebuk, 2023-06-13:
[image]
[image]

It's quite small guage ribbon cable (26 maybe). Maybe it needs 24 or 22?

davebuk, 2023-06-13:
I have just tried powering using a micro USB lead connected to the USB port instead of the 5V and GND pins but it still won't cold boot. Hitting RESET on the board brings everything back to life.

I'll make a thicker gauge loom over the weekend.

metaljay, 2023-06-14:
> I have just tried powering using a micro USB lead connected to the USB port instead of the 5V and GND pins but it still won't cold boot. Hitting RESET on the board brings everything back to life.
> 
> I'll make a thicker gauge loom over the weekend.

I think we have different problems, I can’t even get it to power the display or control any of the functions. Plus you have a different pump revision. Do you get the sounds? @visualapproach any thoughts or should we bin the board?

davebuk, 2023-06-14:
On a cold boot, mains power OFF and ON, I get a 'chirp' but no display power and the D1 mini LED is OFF. I haven't actually checked it the webUI is accessible. Pressing the D1 mini 'RESET' while powered will then bring on the display and the LED comes on.

My D1 mini is a cheap one from Amazon so will be a clone rather than an official wemos one. There are various posts on the net about boot power issues with clone D1 minis.

Maybe I just need to wire in a waterproof push button 'RESET' and have it hanging out side the pump box?

visualapproach, 2023-06-14:
I don't know what your LED is showing, but I reckon either a power present, or the GPIO is ON or OFF. Either way it doesn't necessarily tell you if it's booted properly or not. Check web ui instead.
Other than that- try use pcb V2B pinout and/or try with a level converter. This will not help with the potential power issue but with comms. As for the power, maybe a capacitor over 5V and GND will help.

metaljay, 2023-06-14:
> I don't know what your LED is showing, but I reckon either a power present, or the GPIO is ON or OFF. Either way it doesn't necessarily tell you if it's booted properly or not. Check web ui instead. [...]

Sorry if this sounds dumb, but surely I can’t just use V2B pin out as I have soldered to specific points on the board?

madsk9023, 2023-06-14:
I experience the exact same problem as you metaljay. Power LED is constantly green and lots of "funny" sounds when I press the buttons. I can access the web interface but when e.g. turning on the pump the slider just "jumpes" back after a second. However I am able to change the brightness on the control panel. My pump is a S100101. I hope someone will be able to solve it - fingers crossed :)

metaljay, 2023-06-15:
> I experience the exact same problem as you metaljay. Power LED is constantly green and lots of "funny" sounds when I press the buttons. I can access the web interface but when e.g. turning on the pump the slider just "jumpes" back after a second. However I am able to change the brightness on the control panel from the web app. My pump is a S100101. I hope someone will be able to solve it - fingers crossed :)

Odd, you’re slightly different to me. What Board config do you have? Same for pins.

madsk9023, 2023-06-15:
> Odd, you’re slightly different to me. What Board config do you have? Same for pins.

I followed this guide: https://antanaitis.lt/2022/09/07/wi-fi-on-your-lay-z-spa-make-it-easy-to-remote-control/

With same config 
[image]

madsk9023, 2023-06-15:
> I followed this guide: https://antanaitis.lt/2022/09/07/wi-fi-on-your-lay-z-spa-make-it-easy-to-remote-control/
>
> With same config [image]

I have also added my log file and a video here
[log.txt](https://github.com/visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA/files/11759549/log.txt)

Video: https://photos.app.goo.gl/JFBNdfnxYtAXdS4X8

metaljay, 2023-06-16:
> Hi again guys! Those sounds are just fine. Not in an opinion way, but that's how it's supposed to sound.
> So communication to and from DSP is okay, since the ESP sends a request to the display which in turn answers which button is pressed. The sounds and log tell us that this part is working.
> CIO is sending display data to the ESP which seems to work according to your log (left side C0...FF). It also requests button presses from the ESP (normally it asks the DSP, but the ESP is hijacking that message). Since there is no reaction on button presses it leads me to think that the CIO doesn't hear the ESP. The data is going through CIO data wire both ways and I guess the ESP cannot drop the voltage enough to register as a "0" on the CIO side. If I'm right it could be caused by circuitry on that pin (LED+resistor) or something else. Therefore I'd try moving CIO DATA to pin D1.

Resolderes to D1 and changed out to reflect. Still get the sounds and no display.

log here:
1686925102:0,0,0,0,0,0,0,0,0,0,0	,1B,1B,0,0,0,0,0,0,0,0,0
1686925121:0,0,0,0,0,0,0,0,0,0,0	,2,0,0,0,0,0,0,0,0,0,0
1686925121:0,0,0,0,0,0,0,0,0,0,0	,1B,1B,0,0,0,0,0,0,0,0,0
1686925121:0,0,0,0,0,0,0,0,0,0,0	,13,12,0,0,0,0,0,0,0,0,0
1686925122:0,0,0,0,0,0,0,0,0,0,0	,1B,1B,0,0,0,0,0,0,0,0,0
1686925122:0,0,0,0,0,0,0,0,0,0,0	,13,12,0,0,0,0,0,0,0,0,0
1686925123:0,0,0,0,0,0,0,0,0,0,0	,1B,1B,0,0,0,0,0,0,0,0,0
1686925123:0,0,0,0,0,0,0,0,0,0,0	,11,12,0,0,0,0,0,0,0,0,0
1686925123:0,0,0,0,0,0,0,0,0,0,0	,1B,1B,0,0,0,0,0,0,0,0,0
1686925123:0,0,0,0,0,0,0,0,0,0,0	,12,12,0,0,0,0,0,0,0,0,0
1686925124:0,0,0,0,0,0,0,0,0,0,0	,1B,1B,0,0,0,0,0,0,0,0,0
1686925124:0,0,0,0,0,0,0,0,0,0,0	,10,12,0,0,0,0,0,0,0,0,0
1686925125:0,0,0,0,0,0,0,0,0,0,0	,1B,1B,0,0,0,0,0,0,0,0,0
1686925125:0,0,0,0,0,0,0,0,0,0,0	,0,0,0,0,0,0,0,0,0,0,0
1686925125:0,0,0,0,0,0,0,0,0,0,0	,1B,1B,0,0,0,0,0,0,0,0,0
1686925126:0,0,0,0,0,0,0,0,0,0,0	,8,9,0,0,0,0,0,0,0,0,0
1686925126:0,0,0,0,0,0,0,0,0,0,0	,1B,1B,0,0,0,0,0,0,0,0,0
1686925126:0,0,0,0,0,0,0,0,0,0,0	,3,0,0,0,0,0,0,0,0,0,0
1686925126:0,0,0,0,0,0,0,0,0,0,0	,1B,1B,0,0,0,0,0,0,0,0,0
1686925127:0,0,0,0,0,0,0,0,0,0,0	,1,0,0,0,0,0,0,0,0,0,0
1686925128:0,0,0,0,0,0,0,0,0,0,0	,1B,1B,0,0,0,0,0,0,0,0,0
1686925128:0,0,0,0,0,0,0,0,0,0,0	,1,0,0,0,0,0,0,0,0,0,0
1686925128:0,0,0,0,0,0,0,0,0,0,0	,1B,1B,0,0,0,0,0,0,0,0,0
1686925128:0,0,0,0,0,0,0,0,0,0,0	,0,0,0,0,0,0,0,0,0,0,0
1686925129:0,0,0,0,0,0,0,0,0,0,0	,1B,1B,0,0,0,0,0,0,0,0,0
1686925129:0,0,0,0,0,0,0,0,0,0,0	,0,0,0,0,0,0,0,0,0,0,0
1686925129:0,0,0,0,0,0,0,0,0,0,0	,1B,1B,0,0,0,0,0,0,0,0,0
1686925134:0,0,0,0,0,0,0,0,0,0,0	,0,0,0,0,0,0,0,0,0,0,0
1686925135:0,0,0,0,0,0,0,0,0,0,0	,1B,1B,0,0,0,0,0,0,0,0,0
1686925135:0,0,0,0,0,0,0,0,0,0,0	,0,0,0,0,0,0,0,0,0,0,0
1686925135:0,0,0,0,0,0,0,0,0,0,0	,1B,1B,0,0,0,0,0,0,0,0,0

colorsone, 2023-06-16:
Same issue over here, tried 2 different boards. Wemos D1 V4 and V3, only sounds when pressing the buttons. Web interface works but nothing more. Hope someone can figure out why.

visualapproach, 2023-06-17:
In @metaljay 's case there is no messages coming through from the CIO. Bad connection or you need the LLC I guess

metaljay, 2023-06-17:
> In @metaljay 's case there is no messages coming through from the CIO. Bad connection or you need the LLC I guess

Can you provide a little more info please. I’ve checked my soldering and it’s on point. See my photo above. Maybe the JST connector has different configuration (I.e. not in the same order compared to other pump models?)

visualapproach, 2023-06-17:
> Maybe the JST connector has different configuration

The dsp comm is working so if you're consistent between the two ports that is probably not it.
With bad connection I didn't only mean the soldering. All zeros in your log means that data from cio is not arriving at the esp for any reason. Be it cable, pin, solder or bad luck

normanstanleyspencer, 2023-06-17:
> <visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA/issues/533/1595672056@

I think many of us get the problem of communication with the CIO despite
changing boards, resoldering etc.  So it seems a systematic problem, and
only? with this pump model. Maybe there is something in there may have been
changes in the wiring in the pump?  Or has anyone tried powering the D1
Mini from an external 5V supply?

mieki3, 2023-06-18:
Hi all, I have the same issue. I also have got clone of the wemos d1 mini. Maybe the issue is with the firmware of this board? Mine board looks exactly like metaljay.

colorsone, 2023-06-18:
> Hi all, I have the same issue. I also have got clone of the wemos d1 mini. Maybe the issue is with the firmware of this board? Mine board looks exactly like metaljay.

I tried Wemos D1 v3 and v4 (clones) both have the same issue.

mieki3, 2023-06-18:
> > Hi all, I have the same issue. I also have got clone of the wemos d1 mini. Maybe the issue is with the firmware of this board? Mine board looks exactly like metaljay.
> 
> I tried Wemos D1 v3 and v4 (clones) both have the same issue.

Ok, thx for the update. So it must be the pump then. I've got s100101, bought it in June 2021...

visualapproach, 2023-06-18:
Maybe one of you could try the pcb_v2b with the llc and filters and see if it helps

metaljay, 2023-06-18:
> Maybe one of you could try the pcb_v2b with the llc and filters and see if it helps

Im happy to but I have no idea what you mean - I can select board 'PCB_V2b' from the web portal (and manually adjust to reflect my actual pin out) but what do you mean about llc and filters?

madsk9023, 2023-06-18:
> Hi again guys! Those sounds are just fine. Not in an opinion way, but that's how it's supposed to sound. So communication to and from DSP is okay, since the ESP sends a request to the display which [...]

I tried this as well, but unfortunately with no luck. I would like to give it a go with your PCB, but unfortunately I do not think I have the skill to source and solder it myself.

craigueh, 2023-06-19:
I have this exact same model and issue except that my displpay works correctly. Any button presses register in Homeassistant and bleep accordingly but I have no control over the CIO unit. No state buttons, web interface or Homeassistant requests seem to get to the CIO.
I'm using a genuine NodeMCU V1.0 and short cables to connect everything. My pinouts are as follows:

CIO_DATA (wire #3) : D1 
CIO_CLK (wire #4) : D2
CIO_CS (wire #5) : D5

DSP_DATA (wire #3) : D6 
DSP_CLK (wire #4) : D4
DSP_CS (wire #5) : D3
DSP_AUDIO (wire #6) : D7

I'm not using an LLC as it doesn't seem to be required for this 2021 (Miami) model but I have tried with and without 1k resistors on the 3 CIO data wires with no difference.

craigueh, 2023-06-19:
Ok, after some tinkering with GPIOs I've gotten it working with the following pins and no other changes necessary:

CIO_DATA (wire #3) : D4
CIO_CLK (wire #4) : D3
CIO_CS (wire #5) : D2

DSP_DATA (wire #3) : D7
DSP_CLK (wire #4) : D6
DSP_CS (wire #5) : D5
DSP_AUDIO (wire #6) : D0

I'm not sure why the above GPIOs work properly compared to the original config but it must just be down to the behaviour of one or two GPIOs in particular (D1/D4) while driving a 5v logic from a 3.3v ttl on the NodeMCU
In any case, all is working well now and I have closed up my unit and waiting for it to heat up :-)

normanstanleyspencer, 2023-06-19:
> <visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA/issues/533/1597423060@

Excellent! Can you let us know which CIO wires were swapped around? So
looks as though they changed the cable pinout?

colorsone, 2023-06-19:
> So originally the pins were as follows:
> 
> CIO_DATA : D1 CIO_CLK : D2 CIO_CS : D5
> 
> DSP_DATA : D6 DSP_CLK : D4 DSP_CS : D3 DSP_AUDIO : D7
> 
> This resulted in a functioning display and beeps etc. but no control over the CIO from the buttons or web interface.
> 
> I changed the pins to the following:
> 
> CIO_DATA : D4 CIO_CLK : D3 CIO_CS : D2
> 
> DSP_DATA : D7 DSP_CLK : D6 DSP_CS : D5 DSP_AUDIO : D0
> 
> Result is that now everything works perfectly. Just to confirm, I am using a NodeMCU, no LLC and currently have 1k resistors on the 3 data lines to the CIO but am not sure if these are actually required.

well, we all have the pinout as you changed it too, so you just had it wrong the first time..

metaljay, 2023-06-19:
Yeh I think @craigueh you have something different to us as you never had the right pin out based on your soldering?

mieki3, 2023-06-19:
> Maybe one of you could try the pcb_v2b with the llc and filters and see if it helps

I will try, but take some time till I get PCB and other components from China to Europe ;)

davebuk, 2023-06-20:
I have tried using 24AWG twisted pairs and a 470uF capacitor across 5V-GND but mine still won't boot from COLD. Nothing on the display and no webUI. Resetting then allows everything to boot and work.

I'm thinking GPIO selection like @craigueh.

I'm going to just change D4 (GPIO 2 which is also driving the on board LED) with D1 and see if that works.

craigueh, 2023-06-20:
> well, we all have the pinout as you changed it too, so you just had it wrong the first time..

Possibly but I don't think so. Originally I used the pin-out from the PDF guide and configured the web interface pin assignments accordingly. After changing the pin assignments around in the config and then physically swapping the pins to match things started working correctly so it definitely looks like some pins have issues with communication to/from the CIO and other pins behave better.

On a side note, I forgot to mention that originally my display did not light up but I was powering it from a 5v pin on the ESP module, essentially in pass-through. Once I wired the 5v for the display directly to the source 5v wire coming from the CIO header the display lit up on the next power cycle. Presumably you guys all have the DSP powered directly but thought I'd mention it just in case anyone did the same as me.

colorsone, 2023-06-20:
> Possibly but I don't think so. Originally I used the pin-out from the PDF guide and configured the web interface pin assignments accordingly. After changing the pin assignments around in the config [...]

I understand, but we use a Wemos and so we already used the different pins.

craigueh, 2023-06-20:
> We (in this thread) have a D1 mini not a NodeMCU.

Yeah I had tested on a D1 mini too originally with the same result. Not sure if they're clones but presume they are as they were a 6 pack from Amazon. That's when I switched to a NodeMCU as I presumed being a genuine board maybe it was more 5v tolerant on GPIOs. 

Did you guys update your platform.io config with the different board type for D1 mini as it defaults to NodeMCU and this threw me for a bit at the start too.

craigueh, 2023-06-20:
Ok, just for giggles I switched back from a NodeMCU to a D1 Mini leaving everything else the same and it works fine. Everything functions as expected.

Maybe it doesn't matter but the only difference I can see is that I'm using 1k resistors on the 3 data lines to the CIO. Have any of you tried this?

metaljay, 2023-06-20:
> 

Yes changed the config to reflect the D1.

Can you send a link to what you purchased as the NodeMCU I saw didn’t have a 5v pin out.
For the price of the boards I’m happy to give it a go. I’d buy the official board if I could but the parts take forever

metaljay, 2023-06-20:
> Ok, just for giggles I switched back from a NodeMCU to a D1 Mini leaving everything else the same and it works fine. Everything functions as expected.
>
> Maybe it doesn't matter but the only [...]

Don’t have access to resistors easily but I can have a Google. Can you send a photo of your board (and soldering). You have the same pump as us correct? S100102

craigueh, 2023-06-20:
Yep, it's the same model SI00102 (Miami 2021).

I'll take some pictures shortly but I'm just using normal headers and short 6in dupont cables to the DSP and CIO connectors.
I snipped the 3 CIO data dupont cables and solderes 1k resistors inline. Very simple setup.

metaljay, 2023-06-21:
Yeh I think your issue is totally different @davebuk 

in other news, I redid the entire thing, fresh board, new JST cables and rebuilt the firmware and data files. Still get the same weird sounds and no display / control. 
I tried to price up a board from the main post but the resistors were out of stock (even if buying a batch of 50). 
so I’m out of luck for now…

really feels like a firmware issue, though I might try a LLC on the D1 (any one got any tips?) 
Thanks anyway for the help @visualapproach

uklinux, 2023-06-22:
Have the exact same issue, D1 Mini, chirp when I press any button, hwtest works ok, but when I engage bubbles the slider move back. Attached boot up display video

https://github.com/visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA/assets/5185507/5e543adf-b74f-4575-b9c5-aca7fbb16b3a

[image]

metaljay, 2023-06-22:
> Have the exact same issue, D1 Mini, chirp when I press any button, hwtest works ok, but when I engage bubbles the slider move back. Attached boot up display video

what pump do you have?
Keen not to muddy the water with the original post on this thread as we cannot even get display on our pumps S100102

if your getting display and on the same pump please post photos of your soldering and let us know config of pins

uklinux, 2023-06-22:
> 

I have the S100102
Pin config is as per https://antanaitis.lt/2022/09/07/wi-fi-on-your-lay-z-spa-make-it-easy-to-remote-control/
Pictures not the best but in a rush atm
[image]
[image]
[image]
[image]

normanstanleyspencer, 2023-06-22:
> <visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA/issues/533/1602455395@

I had the same problem with the display not working on that pump but then
changed the pins and the display now works (but the problem with the
buttons chirping, the webpage bubbles switch reverting and the pump
controls not working are the same - see my posts above)

ReckeDJ, 2023-06-30:
Having the exact same issue here. Followed the tutorial on antanaitis.lt, with a d1_mini v3. Same pin setup.
Also tried previous software versions (4 (latest), 3.5, 3.4) but unfortunately the controls were not working with all the versions.
Also, no start-up information on version 4, but I did have them on 3.4 and 3.5 (ip + firmware version)
Having about the same log as the previous comments.
```
1686836152:C0,1,FE,9F,FE,9F,FE,1,FF,F5,FF	,1B,1B,0,0,0,0,0,0,0,0,0
1686836186:C0,1,FE,9F,FE,9F,FE,1,FF,F5,FF	,1,0,0,0,0,0,0,0,0,0,0
1686836186:C0,1,FE,9F,FE,9F,FE,1,FF,F5,FF	,1B,1B,0,0,0,0,0,0,0,0,0
1686836188:C0,1,FE,9F,FE,9F,FE,1,FF,F5,FF	,3,0,0,0,0,0,0,0,0,0,0
1686836188:C0,1,FE,9F,FE,9F,FE,1,FF,F5,FF	,1B,1B,0,0,0,0,0,0,0,0,0
1686836189:C0,1,FE,9F,FE,9F,FE,1,FF,F5,FF	,10,12,0,0,0,0,0,0,0,0,0
1686836189:C0,1,FE,9F,FE,9F,FE,1,FF,F5,FF	,1B,1B,0,0,0,0,0,0,0,0,0
1686836190:C0,1,FE,9F,FE,9F,FE,1,FF,F5,FF	,12,12,0,0,0,0,0,0,0,0,0
1686836190:C0,1,FE,9F,FE,9F,FE,1,FF,F5,FF	,1B,1B,0,0,0,0,0,0,0,0,0
1686836191:C0,1,FE,9F,FE,9F,FE,1,FF,F5,FF	,11,12,0,0,0,0,0,0,0,0,0
1686836191:C0,1,FE,9F,FE,9F,FE,1,FF,F5,FF	,1B,1B,0,0,0,0,0,0,0,0,0
1686836192:C0,1,FE,9F,FE,9F,FE,1,FF,F5,FF	,13,12,0,0,0,0,0,0,0,0,0
1686836192:C0,1,FE,9F,FE,9F,FE,1,FF,F5,FF	,1B,1B,0,0,0,0,0,0,0,0,0
1686836193:C0,1,FE,9F,FE,9F,FE,1,FF,F5,FF	,2,0,0,0,0,0,0,0,0,0,0
1686836193:C0,1,FE,9F,FE,9F,FE,1,FF,F5,FF	,1B,1B,0,0,0,0,0,0,0,0,0
1686836193:C0,1,FE,9F,FE,9F,FE,1,FF,F5,FF	,8,9,0,0,0,0,0,0,0,0,0
1686836194:C0,1,FE,9F,FE,9F,FE,1,FF,F5,FF	,1B,1B,0,0,0,0,0,0,0,0,0
1686836194:C0,1,FE,9F,FE,9F,FE,1,FF,F5,FF	,0,0,0,0,0,0,0,0,0,0,0
1686836194:C0,1,FE,9F,FE,9F,FE,1,FF,F5,FF	,1B,1B,0,0,0,0,0,0,0,0,0
1686836195:C0,1,FE,9F,FE,9F,FE,1,FF,F5,FF	,0,0,0,0,0,0,0,0,0,0,0
1686836195:C0,1,FE,9F,FE,9F,FE,1,FF,F5,FF	,1B,1B,0,0,0,0,0,0,0,0,0
1686836196:C0,1,FE,9F,FE,9F,FE,1,FF,F5,FF	,1,0,0,0,0,0,0,0,0,0,0
1686836196:C0,1,FE,9F,FE,9F,FE,1,FF,F5,FF	,1B,1B,0,0,0,0,0,0,0,0,0
```
Interestingly enough, on 3.5 and 3.4 there were constant repeating messages like:
```
{"CONTENT":"STATES","TIME":1688142845,"LCK":0,"PWR":1,"UNT":1,"AIR":0,"GRN":0,"RED":0,"FLT":0,"CH1":32,"CH2":51,"CH3":51,"HJT":0,"BRT":7,"TGT":33,"TMP":33,"VTF":32.99999237,"VTMC":32.94417572,"VTMF":91.29951477,"AMBC":20,"AMBF":68,"AMB":20,"VTM":32.94417572,"TGTC":33,"TMPC":33,"TGTF":91,"TMPF":91}
{"CONTENT":"TIMES","TIME":1688142845,"CLTIME":1688142144,"FTIME":1688142144,"UPTIME":364,"PUMPTIME":0,"HEATINGTIME":0,"AIRTIME":0,"JETTIME":0,"COST":0.000202222,"FINT":30,"CLINT":14,"KWH":0.000202222,"KWHD":0.000203409,"WATT":2,"TTTT":-311,"T2R":"Never","MINCLK":5,"bq":0,"bqlen":0}
{"CONTENT":"OTHER","MQTT":-1,"PressedButton":"1b1b","HASJETS":false,"FW":"2022-12-20","MODEL":"MIAMI2021"}
{"CONTENT":"STATES","TIME":1688142845,"LCK":0,"PWR":1,"UNT":1,"AIR":0,"GRN":0,"RED":0,"FLT":0,"CH1":32,"CH2":51,"CH3":51,"HJT":0,"BRT":7,"TGT":33,"TMP":33,"VTF":32.99999237,"VTMC":32.94417572,"VTMF":91.29951477,"AMBC":20,"AMBF":68,"AMB":20,"VTM":32.94417572,"TGTC":33,"TMPC":33,"TGTF":91,"TMPF":91}
{"CONTENT":"TIMES","TIME":1688142845,"CLTIME":1688142144,"FTIME":1688142144,"UPTIME":364,"PUMPTIME":0,"HEATINGTIME":0,"AIRTIME":0,"JETTIME":0,"COST":0.000202222,"FINT":30,"CLINT":14,"KWH":0.000202222,"KWHD":0.000203409,"WATT":2,"TTTT":-311,"T2R":"Never","MINCLK":5,"bq":0,"bqlen":0}
{"CONTENT":"OTHER","MQTT":-1,"PressedButton":"1b1b","HASJETS":false,"FW":"2022-12-20","MODEL":"MIAMI2021"}
{"CONTENT":"STATES","TIME":1688142845,"LCK":0,"PWR":1,"UNT":1,"AIR":0,"GRN":0,"RED":0,"FLT":0,"CH1":32,"CH2":51,"CH3":51,"HJT":0,"BRT":7,"TGT":33,"TMP":33,"VTF":32.99999237,"VTMC":32.94417572,"VTMF":91.29951477,"AMBC":20,"AMBF":68,"AMB":20,"VTM":32.94417572,"TGTC":33,"TMPC":33,"TGTF":91,"TMPF":91}
{"CONTENT":"TIMES","TIME":1688142845,"CLTIME":1688142144,"FTIME":1688142144,"UPTIME":364,"PUMPTIME":0,"HEATINGTIME":0,"AIRTIME":0,"JETTIME":0,"COST":0.000202222,"FINT":30,"CLINT":14,"KWH":0.000202222,"KWHD":0.000203409,"WATT":2,"TTTT":-311,"T2R":"Never","MINCLK":5,"bq":0,"bqlen":0}
{"CONTENT":"OTHER","MQTT":-1,"PressedButton":"1b1b","HASJETS":false,"FW":"2022-12-20","MODEL":"MIAMI2021"}
```
even though I did not press any button.
Unfortunately I cannot do the hardware test, because I do not know how to power the board, I do not have an open wire with 5V.
Any ideas left? What could be the next step?

ReckeDJ, 2023-06-30:
FYI, I just remember a strange behaviour when I was trying with V4. After pressing some random buttons (short + long presses), suddenly the pump went on (and heating too I think). After that, the buttons made a different sound, more like a blocking sound. Could not turn the pump off again, had to plug the power.

[20 more replies]

closed 2023-12-15 by github-actions[bot]