visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA #536: [OTHER] T2R "time to ready" value

Tops68, 2023-06-05:
**Hardware (please complete the following information):**
 - Pump model, 2021
 - 6 wires
 - Custom
 - Level shifter type, /

**Software (please complete the following information):**
 - Version 2023-06-03-0938 (beta)

**Your message**
Hi,
On every 1 degree celsius change, I send an accurate ambiant temp to ESP through MQTT and I also have virtual temp calibrated (checked in web UI).
Today, while ambiant temp was 28°C (outdoor),  I've decided to heat water from 31°C to 36°C.
At the starting time, I've read the T2R value: it' was  equal to 10 hours while it took only only 3.5 hours to reach target temp.
Something seems to be wrong, have you an idea of the problem?

visualapproach, 2023-06-06:
The calibration is probably not ok. I would uncheck the box and save and do another calibration when ambient is relatively stable.
Before that you can look at the calculated R-value in settings.json file. Just to compare to next calibration.
If you can graph virtual temp and actual temp you should ideally see the actual temp flip just as the virtual temp hits next integer value.

Tops68, 2023-06-11:
Hi,
Sorry for delay, I 've made some testing.

-R was equal to 20 while "Calibrated" was already checked in web ui...
-Also equal to 20 after I've unchecked from web ui., saved and rebooted..
-And equal to 7.82078886 now, after a new calibration starting from 2am, while actual temp was 30°C and ambiant temp was approximatively 17°C (outdoor).

According these new R value, at this time: virtual temp is 23.7°C and actual temp is 30°C.
So, the estimation is even worth with the last calibration (in my case, I imagine that R should be higher than 20 for better matching).

visualapproach, 2023-06-11:
No worries.
You say actual temp is 30 before AND after the calibration. It does not make any sense to me.
-the temperature has not dropped at all since 02:00. Calibration should not have happened.
-temp did drop and rose again due to high ambient temp. Question is whether you measure temp with pump on or off.
-temp dropped and rose again by heating. The virtual temp should follow measured temp within a degree.

Tops68, 2023-06-11:
Sorry, I understand your confusion because I've forgotten to explain that I heat water when actual temp reaches 28°C.
You should understand reading my attached graphs.
[image]
[image]

visualapproach, 2023-06-11:
I see now why the calibration gives wrong value. Your thermometer is very noisy and switches between different readings before settling. I'm thinking of a manual input for these cases.

Tops68, 2023-06-11:
Actual temp is noisy because I start my pump every 30 minutes for a 4 minutes duration.

But at 1am, I run it for 3 hours to ensure the calibration process.
On the graph, I can see virtual temp matching actual temp at 01:05 (5 minutes after pump starting) and virtual temp collapsing at 03:16 (just after actual temp drop of 1 degree... ending calibration process).
Between 01:00 and 03:16, actual temp was perfectly constant.

Manual input for R value is an interresting wok around, thank you.

visualapproach, 2023-06-11:
Haha you optimized the thing to trick the vtmp 😂
If you run the pump for 6 min every 45 min instead it will register the temp reading from the pump and you will have the same 'duty cycle'. If pump is running less than 5 min it doesn't trust the measurements.
This is not the issue with the calibration though.
Can you get the time @29C from the graph, and the ambient temp at that time?

visualapproach, 2023-06-11:
Ok you already stated the amb was 17. Do you feed that into the esp? That's necessary for it to work. Just checking.

Tops68, 2023-06-11:
Yes, I did the trick... but only for testing virtualization accuracy ;-)
I've sent Ambient temp every 0.1 degree changes to the esp:
16.8°C at 01:00
...
14.5°C at 03:12
[image]

visualapproach, 2023-06-12:
update available

Tops68, 2023-06-16:
I've tried manual calibration.
With R arround 35, I get more accurate values for T2R.

Thank you!

Tops68, 2023-06-16:
R does'nt seem to be persistent.
I've made a new check right now and I've seen that R has been reset to 20 (without changing anything in the calibration menu).

visualapproach, 2023-06-16:
R value is saved in 'settings.json' after you press 'save' from [this page](http://layzspa.local/calib.html) I tried and it reloads the correct value after reboot. Did you do something to that file, or made a hard reset before saving?

Tops68, 2023-06-16:
I did a reboot this morning and check that R wasn't affected. Reset to 20 happened later.

visualapproach, 2023-06-17:
Okay see if you can spot a pattern if it happens again
/visualapproach

visualapproach, 2023-06-18:
Ha! I found the pitfall described in arduinojson regarding default values. Fixed in beta.

Tops68, 2023-06-19:
Hi,
Tested... it's ok now!

closed 2023-06-19 by visualapproach