visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA #552: [FEATURE RQ]

jo541, 2023-06-27:
**Is your feature request related to a problem? Please describe.**

In the window "SPA config" (config.html) after adding some news command I would like remove only one line.

**Describe the solution you'd like**

I would like have the possibility select only one of them.

**Describe alternatives you've considered**

I've tried modifying the JS and HTML to add a checkbox. But the API url ('/addcommand/') doesn't have an option to pass a specific command to remove. 

**Additional context**

One possible solution is to modify the C++ method to add a new reset option.

In file 'bwc.cpp', line 725 , method add_command

Add A new option   

```
if(command_item.cmd == RESETSELECTEDQ)
    {
        _command_que.clear(command_item.selectedItem);
        return true;
    } 
```

visualapproach, 2023-06-29:
Thx it's on my todo list.

visualapproach, 2023-08-06:
Partly done. We can't delete a single item yet but now we can edit the commands. Make the settings as usual and press "SET" on a specific command to replace it with the new data. Deleting and more features are on the horizon :-)
You are welcome to try beta version.

closed 2023-11-23 by visualapproach