visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA #560: [OTHER] Connection to the s200102 pump (Hawaii Hydrojet Pro)

DJDeel, 2023-07-06:
Hardware 
 - Pump model **s200102**
 - 6 wires
 - PCB type: d1_mini

Hello, I'm not sure how exactly I should connect the wires to d1_mini and what pinouts should be set.

On the website https://antanaitis.lt/ I found information about this setting: Pinout is CIO: 4/3/2 and DSP 7/6/5/0 but it applies to the **S100101** pump. What should be in **S200102**? Same?

I read here about various problems with controlling this pump after connecting. Maybe they are due to improper connections?

[image]

DJDeel, 2023-07-10:
@visualapproach ?

visualapproach, 2023-07-14:
In latest versions you don't edit the code. Just set the pinout in the web UI. Suggest you use a level shifter also.

closed 2023-12-01 by github-actions[bot]