visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA #563: [FEATURE RQ] Update MQTT when filter/chlorine counters are reset

MattFryer, 2023-07-08:
**Is your feature request related to a problem? Please describe.**
I have the module connected to Home Assistant via MQTT and everything is working perfectly with one minor issue. When I press the button in HA to reset either the filter or chlorine counters, HA doesn't display the reset until the next MQTT telemetry push from the module. The module's web interface shows the reset immediately.

**Describe the solution you'd like**
The module should update the MQTT telemetry on reset of either counter. 

**Describe alternatives you've considered**
Now aware of any

**Additional context**
N/A

visualapproach, 2023-07-09:
Got it thank you

visualapproach, 2023-07-14:
Fixed (in beta)

closed 2023-07-14 by visualapproach

MattFryer, 2023-07-14:
Tested and appears to be working perfectly. Many thanks!