visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA #564: [BUG] CIO not reacting, DSP OK

espenfjo, 2023-07-09:
**Describe the bug**
Nothing I do seems to affect the CIO at all. Button presses on the DSP unit or changes in the webui have no effect.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'web page'
2. Click on 'bubbles'

**Expected behavior**
Pump should spew air, it doesnt.

**Screenshots**
[Screenshot of HW config](https://i.imgur.com/XebmKxk.png)

**Hardware (please complete the following information):**
 - S100101
 - 6 wires
 - D1_mini custom solder
 

**Software (please complete the following information):**
 - Current development v4

**Additional context**
hwtest shows that all the pins have contact with eachother. 
Display seems to work as expected since the correct button presses show up in MQTT.

Debug log seems to correctly indicate no connection with the CIO. This log is from a "Bubbles" press on the DSP, as well as turning on the bubbles from the web page.

```
1688894354:C0,1,FE,B7,FE,CD,FE,1,FF,F5,FF	,1B,1B,0,0,0,0,0,0,0,0,0
1688894358:C0,1,FE,B7,FE,CD,FE,1,FF,F5,FF	,12,12,0,0,0,0,0,0,0,0,0
1688894359:C0,1,FE,B7,FE,CD,FE,1,FF,F5,FF	,1B,1B,0,0,0,0,0,0,0,0,0
```

CIO side is soldered like this: 
1. 5v
2. GND
3. D8/15
4. D7/13
5. D6/12  
6. D5/14 ( I know this one is not in use).

visualapproach, 2023-07-09:
Hi. You have given the GPIO numbers instead of Dx. For D8 put 8 in the box and so on.

espenfjo, 2023-07-09:
> <visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA/issues/564/1627674030@
> github.com>

Ah, sorry, I should probably have noted that I did edit the code to use
gpio pins instead of D-pins. The edit at the bottom of the post shows that
the cio behaves exactly the same with the original code and D-pins
configured in the webui.

visualapproach, 2023-07-10:
Okay it seems D1 mini solution only works if lucky.

espenfjo, 2023-07-10:
> <visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA/issues/564/1628841118@

I haven't gone through the code in detail yet, but the ESP communicates
with the cio by pulling the pins low? Ie they are high by default?

So the issue would likely be the ESP not pulling the pins low enough for
the cio to register it as a change?

visualapproach, 2023-07-10:
Rather: CIO expects 5V signals and Esp only delivers 3V3. In some cases that's over the threshold but probably not in your case. You should try with LLC

closed 2023-09-13 by espenfjo