visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA #569: [BUG]

acolod, 2023-07-11:
I have been getting a constant warning in HA that reads: 
`Template variable warning: 'dict object' has no attribute 'RS' when rendering '{{ value_json.RS }}'`

I've been trying to narrow it down and seems to have something to do with the "Ready State sensor (Never, Ready, Not Ready)" and is likely because Home Assistant is receiving an MQTT message from the Lay-Z-Spa device that doesn't include the 'RS' attribute, and it's warning me that it tried to access this missing attribute?

I am on HAOS and running the Development4 version of this code on my wifi controller.

Any thoughts?

visualapproach, 2023-07-11:
I'll fix it. Thx

visualapproach, 2023-07-14:
Fixed (in beta)

closed 2023-07-14 by visualapproach

acolod, 2023-07-17:
> Fixed (in beta)

Thanks!