visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA #574: MQTT very slow (only layzspa, rest ok)

NitramH, 2023-07-14:
Dear all,

I have trouble with MQTT. All the rest was checked, I change the electronics to exclude it, no change.

It seems that it takes seconds until a request is processed (like pump on).
With this, a temperature control is impossible and creates a lot of rubbish actions -> not useable any more.

When turning MQTT off at layzspa, control over HTTP is completely ok.

Any idea where to search and how to log? I use Homeassistant on Proxmox VM and did not had problems before.

Thanks for help, I love the project and would be a pitty to kick it out from HA.

Martin

NitramH, 2023-07-14:
It seems having a look into the HA log, that the spa controller is flooding messages to HA server,
dozen messages within milliseconds

visualapproach, 2023-07-14:
You can check what messages are sent with MQTTExplorer on a PC. I suspect that something is toggling very rapidly and the ESP feels obligated to report it. It wouldn't generate dozens of messages per millisecond but rather tens of messages per second. You can also check browser/inspect/console if it sends more than one batch of messages per 2 seconds. And what's different between them.

visualapproach, 2023-07-14:
My guess is that the ESP tries to connect to mqtt broker and fails. What's the mqtt status on the UI page?

NitramH, 2023-07-14:
The broker is fine since I have 85 entities and all are working.
I will check with mqtt explorer.

visualapproach, 2023-07-14:
I didn't suggest the broker was bad.

NitramH, 2023-07-14:
Ah, sorry, misunderstood.

Status is: MQTT connected
Meanwhile i found  doubled MQTT entities in HA. Maybe from old installation which I migrated. But I do not understand why it worked for weeks with no further change.

visualapproach, 2023-07-14:
No problem. Is the Wi-Fi good?

NitramH, 2023-07-14:
RSSI -60, is good
MQTT Explorer, every 1s a message arrives.

It seems to be something inside my HA environement. There I found old MQTT entities which seems to point to the same topic two times. But I do not know how to delete, HA refuses it

NitramH, 2023-07-14:
mqtt:
  sensor:
    - name: Spa Button
      unique_id: sensor.spa_button
      state_topic: "layzspa/button"
      availability_topic: "layzspa/Status"
      payload_available: "Alive"
      payload_not_available: "Dead"

    # Number of times MQTT has reconnected
    - name: Spa Connect Count
      unique_id: sensor.spa_connect_count
      state_topic: "layzspa/MQTT_Connect_Count"

    # Spa temperature
    - name: Spa Temperature
      unique_id: sensor.spa_temperature
      state_topic: "layzspa/message"
      unit_of_measurement: "°C"
      value_template: "{{ value_json.TMP }}"
      expire_after: 700
      availability_topic: "layzspa/Status"
      payload_available: "Alive"
      payload_not_available: "Dead"

    # Spa target temperature
    - name: Spa Target Temperature
      unique_id: sensor.spa_target_temperature
      unit_of_measurement: "°C"
      state_topic: "layzspa/message"
      value_template: "{{ value_json.TGT }}"
      expire_after: 700

    # Estimated time until target temperature is reached. Needs two+ temperature changes to make calculation
    - name: Spa Time to Target
      unique_id: sensor.spa_time_to_target
      icon: mdi:clock
      unit_of_measurement: "hours"
      state_topic: "layzspa/times"
      value_template: "{{ (value_json.TTTT / 3600 | float) | round(2) }}"
      expire_after: 700

    # Estimated spa energy usage
    - name: Spa Energy
      unique_id: sensor.spa_energy
      device_class: energy
      state_class: total_increasing
      unit_of_measurement: kWh
      icon: mdi:flash
      state_topic: "layzspa/times"
      value_template: "{{ value_json.KWH }}"
      availability_topic: "layzspa/Status"
      payload_available: "Alive"
      payload_not_available: "Dead"
      expire_after: 700

    # Time when Chlorine last was added to water
    - name: Spa Cl Added
      unique_id: sensor.spa_cl_added
      unit_of_measurement: days
      icon: hass:hand-coin-outline
      state_topic: "layzspa/times"
      value_template: "{{ ( ( (now().timestamp()|int) - value_json.CLTIME|int)/3600/24) | round(2) }}"
      availability_topic: "layzspa/Status"
      payload_available: "Alive"
      payload_not_available: "Dead"
      expire_after: 700

    # Time when filter last was changed
    - name: Spa Filter Changed
[... 194 lines cut ...]
#          heat
#        {% else %}
#          off
#        {% endif %}
#      action_topic: "layzspa/message"
#      action_template: >-
#        {% if value_json.RED == 1 %}
#          heating
#        {% elif value_json.GRN == 1 %}
#          idle
#        {% else %}
#          off
#        {% endif %}
#      temperature_state_topic: "layzspa/message"
#      temperature_state_template: "{{ value_json.TGT }}"
#      current_temperature_topic: "layzspa/message"
#      current_temperature_template: "{{ value_json.TMP }}"
#      temperature_command_topic: "layzspa/command"
#      temperature_command_template: '{"CMD":0,"VALUE":{{ value|int }},"XTIME":0,"INTERVAL":0}'
#      power_command_topic: "layzspa/command"
#      payload_on: '{"CMD":3,"VALUE":1,"XTIME":0,"INTERVAL":0}'
#      payload_off: '{"CMD":3,"VALUE":0,"XTIME":0,"INTERVAL":0}'
#      availability_topic: "layzspa/Status"
#      payload_available: "Alive"
#      payload_not_available: "Dead"

NitramH, 2023-07-14:
Is this still necessary or posted by the ESP (sorry for stupid question, but just to be sure):

# Home Assistant package file for https://github.com/visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA
# Save in packages directory in your Home Assistant configuration directory. Create it if it doesn't already exist.
# Add the following to configuration.yaml:
#homeassistant:
#  packages: !include_dir_named packages

visualapproach, 2023-07-14:
Remove all yaml references to layzspa. ESP takes care of it.

NitramH, 2023-07-14:
Removed it. It works but still jittering a lot in Homeassistant. Very crazy behaviour.
I use 11.3.0, maybe need to update?

NitramH, 2023-07-14:
Finally I set the CPU speed to 160 instead of 80. Seems to solve the Jitter.

Really don't know why it was working in the first half of the season wihtout problems :-(

visualapproach, 2023-07-15:
> update from 3.2.0 to 4 using OTA is possible?

Yes it should be fine. After updating you can visit `layzspa.local/update_beta/` to get all the html files updated. It can be problematic sometimes to upload them OTA.

closed 2023-11-23 by visualapproach