visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA #576: [OTHER]

TazmanianV, 2023-07-18:
**Hardware

Pump model Hawaii Hydrojet Pro Square S200102
6 wires
PCB type v2
Wire connection: PCB_V2B
Level shifter: blue
No resistors soldered

**Software

Date: 2022-12-17

Hi all,

I have been running my lazy spa with the wifi module for just about 2 months now. 
However, I notice that the timer of the jets is not running correctly.
Currently I have an uptime of 51 days and according to the app the hydroject should have run for 125 days ;)

I suspect there is a bug in the code?
Can I fix this?

Kind regards 
Michael

TazmanianV, 2023-07-24:
Hi,

A reset of the timers makes no difference, the problem remains.
And if the jets are not working the time does not go up. 
This afternoon I took a screenshot and looked again just now. The jet time has jumped about 1 day and a 2 hours further. While only the pump and heaters worked today.

visualapproach, 2023-07-24:
Alright, did you disengage the warp speed generators?

visualapproach, 2023-07-25:
Just kidding. I found a small bug that I fixed in beta. Let's see if that makes any difference. *Fingers crossed*

TazmanianV, 2023-07-26:
I'd rather have a time machine to go back in time instead of further into the future, I'm getting old ;-p

Thanks for checking.

closed 2023-11-23 by visualapproach