visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA #578: [OTHER] sometimes Button in web UI works, but with delay - most time no reaction of the pump when Button pressed

fiberkev, 2023-07-21:
**Hardware:**
 - Pump model: #54173 primary and #54138 for testing available
 - 4 wires
 - PCB type BWC v2
 - Level shifter type, Transistor based LLC (the red one)
 - capacitors
 - resistors

**Software:**
 - first 2023-05-19-1021
 - Current: Version 4.1.4 - 2023-07-14-1259

**Your message**
Hi all, 

first a big thank you for your work and this great project. 
i tried to get my pump include to my smarthome system (with IOBroker) - unfortunately first step to control over the web UI wouldn´t work than thought. I tried a few things but now i need your help or an tip. ;-) 

First Test yesterday:
The web UI showed the right actually temp. (same as in DISP) 
When i used the Button "Take Control" in the web UI the Button "flipped back" to "off". The same happend with the "Bubbles-Button". 
I tried it about 10 times - switched on / off via web UI. And then suddenly the Pump started the Bubbles - only about 5 sek. then stopped again. I played several times and the Pump reacted sometimes with delay. 

Then i tried the Buttons on the DISP. Here was also a delay. 

Test today: 
today i tried again some things and hoped to find out when the Pump is starting. 
But - don´t ask me why - suddenly pump or heater (not the bubbles) started. Because i was afraid that the pump would be damaged i switched the power off in this moment. Since then now i always get an E013 Error on the DISP. 
Could something have broken? I have already replaced the ESP8266 and the LLC... 

The situation is still the same, however, that I can sporadically control the pump via the web UI. (but the DISP no longer works. 

See the Video attached at 1:03 min > there the Pump starts bubbles. (sorry for the Error "piep")

How can I find out what is causing the two problems?

I readed about an "hardware test" - but i can´t find an information how i can start the test. 

I have also a new Pump #54138 here and tested with thist one too. No change. 

Thanks in advance

[image]
[image]
[image]
[image]
[image]

https://github.com/visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA/assets/140191757/dca0e00b-98bc-4dab-b201-87a556beb5ff

visualapproach, 2023-07-25:
Hi! Page 16 in the manual is about the hardware test. Possibly you may have reversed the RX/TX pins. Try flipping them on the hardware cofig page. If that doesn't work you could try H1A/H2A ports instead.

KHouse75, 2023-07-25:
Could be a wifi distance or signal quality issue.  Do you drop packets when you continuously ping the esp8266?

fiberkev, 2023-07-25:
thanks for your answers - i try it and give feedback

fiberkev, 2023-07-28:
No success at the H1B/H2B ports, so tried the H1A/H2A ports instead. 
But i am confused with the results of the HW-Test. 

Test with bridge beetween D1 > D4 and D2 > D5 (see photo) 

Result HW-test:
CIO to DSP pin 3 success!
CIO to DSP pin 4 fail!
CIO to DSP pin 5 fail!

DSP to CIO pin 3 success!
DSP to CIO pin 4 fail!
DSP to CIO pin 5 fail!

Hardware Config:
CIO data/td/rx 1 	clk/tx 2		cs/ld "emty"
DSP data/td/tx 4	clk/rx 5		CS/ld "emty" audio "emty"

_____

Test with bridge beetween D1 > D4 and D2 > D5 + flipped Settings RX/TX:

Result HW-test:
CIO to DSP pin 3 fail!
CIO to DSP pin 4 success!
CIO to DSP pin 5 fail!

DSP to CIO pin 3 fail!
DSP to CIO pin 4 success!
DSP to CIO pin 5 fail!

Hardware Config:
CIO data/td/rx 2 	clk/tx 1		cs/ld "emty"
DSP data/td/tx 5	clk/rx 4		CS/ld "emty" audio "emty"
+ "custom" activated

@ KHouse75: Thanks for your notice but no packet loss with ping at the IP of the ESP8266. 

[image]

shoud i use a new circuit board without the resistors and capacitors?

visualapproach, 2023-07-28:
Seems like your connection D2-D5 is not working. ESP, LLC, wires or solder joints.

closed 2023-11-23 by visualapproach