visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA #584: [OTHER]

LateWiksi, 2023-08-07:
**Hardware (please complete the following information):**
 - Lay-Z-Spa Miami S100101 (B-BWP05328)
 - 6 wires

PCB details and wiring:
Display directly to the Wi-Fi module
- https://wiksi.biz/i/image/QNT9Rg8Yvs9dFWgB.png
Wi-Fi module to the pump:
- https://wiksi.biz/i/image/tkhlGkMnGpQyxH5Y.png

**Software (please complete the following information):**
 - Original firmware version was: 2023-05-13-1551
 - Newly tried firmware: the one available on "master" branch

**Problem**

I was yesterday doing weekly maintenance for the Spa and while that I ran Spa pump etc. for a while via the touch screen. Everything went smoothly as normal. Suddenly later yesterday I noticed that Wi-Fi module is having network issues, so I decided remotely reboot it http://layzspa.local/reboot/. The problem continued, so I restarted Wi-Fi AP's etc. and after that I decided to do a remote reboot for the Wi-Fi module again. After that pump did not respond anymore via Wi-Fi. I went to the touch screen and it did not react at all. I powered whole Spa off and powered it on again. Nothing happened. Touch screen/display is dead, but I can hear beeping sounds coming very fast but not loud, like beep, beep, beep, beep... I opened up the Wi-Fi module box and noticed that the blue led is blinking very fast and beeping comes in the same pace. My tough was ok, the firmware is some how corrupted. Let's try to upload the firmware again to he device.

I followed the instructiosn for the upload and I did not get any error. I did re-attach the Wi-FI module to the pump after upload and tried to power on the pump. Same blue led blinking and beep sounds in a fast pace.

Any ideas?

VSC log from the firmware flashing:
` Executing task: C:\Users\Wiksi\.platformio\penv\Scripts\platformio.exe run --target upload 

Processing nodemcuv2 (platform: espressif8266@^4; board: nodemcuv2; framework: arduino)
--------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif8266/nodemcuv2.html
PLATFORM: Espressif 8266 (4.2.1) > NodeMCU 1.0 (ESP-12E Module)
HARDWARE: ESP8266 160MHz, 80KB RAM, 4MB Flash
PACKAGES:
 - framework-arduinoespressif8266 @ 3.30102.0 (3.1.2)
 - tool-esptool @ 1.413.0 (4.13)
 - tool-esptoolpy @ 1.30000.201119 (3.0.0)
 - tool-mklittlefs @ 1.203.210628 (2.3)
 - tool-mkspiffs @ 1.200.0 (2.0)
 - toolchain-xtensa @ 2.100300.220621 (10.3.0)
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 48 compatible libraries
Scanning dependencies...
Dependency Graph
|-- ArduinoJson @ 6.21.3
|-- WebSockets @ 2.4.1
|-- PubSubClient @ 2.8.0
|-- ESPAsyncTCP @ 1.2.2
|-- WiFiManager @ 0.16.0
|-- EspSoftwareSerial @ 8.1.0
|-- DallasTemperature @ 3.11.0
|-- BWC_unified
|-- ArduinoOTA @ 1.0
|-- ESP8266WiFi @ 1.0
|-- DNSServer @ 1.1.1
|-- ESP8266HTTPClient @ 1.2
|-- ESP8266httpUpdate @ 1.3
|-- ESP8266WebServer @ 1.0
|-- LittleFS @ 0.1.0
|-- OneWire @ 2.3.7
|-- Ticker @ 1.0
Building in release mode
Retrieving maximum program size .pio\build\nodemcuv2\firmware.elf
Checking size .pio\build\nodemcuv2\firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [=====     ]  54.3% (used 44444 bytes from 81920 bytes)
Flash: [======    ]  64.4% (used 673136 bytes from 1044464 bytes)
Configuring upload protocol...
AVAILABLE: espota, esptool
CURRENT: upload_protocol = esptool
Looking for upload port...
Auto-detected: COM5
[... 384 lines cut ...]
esptool.py v3.0
Serial port COM5
Connecting....
Chip is ESP8266EX
Features: WiFi
Crystal is 26MHz
MAC: 48:3f:da:64:7a:84
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 921600
Changed.
Configuring flash size...
Compressed 1024000 bytes to 55756...
Writing at 0x00300000... (25 %)
Writing at 0x00304000... (50 %)
Writing at 0x00308000... (75 %)
Writing at 0x0030c000... (100 %)
Wrote 1024000 bytes (55756 compressed) at 0x00300000 in 1.7 seconds (effective 4877.7 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
===================================================================================== [SUCCESS] Took 8.89 seconds =====================================================================================
 *  Terminal will be reused by tasks, press any key to close it. `

LateWiksi, 2023-08-07:
Blue led is also keeping constantly after firmware upload is completed.

visualapproach, 2023-08-07:
Hi! Run the pump without the device to see if it has any errors like E02. If it runs fine, and the ESP was working before, my guess is something physical has happened. Corrosion or similar.

LateWiksi, 2023-08-07:
Pump works normally without Wi-Fi module. So most likely the Wi-Fi module it self or wiring has happened something then...

LateWiksi, 2023-08-07:
This is odd...
Used network sniffer while the Wi-Fi module is attached to PC via USB and found out the device has lost some of it's configs, but responds via other IP address, but still seems to remember Wi-Fi network password.

HW config at the moment, seems to be messed too:
https://wiksi.biz/i/image/LGblH8968v26x74W.png

LateWiksi, 2023-08-07:
After trying to change settings from the WebUI, restarted Wi-Fi module and it did not wake up anymore. Applied firmware to the Wi-Fi module again as well data files, and now flashing firmware seemed to work.

I really do not know what has happened, but clearly software got somehow messed, because partly e.g. network configurations were missing.

Now I  did setup HW config as follows:
Select your CIO (pump) model: 6 wire, 2021 (air)
Select your DSP (display) model: 6 wire, 2021 (air)
Select your PCB: V1

After restarting Wi-Fi module beeping noise stopped and touch screen display is working as expected. Also Wi-Fi module now responds. Blue led is still blinking, but I suppose that's normal.

Really weird, but seems to work now.

gorper, 2023-08-07:
Can´t provide any more intelligent than that I have seen some strange things. I have a working system running 2023-07-14-1259, but I am fiddling with a second identical esp module. That one worked fine with 2023-05-13-1551 firmware. I updated to nem master thru web interface then to beta. As i remember master was 413 and beta 414 at that time. Then all is broken. But in my case i thought I was the stupid one to upgrade to beta then down to mater back and forth just to see differences. So I make a boy guess that the file system got corrupted when doing like this. When I occasionally get it running some of webpage's works and other not and it seems that some files has 0 bytes and it correspond to the pages not working, also i have seen the firmware statement on web page refer to 2023-05-13-1551 but it should have stated something newer in my opinion.  But behavior is as described blu led flicker and so on. My next try will be to factory reset the esp because it don´t begin to work to upload the 2023-05-13-1551 again. I tried so many things so i am not shure what broke or semi fixed it. So I will try to reset it and begin from scratch.

LateWiksi, 2023-08-07:
Ran some tests and looks like it everything works now as expected.

The symptoms I had:
- Display stuck or after reboot completely black, did not react
- Continuous beep sounds in fast pace when starting "beep, beep, beep"
- Wi-Fi connection "died" (honestly seemed to use different IP) did not respond to ping or http
- Wi-Fi AP "Lay-Z-Spa Module" did not wake (should be brought up when connection fails)
- Firmware upgrade did not work, the beeping problem was continuous

How I was able to "fix" the problem:
- I ran a network scan for home network, this can be done for for subnet such as 192.168.0.0/24 with nmap from Linux CLI: nmap -n -sP 192.168.0.0/24)
- Connected to the device with a new IP address (I had static IP configured, but device somehow did not use it and static IP checkbox wasn't even checked and IP address config was changed).
- Checked hardware config an noticed it's not what it's supposed to be, fixed hardware settings
- Rebooted ESP

After these steps beeping sounds stopped and display started to work again.
Based on this I think the problem was faulty hardware config.

After Wi-Fi module started to work again, I ran a firmware upgrade. Upgrade was successful and now also Lay-Z-Spa Module Wi-Fi AP seems to be brought up when Spa is not capable of connection to the Wi-Fi network that it is supposed to.

Why settings were messed up? I do not know. I had no any script automation for Spa that would have ran after me manually being operating Spa device from the display. I did reboot ESP using web-browser a couple of times. Perhaps reboot request came too closely each other, during the first reboot or something and that messed the config. Can really tell.

visualapproach, 2023-08-09:
Yeah something has changed and firmware update buttons are broken. I've put two days on it but it looks like we need more memory than I can free up...

gorper, 2023-08-09:
Well I gave you a coffe for your efforts.

visualapproach, 2023-08-09:
Jag såg det 😃 det blir många koppar! Thanks a lot my friend! Motivation level raised!

visualapproach, 2023-08-13:
Update problem fixed. Update from VCS/platformio once and the UI updates should work again.

closed 2023-08-13 by visualapproach