visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA #585: [OTHER] 0 Byte files after update

Darkem1, 2023-08-07:
Hi all,

I recently updated directly over the "Check firmware update page".
After the update, the following files have 0 Byte and the sites are not working:

chkupdatefw.html.gz
config.html.gz
favicon.ico.gz
favicon.png
hwconfig.html.gz
furelise.mel
mqtt.html.gz
upload.html.gz

How can I upload the correct files? If I try to upload the whole image with Visual studio, I always get an error.
Do I really have to take out my board and flash it new over USB? :(

Regards

visualapproach, 2023-08-07:
You can upload each file via the web ui upload file page. Choose the .gz files.

Darkem1, 2023-08-07:
Unfortunatly not, because the upload page also has 0 byte :(

visualapproach, 2023-08-07:
Update again and hope it works: http://layzspa.local/update/

Otherwise upload with usb

Darkem1, 2023-08-07:
No chance, everything stays like it is. I was already afraid that I would have to unscrew everything again...

closed 2023-11-23 by visualapproach