visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA #588: [BUG] Updated via webUI, now can't update again.

acolod, 2023-08-09:
**Describe the bug**
I clicked the "BETA: Update data file, then firmware" button on the chkupdatefw.hml page. It seemed to react the same way as usual but the FW number didn't change once the main page reloaded.  Checking for update again, it does not show anything available for Beta or Master. Everything seems to be working as expected still but would like to fix this so I can update.

**Screenshots**
[image]

visualapproach, 2023-08-09:
see end of this thread #584 
Thanks for your report!

visualapproach, 2023-08-13:
Fixed, but you have to upload the old fashioned way once.

closed 2023-08-13 by visualapproach