visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA #596: [BUG] Watchdog crash when index page makes websocket connection

brainthee, 2023-08-15:
**Describe the bug**
After setting the hardware configuration and rebooting - viewing the index page causes a websocket connection which in turn crashes the D1 mini I'm using. Keeping the index page open causes in effect a crash loop as the browser tries to re-establish the websocket connection causing it to crash.

**To Reproduce**
Steps to reproduce the behavior:
1. Download and flash v4.2.0 on a D1 mini (board info below)
2. Join to your wifi network and connect on new IP
3. Go to the Hardware configuration page, set the model. In my case I have a 4 wire Miami so am setting 54123 for both DSP and CIO.
4. Reboot
5. Load the index page; see the crash in the monitor:

```
--------------- CUT HERE FOR EXCEPTION DECODER ---------------

Soft WDT reset

Exception (4):
epc1=0x4000dd2b epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

Level1Interrupt: Level-1 interrupt as indicated by set level-1 bits in the INTERRUPT register

>>>stack>>>

ctx: sys
sp: 3ffffaa0 end: 3fffffb0 offset: 0160
3ffffc00:  00000000 00006447 210624dd 0073f23e  
3ffffc10:  00000000 0187b5dd 4010b31c 4010ad34  
3ffffc20:  4010b31c 00006447 00000000 4010ad34  
3ffffc30:  4010b31c 00000010 00000100 4021d242  
3ffffc40:  00005912 00000100 4029a920 00000004  
3ffffc50:  00000000 0021002f 00000000 4010ad04  
3ffffc60:  4010b21c 4010ad34 00000110 4021d52d  
3ffffc70:  00000001 00000003 4029a8ec 00000002  
3ffffc80:  10017e81 00000000 00000000 00000000  
3ffffc90:  40290000 00000004 4029a7b4 00000003  
3ffffca0:  00000000 00000004 00000001 00000001  
3ffffcb0:  00000001 00000001 00000000 00000007  
3ffffcc0:  4029a7e0 00000007 4029a7e8 4010b21c  
3ffffcd0:  4010ad34 00000110 4010ad04 4021c619  
3ffffce0:  00000000 00000005 4029a808 00000006  
3ffffcf0:  00000001 4010b158 00000000 00000007  
3ffffd00:  d4fdfac2 4046a978 3ffffd20 40240b14  
3ffffd10:  4010ad44 3ffefccc 3ffefdbc 40214719  
3ffffd20:  4010b21c 0110014f 80000000 00000000  
3ffffd30:  00000001 4010ad04 4010ad34 40240b40  
3ffffd40:  00000001 4010ad44 00000002 4021d200  
3ffffd50:  3fff42ac 4010ad04 4010ad34 4021cf7a  
3ffffd60:  00000000 00000080 4010adfc 4022dc6a  
3ffffd70:  00000000 0d6f6e69 00000000 4010ad34  
3ffffd80:  4010ad04 3fff42ac 00df00df 80000000  
3ffffd90:  40240af8 3fff3f8c 001c001f 80000000  
3ffffda0:  00000000 00000001 4010adfc 4022e052  
3ffffdb0:  00000000 3ffffde0 00000020 40101063  
3ffffdc0:  40232728 00000030 00000014 00000001  
3ffffdd0:  00000002 3fff2c74 00000002 401001a9  
3ffffde0:  3fff2c01 4010adfc 3ffffe00 40241bbd  
3ffffdf0:  00001890 00000312 3ffe8678 402334a8  
3ffffe00:  02a00000 a98ffaf0 00000000 4010ad04  
3ffffe10:  3fff4209 00000000 3ffe8678 40100c02  
3ffffe20:  48910a08 4010adfc 3ffffe54 00000001  
3ffffe30:  3fff4209 00000002 3fff2c74 40233530  
3ffffe40:  00000000 000058f8 824dd2f1 0066df56  
3ffffe50:  4010ad04 3fff3f8c 3fff3bcc 3fff3fbc  
3ffffe60:  3ffffeb0 3fff3bcc 3fff2cd4 40232aac  
3ffffe70:  00000209 3fff4002 3fff3bcc 3fff2c74  
3ffffe80:  00000000 000000e5 3ffe8678 40100c02  
3ffffe90:  3fff000a 4010b194 3fff2cd4 00000000  
3ffffea0:  3fff3fbc 3fff2cd4 3fff3bcc 40232b0c  
3ffffeb0:  3fff2e80 3fff2e74 00000000 3fff07c8  
3ffffec0:  3fff0a3c 3fff07c8 3fff3fbc 3fff0a84  
3ffffed0:  3fff2cd4 3fff0a88 3fff0a89 402531fd  
3ffffee0:  00000000 3fffdad0 3fff0350 00000030  
3ffffef0:  00000018 00000000 3ffec1f4 3fff3364  
3fffff00:  3fff0000 00000000 00000020 3fff07c8  
3fffff10:  3fff0a3c 3fff3fda 3fff3fbc 40258299  
3fffff20:  00000014 3fff07c8 00000000 4024fc6d  
3fffff30:  00000000 00000312 3ffe8678 40100c02  
3fffff40:  40288718 3fff3f8c 3fff2adc 3fff3364  
3fffff50:  3fff07c8 00000008 3fff3fbc 4024f335  
3fffff60:  3fffdc80 3fff2adc 3fff3f8c 4024f148  
3fffff70:  4028b8e9 3fff2adc 3fff3f8c 4028b8fb  
3fffff80:  3fff3fcc 3fff3fbc 00000000 3fffdab0  
3fffff90:  40288347 00000000 3fff3f8c 4028cbbf  
3fffffa0:  40000f49 3fffdab0 ffffff00 40000f49  
<<<stack<<<

0x4021d242 in WebSockets::write(WSclient_t*, unsigned char*, unsigned int) at ??:?
0x4029a920 in etharp_output at ??:?
0x4021d52d in WebSockets::sendFrame(WSclient_t*, WSopcode_t, unsigned char*, unsigned int, bool, bool) at ??:?
0x4029a8ec in etharp_output at ??:?
0x40290000 in wifi_softap_get_config_default at ??:?
0x4029a7b4 in etharp_output at ??:?
0x4029a7e0 in etharp_output at ??:?
0x4029a7e8 in etharp_output at ??:?
0x4021c619 in WebSocketsServerCore::broadcastTXT(unsigned char*, unsigned int, bool) at ??:?
0x4029a808 in etharp_output at ??:?
0x40240b14 in WebSocketsServerCore::runCbEvent(unsigned char, WStype_t, unsigned char*, unsigned int) at ??:?
0x40214719 in sendWS() at ??:?
0x40240b40 in WebSocketsServerCore::runCbEvent(unsigned char, WStype_t, unsigned char*, unsigned int) at ??:?
0x4021d200 in WebSockets::write(WSclient_t*, unsigned char*, unsigned int) at ??:?
0x4021cf7a in WebSocketsServerCore::handleHeader(WSclient_t*, String*) at ??:?
0x4022dc6a in String::changeBuffer(unsigned int) at ??:?
0x40240af8 in std::_Function_handler<void (bool, void*), std::_Bind<void (WebSocketsServerCore::*(WebSocketsServerCore*, WSclient_t*, String*))(WSclient_t*, String*)> >::_M_invoke(std::_Any_data const&, bool&&, void*&&) at ??:?
0x4022e052 in String::concat(char const*, unsigned int) at ??:?
0x40101063 in malloc at ??:?
0x40232728 in AsyncServer::_accept(tcp_pcb*, int) at ??:?
0x401001a9 in std::function<void (bool, void*)>::operator()(bool, void*) const at ??:?
0x40241bbd in std::function<void (bool, void*)>::function(std::function<void (bool, void*)> const&) at ??:?
0x402334a8 in AsyncTCPbuffer::_handleRxBuffer(unsigned char*, unsigned int) at ??:?
0x40100c02 in umm_free_core at umm_malloc.cpp:?
0x40233530 in AsyncTCPbuffer::_rxData(unsigned char*, unsigned int) at ??:?
0x40232aac in AsyncClient::_recv(std::shared_ptr<ACErrorTracker>&, tcp_pcb*, pbuf*, int) at ??:?
0x40100c02 in umm_free_core at umm_malloc.cpp:?
0x40232b0c in AsyncClient::_s_recv(void*, tcp_pcb*, pbuf*, int) at ??:?
0x402531fd in tcp_input at /local/users/gauchard/arduino/arduino_esp8266/origin/tools/sdk/lwip2/builder/lwip2-src/src/core/tcp_in.c:501 (discriminator 1)
0x40258299 in ip4_input at /local/users/gauchard/arduino/arduino_esp8266/origin/tools/sdk/lwip2/builder/lwip2-src/src/core/ipv4/ip4.c:1467
0x4024fc6d in mem_malloc at /local/users/gauchard/arduino/arduino_esp8266/origin/tools/sdk/lwip2/builder/lwip2-src/src/core/mem.c:210
0x40100c02 in umm_free_core at umm_malloc.cpp:?
0x40288718 in ppRecycleRxPkt at ??:?
0x4024f335 in ethernet_input_LWIP2 at /local/users/gauchard/arduino/arduino_esp8266/origin/tools/sdk/lwip2/builder/lwip2-src/src/netif/ethernet.c:188
0x4024f148 in git2glue_err at /local/users/gauchard/arduino/arduino_esp8266/origin/tools/sdk/lwip2/builder/glue-lwip/lwip-git.c:118
 (inlined by) esp2glue_ethernet_input at /local/users/gauchard/arduino/arduino_esp8266/origin/tools/sdk/lwip2/builder/glue-lwip/lwip-git.c:494
0x4028b8e9 in ethernet_input at /local/users/gauchard/arduino/arduino_esp8266/origin/tools/sdk/lwip2/builder/glue-esp/lwip-esp.c:365
0x4028b8fb in ethernet_input at /local/users/gauchard/arduino/arduino_esp8266/origin/tools/sdk/lwip2/builder/glue-esp/lwip-esp.c:373
0x40288347 in ppPeocessRxPktHdr at ??:?
0x4028cbbf in ets_snprintf at ??:?

--------------- CUT HERE FOR EXCEPTION DECODER ---------------

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 3424, room 16 
tail 0
chksum 0x2e
load 0x3fff20b8, len 40, room 8 
tail 0
chksum 0x2b
csum 0x2b
v000a6000
~ld
��␃�␓�o�s��g|�␄l␌�␌l ␃␜b�␛␃␌�|␃;�␃$�g�␌�g�␀␄l`␃��r�d�l␛�␌␌␄�
Start
...............................
startmqtt
Failed to read mqtt.json. Using defaults.
10.yyy.zzz.xxx
End of setup()
3576
```

**Expected behavior**
Not crash :)

**Screenshots**
No screenshots 

**Hardware (please complete the following information):**
 - Pump model: 54123
 - 4 wires
 - PCB type - custom, pins 7/6 for CIO and pins 3/2 for DSP. Pins go from D1 mini to level figter which goes direct to CIO/DSP plugs.
 - Level shifter type: 4 Channel IIC I2C Logic Level Converter Bi-Directional Module 3.3V to 5V Shifter (https://www.amazon.co.uk/gp/product/B082F6BSB5/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1)

**Software (please complete the following information):**
 - Version: v4.2.0
 - Latest VSCode/PlatformIO/etc - lemme know if more info is needed.

**Additional context**
I tried back in June to do this but came across other issues where the DSP and CIO didn't seem to be talking to each other. I then used a v3 firmware and it worked but would occasionally cause the pump to error p2 and need a power cycle to recover. I shelved it and left it till now. I've completely erased and reflashed to new so hoping to get some decent debugging in. If you need any more info or tweaks made I'm comfy enough to make the changes. Keen to find a way to make this work before summer is up :D Thanks!

visualapproach, 2023-08-15:
Thanks for splendid report. I will examine

brainthee, 2023-08-15:
Oke doke, had some chance to do some in-situ testing (laptop on an inflatable ring... slightly scared lol!). v3.5.1 is working fine with 
```
#define NO54123 
// and
const int CIO_RX = D7;
const int CIO_TX = D6;
const int DSP_TX = D3;
const int DSP_RX = D2;
```

I had to tweak `WebSocket.js` at line 293 as it was `val == 'toggleUnit'` and I'm guessing it's meant to be `cmd ==`. After that, I could take control fine and both display and CIO are behaving. I'm just running it for a while now to see if it errors with P-02 as it did before.

I know v3 branch is legacy but was keen to get a working setup to know a good starting point for v4 (also to validate my hardware setup).

Will do some more testing/digging on v4 in the next day or so and feedback what I find.

visualapproach, 2023-08-16:
Think I fixed it. It was out of memory again. I think the 4-wire is needing a bit more memory than the 6-wire. There is two heaps, one is around 40KB and the other 16KB. HTTPS connections need about 26KB so I moved a lot of stuff to the 16KB heap, but it became overfull appearantly. So moved back some stuff and the https (update page) seem to work still.

closed 2023-11-23 by visualapproach

brainthee, 2023-11-23:
Sorry I didn’t reply! We ended up taking everything down for winter. I’m 90% sure I flashed it and it worked… but it was a while ago and I tried a few different versions. Thank you for spending time looking into and sorting it. When summer comes back round I’ll try and remember to let you know for sure :-)

visualapproach, 2023-11-24:
@brainthee no worries, I'm just tidying. You are welcome to reopen or open new issues when you want.