visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA #604: VS Code Problems (6)[BUG]

andrewdaviesj, 2023-08-21:
When I upload the VS Code, I get 6 problems flag up. 

I have been following the step, click the first upload button (the right arrow) and it start to compile. During compile the following 6 problems report each time. The process continues and after the upload says SUCCESS.

/Code/.pio/libdeps/nodemcuv2/OneWire/OneWire.cpp",
	"owner": "cpp",
	"severity": 4,
	"message": "unused variable 'reg' [-Wunused-variable]",
	"startLineNumber": 167,
	"startColumn": 24,
	"endLineNumber": 167,
	"endColumn": 24

/Code/.pio/libdeps/nodemcuv2/OneWire/OneWire.cpp",
	"owner": "cpp",
	"severity": 4,
	"message": "unused variable 'reg' [-Wunused-variable]",
	"startLineNumber": 201,
	"startColumn": 24,
	"endLineNumber": 201,
	"endColumn": 24

/Code/.pio/libdeps/nodemcuv2/OneWire/OneWire.cpp",
	"owner": "cpp",
	"severity": 4,
	"message": "unused variable 'reg' [-Wunused-variable]",
	"startLineNumber": 229,
	"startColumn": 24,
	"endLineNumber": 229,
	"endColumn": 24

/Code/.pio/libdeps/nodemcuv2/PubSubClient/src/PubSubClient.cpp",
	"owner": "cpp",
	"severity": 4,
	"message": "comparison of integer expressions of different signedness: 'unsigned int' and 'int' [-Wsign-compare]",
	"startLineNumber": 523,
	"startColumn": 16,
	"endLineNumber": 523,
	"endColumn": 16

/Code/.pio/libdeps/nodemcuv2/WebSockets/src/SocketIOclient.cpp",
	"owner": "cpp",
	"severity": 4,
	"message": "'void WebSocketsClient::loop()' is deprecated [-Wdeprecated-declarations]",
	"startLineNumber": 175,
	"startColumn": 23,
	"endLineNumber": 175,
	"endColumn": 23

/Code/.pio/libdeps/nodemcuv2/WebSockets/src/SocketIOclient.cpp",
	"owner": "cpp",
	"severity": 4,
	"message": "'void WebSocketsClient::loop()' is deprecated [-Wdeprecated-declarations]",
	"startLineNumber": 175,
	"startColumn": 28,
	"endLineNumber": 175,
	"endColumn": 28

[image]

Many Thanks

visualapproach, 2023-08-21:
I reckon there is a question in there between the lines..
So here's the answer: Yes it's okay and you can look the other way. It's just warnings in some of the libraries used (not my code).

andrewdaviesj, 2023-08-21:
Thanks for the quick reply. 

About 5 months ago I got some PCBs V2 made and have been using one with the St Lucia (Miami2021) hot tub with no issues. 
After a power outage a couple weeks ago the hot tub wouldn't work, so took out the PCB and connected the tub without the Wi-Fi mod, all was working but obviously not ideal. 

I made a new PCB with the new code and new TXS0108E and new NodeMCU and i can't even get the LEDs to come on on the dsp. Web page works every time, but nothing is working. Set the hardware to Miami2021 & PCB v2, but still no joy. Made another board and another TXS0108E, same issue. Swapped things in and out but still no joy....

To night I thought I would try something different and tried the D1 Mini variant from here https://antanaitis.lt to try and swap stuff around. 

Again web page is ok, nothing seems to go to the dsp and nothing to the control board. 

The only common issue I can see are these errors. Any suggestions? 

The unused variable isn't ideal, but I would suspect it wouldn't cause any issues. The other 3 might be causing issues, these errors are the same if I make board = nodemcuv2 or board = d1_mini. As you say they are pointing to the libraries so not surprising. 

Do you have any libraries that i can try? 

Thanks

andrewdaviesj, 2023-08-22:
> You don't have to worry about those warnings.
> If you have a 4-wire pump you should note that the dsp tx/rx is reversed in the hardware config page. Not nice but it was necessary for other reasons

Thanks,Is a 6 wire Miami 2021 version. Can’t remember the full number but it ends in 01. This is the same set up as previously working but with the new code and parts. I got somewhere with the d1 mini last night where by it would work if the usb was connected to the d1 mini. I use a cable without a 5v line to save any potential issues. So tx, ex, gnd. When I open a serial monitor the system kicks into life…. Not looked into this as yet but interested if anyone has any suggestions. Many ThanksAndrew Davies