visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA #606: [BUG] Web interface inaccessible, MQTT and control still working

iawa2k, 2023-08-23:
**Describe the bug**
After 12-24 hours I cannot access the web interface however I can control the device via MQTT or the physical hot tub.

**To Reproduce**
Steps to reproduce the behavior:
Leave the device running for around 12-24 hours

**Expected behavior**
Being able to access the web interface

**Screenshots**
N/A

**Hardware (please complete the following information):**
 - Pump model - S200102
 - 6 wires
 - PCB_V2B
 - Red Transistor Based 8 channel bidirectional level converter

**Software (please complete the following information):**
 - Version: 2023-08-19-2340

**Additional context**
NodeMCU Lolin v3 Board in use is: https://www.amazon.co.uk/gp/product/B074Q2WM1Y/ref=ppx_yo_dt_b_asin_title_o04_s00?ie=UTF8&th=1

iawa2k, 2023-08-23:
Good call on command schedule, didn't even notice there was one in there for Reboot ESP.

Added a script to my server that checks for text on the webpage and pushes the MQTT command to reset and also log the time it did it in a file so I can have the data to hand of roughly when it died (it's running every 30 mins).

Also added a HomeBridge control lol

[image]

visualapproach, 2023-08-25:
Do the new beta version behave better?

iawa2k, 2023-08-25:
I will install now, this is the log from my script so far:

08/23/2023 12:36:26, Page Down, Rebooting ESP
08/25/2023 03:10:12, Page Down, Rebooting ESP
08/25/2023 03:40:11, Page Down, Rebooting ESP
08/25/2023 04:10:11, Page Down, Rebooting ESP
08/25/2023 04:40:11, Page Down, Rebooting ESP
08/25/2023 05:10:11, Page Down, Rebooting ESP
08/25/2023 05:40:11, Page Down, Rebooting ESP
08/25/2023 06:10:11, Page Down, Rebooting ESP
08/25/2023 06:40:11, Page Down, Rebooting ESP
08/25/2023 07:10:11, Page Down, Rebooting ESP
08/25/2023 16:56:21, Page Down, Rebooting ESP
08/25/2023 17:10:12, Page Down, Rebooting ESP

I'll report back in a day or so after installing beta to see how it is.

zolonn, 2023-08-28:
I currently experience the same problem - MQTT working, web UI: "ERR_CONNECTION_REFUSED"
board is PCB_V2B, pump model S100101 (6wire), level converter TXS0108E
FW version is 2023-08-19-2340 compiled via Visual Studio Code

visualapproach, 2023-09-06:
Yes it's very weird. I know the little ESP is pushed to the limits but it should either work or not. I get occasional builds that doesn't work and then I change the smallest thing and the problem disappears.

iawa2k, 2023-09-06:
Are there any logs you need or anything I should try hardware wise? 

The monitor I have set up to check the page and reboot ESP over MQTT command works well so again, not a priority for me personally but not sure how many others it’s affecting.

I did use headers for the boards so may flash another board (I bought three) and swap that out in the coming week just to make sure it’s not a dodgy board etc.

Feels like maybe it’s filling memory or something but not an easy one to fix most likely.

Does it log by default? Just wonder if a toggle to not having it write much to it might help?

visualapproach, 2023-09-06:
Thanks, but I don't need logs for the moment. There is no default logging but you can turn it on @ layzspa.local/debug-on/ however it will not log anything useful for this problem. So I think one would have to dig deeper into the web server library and make custom logging. Last time I checked it wasn't out of memory but it just didn't send the whole files.

visualapproach, 2023-09-06:
You could try beta 2023-09-06-2103
I made a small change in how the index page loads. May be better.

visualapproach, 2023-09-12:
Okay I'll see what I can do

iawa2k, 2023-09-12:
Mine has been better on 2023-09-06-2103, requiring fewer reboots but it’s still occurring every 2 days or so.

Apologies, busy week and didn’t get chance to update.

ZoXx, 2023-09-18:
same problem here.

ZoXx, 2023-09-18:
latest software and pump model S100101

visualapproach, 2023-09-18:
Latest as in yesterdays beta?

ZoXx, 2023-09-18:
tried not the beta channel. Iam on the official latest release (v4.2.1)

visualapproach, 2023-09-18:
Try beta and see if it's still problematic

ZoXx, 2023-09-18:
update in progress. 
Will tell you the next hours / days, if it is working or the webinterface still not available.

ZoXx, 2023-09-18:
Iam wrong. Iam on firmware Firmware: 2023-08-16-1957. Update not possible to newest and beta. Its downloading for 1-2 seconds and then refresh to the home screen without update. restart didnt change anything.

ZoXx, 2023-09-23:
Any ideas how i can resolve my problem with update? :(

visualapproach, 2023-09-23:
Download as first time install and upload via OTA or cable

visualapproach, 2023-09-27:
Http server still stops after long runtime. I'm logging some stuff but it takes time to get it to fail so be patient.

ZoXx, 2023-09-28:
Iam on version 2023-07-14-1259 now.
But got blank screen for updates:
[image]

visualapproach, 2023-09-28:
Yes the secure connections is very demanding on the ESP and some earlier versions can have problems with this function. So normal update is needed. Either OTA or by cable from visual studio code

ZoXx, 2023-09-28:
Latest Firmware Versin via Github is: 2023-08-16-1957
OTA Update still not working after flashing again.

visualapproach, 2023-10-02:
Okay, so I had it running for several days now and stress tested it with max no of clients (4-5) which fragmented the heap but not to the point of failure. Then this night I saw it dropping 5 KB from iRam and 11 from heap and web server broke. The time was when my router made an auto restart. Is that the case for you guys also? I hope I can pinpoint the problem but need to investigate some more...

visualapproach, 2023-10-23:
New (development branch) version may fix this issue #606

closed 2023-11-23 by visualapproach