visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA #612: red levelshifter not to be found

kallejoken, 2023-08-25:
**Hardware (please complete the following information):**
 - not yet available (helsinki is ordered)
 - probably 6 wires
 - PCB type V2 is ordered

i can't find the red levelshifter in germany - only the blue one can be found.
is the red one absolutely necessary?

visualapproach, 2023-08-26:
It's a DIY project and expect to have to try both if the first don't work. I think the red one has a better track record but it's not guaranteed. I bet eBay ships to Germany and there are both colors for sale.

vantech14, 2023-09-25:
@kallejoken did the blue shifter work for you?

kallejoken, 2023-09-25:
> @kallejoken did the blue shifter work for you?

Yes and the Red shifter too

kallejoken, 2023-09-25:
> @kallejoken did you need the 10k resistor? Thanks. I am making one of these up.

No Not resistor

closed 2023-11-23 by visualapproach