visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA #618: [OTHER]

GRINDER1975, 2023-09-03:
**Hardware (please complete the following information):**
 - Pump model S200102
 -6 wires
 - VT2B
 - RED

**Software (please complete the following information):**
 - Version most recent 

**Your message**

hi i have managed to build this project and gain control of the pump fia the web interface, but seem to have lost any ability to control the pump from the touch panel.

have checked soldering etc cannot seem to find issue

any help would be really appreciated

Marc

visualapproach, 2023-09-03:
Are your buttons enabled in spa config?

GRINDER1975, 2023-09-03:
yes they are.    i have run the hardware test and fail.  Been scratching my head most oif the day.  have several boards and can get none of them to work.  wonder if pcb fault.  its probably someting stupid. 
[image]

GRINDER1975, 2023-09-03:
wires in the middle for hardware test

GRINDER1975, 2023-09-03:
CIO to DSP pin 3 fail!
CIO to DSP pin 4 fail!
CIO to DSP pin 5 fail!

DSP to CIO pin 3 fail!
DSP to CIO pin 4 fail!
DSP to CIO pin 5 fail!

GRINDER1975, 2023-09-03:
this is what i am getting

GRINDER1975, 2023-09-04:
progress of sorts.  are you able to point me to where i need to look re pin 4 fail and pin 5 0%bad?

CIO to DSP pin 3 success!
CIO to DSP pin 4 fail!
CIO to DSP pin 5 0% bad

DSP to CIO pin 3 success!
DSP to CIO pin 4 success!
DSP to CIO pin 5 success!

visualapproach, 2023-09-06:
Either bad solders or wires, or broken LLC or Esp. Likely the latter since it's working one way. Guess it's not impossible for a bad solder to make a diode but I'd try new parts.

GRINDER1975, 2023-09-06:
I am beginning to think this may be a faulty PCB.  i have built 5 from the same batch and none of them working.  Will order some new 2b Boards.

GRINDER1975, 2023-09-06:
Answer to your last question - built another :)

visualapproach, 2023-09-14:
same problem as @metaljay?

metaljay, 2023-09-15:
> same problem as @metaljay?

Oh wow I think it could be, did you also order from JLPCB? @GRINDER1975 
I appear have 6 dead boards out of my batch. Wasn’t sure if I should contact them

closed 2023-11-23 by visualapproach