visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA #623: [BUG] no matching function for call to configTime

Gifford47, 2023-09-10:
**Describe the bug**
the custom ntp feature doesn´t compile.
Error: 
```
error: no matching function for call to 'configTime(int, int, String&, const __FlashStringHelper*, const __FlashStringHelper*)'
  450 |     configTime(0,0,wifi_info.ip4NTP_str, F("pool.ntp.org"), F("time.nist.gov"));
      |                                                                               ^
In file included from src\main.h:1,
                 from src\main.cpp:1:
C:\Users\hannki\.platformio\packages\framework-arduinoespressif8266\cores\esp8266/Arduino.h:269:6: note: candidate: 'void configTime(int, int, const char*, const char*, const char*)'
  269 | void configTime(int timezone, int daylightOffset_sec, const char* server1,
      |      ^~~~~~~~~~
C:\Users\hannki\.platformio\packages\framework-arduinoespressif8266\cores\esp8266/Arduino.h:269:67: note:   no known conversion for argument 3 from 'String' to 'const char*'
  269 | void configTime(int timezone, int daylightOffset_sec, const char* server1,
      |                                                       ~~~~~~~~~~~~^~~~~~~
C:\Users\hannki\.platformio\packages\framework-arduinoespressif8266\cores\esp8266/Arduino.h:272:6: note: candidate: 'void configTime(const char*, const char*, const char*, const char*)'
  272 | void configTime(const char* tz, const char* server1,
      |      ^~~~~~~~~~
C:\Users\hannki\.platformio\packages\framework-arduinoespressif8266\cores\esp8266/Arduino.h:272:6: note:   candidate expects 4 arguments, 5 provided
*** [.pio\build\nodemcuv2\src\main.cpp.o] Error 1
```

**To Reproduce**
Steps to reproduce the behavior:
1. clone repo (10.09.23)
2. switch to branch dev_v4 or master
3. compile

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**

**Hardware (please complete the following information):**

**Software (please complete the following information):**
 - Version

**Additional context**
Workaround:
disable custom ntp and revert to stock ntp:
```
configTime(0,0,"pool.ntp.org", "time.nist.gov");
//configTime(0,0,wifi_info.ip4NTP_str, F("pool.ntp.org"), F("time.nist.gov"));
```

visualapproach, 2023-09-11:
It compiles fine here. Do you have up to date libraries?

Gifford47, 2023-09-12:
Yes, all of the libs in the ini file are up to date

closed 2023-11-23 by visualapproach