visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA #627: [OTHER] Connection to the s200102 pump (Hawaii Hydrojet Pro)

codac, 2023-09-13:
**Hardware:**
 - Pump model: S100101
 - 6 wires
 - PCBV2B
 - Level shifter type: red,

**Software (please complete the following information):**
 - Version 4.2.1

How do I need to do a correct wiring regarding the provided photos?

As you can see on the pump side the following colors are set:
[image]

I have wired the cables on the board, following the instructions of the manual.
Unfortunately, when I stick the connectors together, the red/black cable of the board are on the opposite side of the connector of the pump. So I guess my wiring is wrong.
[image]

Any help is aprreciated.

visualapproach, 2023-09-13:
Do not trust the colors! They are not consistent or logical. Find out which is 5V power and that it goes to the right pin. Then the other pins should be in order on your model

codac, 2023-09-19:
Pretending, that 5V is one of the outer pins, GND ist next to 5V and NC is on the oposite side, and that the red wire of the pump is 5V, I measured the GND vs NC and had 0V. So I knew that NC as the exact oposite of 5V is not 5V. 
After that I changed to cables in that order, so that they properly fit the male and female adapter of the pump and it worked.

closed 2023-09-19 by codac

Trikap82, 2024-05-15:
> Pretending, that 5V is one of the outer pins, GND ist next to 5V and NC is on the oposite side, and that the red wire of the pump is 5V, I measured the GND vs NC and had 0V. So I knew that NC as the [...]

Hi Codac... I'm having trouble trying to use a PCBV2B... 

Hardware:

Pump model: S100101 (Lay-Z Spa St-Moritz, square pump) #58113
6 wires
PCBV2B
Level shifter type: **NO LEVEL SHIFTER**    <--- really requisite ?

bought it "plug and play"... well... So I need to investigate by myself.

Could you please post pictures of your final wiring ? Thx

Any help is aprreciated.

visualapproach, 2024-05-15:
> Level shifter type: **NO LEVEL SHIFTER** <--- really requisite ?

In some lucky cases it will work without (but not on my PCBs without bridging). Since your setup doesn't work the answer is yes.