visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA #632: HELP e13 54138

pipppy1, 2023-09-16:
**Hardware (please complete the following

 information):**
 - Pump model, 54138
 - 4 wires
 - PCB type v2b
 - Level shifter type,txs0108e

**Software (please complete the following information):**
 - Version all

i just cannot get it to run e13 all the time
[image]

my settings are [image]

visualapproach, 2023-09-16:
The wires should be in the white squares

pipppy1, 2023-09-16:
thanks just moving them now

pipppy1, 2023-09-16:
what should the pins be

pipppy1, 2023-09-16:
[image]

pipppy1, 2023-09-16:
[image]

visualapproach, 2023-09-16:
As in your picture, but you need to move the wires to those pins. Now in port H1A you have them on D1 and D2, instead of D2 and D5. Same for the other port. move to D4 & D3 as the config says

pipppy1, 2023-09-16:
all wires are moved
[image]

pipppy1, 2023-09-16:
still error 13

pipppy1, 2023-09-16:
ok got it going
took out the D2 ect and just left it at 2 in ect in the config

pipppy1, 2023-09-16:
bit confused though it works for a bit then goes into error 13

closed 2023-09-17 by pipppy1

georgemckee999, 2024-09-11:
Do the wires need to be in H1A and H2A? Mine are in H1B and H2B.

visualapproach, 2024-09-12:
> @visualapproach Do the wires need to be in H1A and H2A? Mine are in H1B and H2B. 
> 
> [image]
> 
> 

The beauty of DIY... They should be where they work. You have to find out by trying I'm afraid.

georgemckee999, 2024-09-12:
Oh that’s a shame, I don’t know what any of them do. I brought this already wired up, I assumed it was in the right location until I saw this photo, now I’m not sure. All I get is e13

visualapproach, 2024-09-14:
Yeah ok. There are many variants of pumps and there is no one size fits all solution, as I understand it. You have to find out which wire goes where. There is only a few combinations though. H1A or B is only to mitigate potential signal problems.
You should update to latest firmware if you haven't, and use the available tools to diagnose. Several threads here about that.