visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA #633: Enabled display buttons. All

pipppy1, 2023-09-17:
when i enable this can i use the panel and web interface at the same time

pipppy1, 2023-09-17:
4 pin model

pipppy1, 2023-09-17:
and also if it gets an error eg like e13 is it possible for it to show up in the web display

visualapproach, 2023-09-17:
> and also if it gets an error eg like e13 is it possible for it to show up in the web display

isn't it already?

pipppy1, 2023-09-29:
no

pipppy1, 2023-09-29:
and one more question if possible. if i press one of the buttons on  the lazy spa does it relay back to the web server to say i pressed it

visualapproach, 2023-09-29:
yes

pipppy1, 2023-09-29:
but mine does not show no errors on the web display . but it will show it on the spa

pipppy1, 2023-09-29:
btw this is a fantastic project

visualapproach, 2023-10-12:
Thx, I think some errors are triggered internally to the display, and not sent to the pump, and therefore the ESP will not know about it. Some errors will be triggered by the pump and it has to tell the display, so we can catch the message.

closed 2023-11-23 by visualapproach